cyoung/stratux #106: infinite SetFreqCorrection error

bradanlane, 2015-11-11:
setting frequency correct to zero is invalid and returns an error which results in teh sdrConfig code being called repeatedly.

Fix submitted in pull request.

bradanlane, 2015-11-11:
fix submitted with pull request #107

jpoirier, 2015-11-11:
Should already be fixed in the gortlsdr library. Setting the freq
correction to zero isn't the problem, the problem is that librtlsdr returns
an erroneous error (-2) when the requested PPM is the same as the currently
configured PPM. The gortlsdr library handles that now.

You really need to read the commit history, and reason out the exact
problem problem before coming to a conclusion.

bradanlane, 2015-11-11:
Thanks @jpoirier, I did read the commit history but it didn't indicate I needed to do something special to get a build to work. 

Thankfully @cyoung posted a followup comment to let everyone know that we need to do "go get -u all".

I was reading the commits and missed chris's important notice.

bradanlane, 2015-11-11:
I've closed the pull request.

closed 2015-11-11 by bradanlane

jpoirier, 2015-11-11:
Roger. Point taken and I'll be sure and annotate/emphasize/explain when
commits cause side effects to external dependencies.

bradanlane, 2015-11-11:
thanks. we all back on the same track again. :-)