cyoung/stratux #115: Makefile changes with 9dda52d cause 'make install' to fail

loather, 2015-11-12:
The 'ln' commands at the end of the Makefile lack -f to clobber any existing symlinks.

cyoung, 2015-11-12:
You should see about making pull requests :)

closed 2015-11-12 by cyoung

loather, 2015-11-12:
I will do that :) I need to better learn how to use Github and Git in general.

kdknigga, 2015-11-12:
This is a fun-ish way to learn:

http://
<http://www.google.com/url?q=http%3A%2F%2Fpcottle.github.io%2FlearnGitBranching%2F&sa=D&sntz=1&usg=AFQjCNHJITlkdUc2PkGUokzfIXWh76-r2w>
pcottle.github.io
<http://www.google.com/url?q=http%3A%2F%2Fpcottle.github.io%2FlearnGitBranching%2F&sa=D&sntz=1&usg=AFQjCNHJITlkdUc2PkGUokzfIXWh76-r2w>
/
<http://www.google.com/url?q=http%3A%2F%2Fpcottle.github.io%2FlearnGitBranching%2F&sa=D&sntz=1&usg=AFQjCNHJITlkdUc2PkGUokzfIXWh76-r2w>
learnGitBranching
<http://www.google.com/url?q=http%3A%2F%2Fpcottle.github.io%2FlearnGitBranching%2F&sa=D&sntz=1&usg=AFQjCNHJITlkdUc2PkGUokzfIXWh76-r2w>
/
<http://www.google.com/url?q=http%3A%2F%2Fpcottle.github.io%2FlearnGitBranching%2F&sa=D&sntz=1&usg=AFQjCNHJITlkdUc2PkGUokzfIXWh76-r2w>

loather, 2015-11-12:
Thanks. i'll give those a look!

commit 724a87de (2015-11-12): Fixes #115.