cyoung/stratux #117: Possible build error?

Axtel4, 2015-11-13:
I pulled the latest archive for 0.4r4 through commit 519b000 and attempted a build. I received a build error concerning a "flag provided but not defined" error. 

pi@raspberrypi ~/stratux $ make
cd dump978 && make lib
make[1]: Entering directory '/home/pi/stratux/dump978'
gcc -c -O2 -g -Wall -Werror -Ifec -fpic -DBUILD_LIB=1 dump978.c fec.c fec/decode_rs_char.c fec/init_rs_char.c
gcc -shared -lm -o ../libdump978.so dump978.o fec.o decode_rs_char.o init_rs_char.o
make[1]: Leaving directory '/home/pi/stratux/dump978'
sudo cp -f ./libdump978.so /usr/lib/libdump978.so
go get -t -d -v ./...
go build "-ldflags -X main.stratuxVersion=`git describe --tags --abbrev=0` -X main.stratuxBuild=`git log -n 1 --pretty=%H`" main/gen_gdl90.go main/traffic.go main/ry835ai.go main/network.go main/managementinterface.go main/sdr.go main/uibroadcast.go
flag provided but not defined: -ldflags -X main.stratuxVersion

Build compiles the packages named by the import paths, along with their dependencies, but it does not install the results.
… 
-ldflags usage description 
... 
See also: go install, go get, go clean.
Makefile:8: recipe for target 'all' failed
make: *** [all] Error 2

Axtel4, 2015-11-13:
Update: I got it to build. I had to edit line 4 in the Makefile from:
BUILDINFO="-ldflags -X main.stratuxVersion=`git describe --tags --abbrev=0` -X main.stratuxBuild=`git log -n 1 --pretty=%H`"

To:
BUILDINFO=-ldflags "X main.stratuxVersion=`git describe --tags --abbrev=0` -X main.stratuxBuild=`git log -n 1 --pretty=%H`"

jpoirier, 2015-11-14:
Sorry about that! It's an edit error on my part for a commit that got the
ci builder running. I'll submit a fix.

-joe

cyoung, 2015-11-14:
Fixed with abd4a50386a79a913245bd905a136919e07f73ea.

closed 2015-11-14 by cyoung