cyoung/stratux #119: WiFi stops working after Raspbian apt-update / apt-upgrade

Axtel4, 2015-11-15:
I don’t know if this is a Raspbian or Stratux issue, or if it belongs on Reddit or on Github, so I’ll place it here.

I am attempting to build a second Stratux system and in doing so, I am updating Raspbian from the Aug 20 Version to the latest version. When updating Raspbian to the latest version via apt-get update / apt-get upgrade I get the following error during start-up:

“starting advanced ieee 802.11 management:hostapd failed!”

I had previously successfully updated my first Stratux system to Raspbian version 4.1.7-v7+ per the information contained in Github Stratux Issue 63:
root@raspberrypi:~/stratux# vcgencmd version
 Sep 23 2015 12:12:01 
 Copyright (c) 2012 Broadcom
 version c156d00b148c30a3ba28ec376c9c01e95a77d6d5 (clean) (release)
 root@raspberrypi:~/stratux# uname -a
 Linux raspberrypi 4.1.7-v7+ #817 SMP PREEMPT Sat Sep 19 15:32:00 BST 2015 armv7l GNU/Linux
 root@raspberrypi:~/stratux# 

Running the hostapd command manually I get the following error:
pi@raspberrypi ~ $ sudo hostapd /etc/hostapd/hostapd.conf
Configuration file: /etc/hostapd/hostapd.conf
Line 2: invalid/unknown driver 'rtl871xdrv'
1 errors found in configuration file '/etc/hostapd/hostapd.conf'

This is the contents of the hostapd.conf file:
interface=wlan0
driver=rtl871xdrv
ssid=stratux
hw_mode=g
channel=1
wme_enabled=1
ieee80211n=1
ignore_broadcast_ssid=0

Thanks in advance for any insight.

skypuppy, 2015-11-15:
apt-get upgrade has caused me nothing *but* problems every time I've
tried to do it, whether it is desktop units, virtual Linux, or the SOC
"embedded" units..  apt-get dist-upgrade has never, ever actually worked
for me either. 
You might like to save any extra work on another medium, and just load
the SD card with the full image from cyoung, then copy your other stuff
back onto the SD card (after re-enlarging it, of course.)

David M.

AvSquirrel, 2015-11-15:
I'd guess that apt-get upgrade may have replaced your hostapd binary.

This project is using an older / customized version of hostapd that supports the proprietary rtl871xdrv driver, needed by the chipset in the "nano" sized WiFi adapters. See if grabbing hostapd from a known-good install and dropping it into /usr/sbin fixes it.

For what it's worth, I've had some success with a [version 2.4 patch](https://github.com/lostincynicism/hostapd-rtl8188) that can be compiled with support for other WiFi devices, but could never quite get ifconfig to reliably assign an IP address at boot.

bradanlane, 2015-11-15:
I also use the 2.4 release of hostapd to support both the official wifi dongle and some other common adapters.

cyoung, 2015-11-21:
@Axtel4 - try this: http://blog.sip2serve.com/post/38010690418/raspberry-pi-access-point-using-rtl8192cu (http://dl.dropbox.com/u/1663660/hostapd/hostapd.zip)

The newer hostapd with patch seems to work fine for me, were there any improvements since then such that we should upgrade in the release image?

Axtel4, 2015-11-24:
Thanks cyoung. 
Sorry about the delay in responding. I have been tied up with other things and finally had a chance to work with Stratux over the last few days.  I discovered I was fighting two parallel issues. One was an incompatible SB Card and the other was the hostapd issue.

SD Card Issue: 
Until recently I have been using 8 GB San Disk Ultra Cards, however I recently picked up a couple of Microcenter 8 GB Class 10 cards on sale and started using them in my R-Pi.   I found that even though I use the sudo shutdown command when shutting down Stratux, these cards would corrupt. When they do corrupt one of the symptoms was a network initialization issue that looked like the hostapd issue. Also, with the Microcenter cards I could not successfully update Raspbian as it would error out part way through the update. However, switching back to the san Disk cards solved these issues.  It looks like with the Microcenter cards YMMV.  Here are a few sites that talk about the SD Cards in the Raspberry Pi if it would be of help to anyone else:
http://www.raspberry-pi.co.uk/2012/06/07/compatible-sd-cards/
http://elinux.org/RPi_SD_cards
http://reprage.com/post/what-are-the-best-sd-cards-to-use-in-a-raspberry-pi/

hostapd: 
After getting the above squared away, I updated Raspbian to Version 4.1.7-v7+ returning to the  hostapd error.  I copied over hostapd from a previous stable build (based on 0.3b4) to my current build and I was able to boot without errors and have WiFi capability again.

I don’t think I am following your connent on 21-November. 
I downloaded the file that was in link above and attempted to install it on my Stratux device with an error that it was the same file. I performed a difference check on the downloaded file and the file from the previous build (diff -s file1 file2  and they are the same file.  I overwrote the original file with the file you linked and I did not detect any difference in performance.

I also have another issue I have been chasing since updating the builds through the 0.4r4 changes. I am running two Stratux systems. The only difference between the two is the WiFi dongle. One uses the Edimax EW-7811Un and the other uses the TP-LINK TL-WN725N. After about an hour of operation I have a “Kernel Panic” with the TP-Link dongle but not with the Edimax dongle. It seems to happen when the receiving device has been asleep for an extended period of time. Here is the start of the “panic”:

pi@raspberrypi ~ $ date
Tue Nov 24 15:24:51 CST 2015
pi@raspberrypi ~ $
Message from syslogd@raspberrypi at Nov 24 16:21:47 ...
 kernel:[ 3613.280030] Internal error: Oops: 17 [#1] PREEMPT SMP ARM

Message from syslogd@raspberrypi at Nov 24 16:21:47 ...
 kernel:[ 3613.599644] Process RTW_CMD_THREAD (pid: 1704, stack limit = 0xb86ee210)

I could provide logs if you need them.

Thanks again for all your help.

Axtel4, 2015-11-24:
Sorry, I hit the wrong button and didn't mean to close it.

cyoung, 2015-11-24:
Well, it seems like you have solved the problem by copying over the hostapd file from the previous installation. The recommended hardware was chosen deliberately and, in my opinion, there's no good reason to troubleshoot the TP-LINK kernel panic issue as an Edimax will cost you some $8 and it works fine.

Axtel4, 2015-11-24:
I am OK with that. I had a TP-Link laying around and thought I'd try it.  The TP-Link is listed as a supporter adaptor in the Readme.md.  Does it need to be deleted then?

cyoung, 2015-11-26:
Might be a good idea to delete it as a supported adapter. Are you able to reproduce the kernel panic if you swap the TP-Link and the Edimax?

Axtel4, 2015-11-26:
I have not seen a kernel panic with the Edimax adapter. I had Stratux on for about 3 hours today at the airport demo-ing for some of our club members. I did not have any issues at all with the Wifi. 

With the TP-Link, I would typically see an event within hour or so after initial turn on. The only reason I noticed it the first time was because I was monitoring the R-Pi via the local console and I saw the kernel dump. After noticing it the first time, I observed it momentarily drop the traffic and weather on my app and then it would start displaying again. I was able to collate that behavior with the kernel dump in the console and with the TP-Link dongle.

bradanlane, 2015-11-26:
I wasn't putting 2 and 2 together. @cyoung, you may remember, early on in my work on the WebUI, I had an issue with my wifi adapter. It too was a Kootek which uses the same chips as the Edimax. It worked for about 2 weeks and then started faulting in Stratux.

It seems these "Edimax compatible" wifi adapters have something wrong. 

My solution was to use a newer version of "hostapd" which has built-in support for the canakit wifi adapter (R5370 based).

Given the data, the wise thing is to update the documentation to list support for only Edimax.

Hal2131, 2015-11-27:
I also have need to use a RT5370 based adapter. What exactly did you do to to install a new version of hostapd?  Did you just use apt-get?
Thanks,
Hal

bradanlane, 2015-11-27:
I had to fetch the source for hostapd 2.4 and compile (with appropriate flags) to get a newer version with support for other wifi adapters. I then had to install the new hostapd. Finally I edited  the hostapd.conf to use the wifi adapter I had.

commit f315ca57 (2015-12-02): Remove TP-LINK from supported adapters. Close #119.
closed 2015-12-02 by cyoung