cyoung/stratux #120: GPS/AHRS page and Weather page do not respect settings choices

bradanlane, 2015-11-15:
The various details pages need to display only the related information based on the hardware which has been enabled. 

If only GPS, then just the map section.
If only AHRS then just the airplane section.
If neither then the page should display an informational message.

Likewise, if a user has configured a 1090ES only Stratux, the weather page should display an informational message.

bradanlane, 2015-11-20:
I should have this fixed by the end of the day. In reviewing the code, I realized it should be refactored so I want to do an extra level of testing.

cyoung, 2018-01-17:
`ng-show="visible_gps"` like in status.html.

Reopen or reference if the feature is added.

closed 2018-01-17 by cyoung