cyoung/stratux #121: Bad Elf ADS-B KickStarter Campaign

rgutlon, 2015-11-16:
Just for reference wanted to pass this along ... interesting market positioning / price points. Notice they are looking for $500K otherwise project will not be funded.

*The first affordable ADS-B weather and traffic receiver with universal app compatibility for pilots.*
https://www.kickstarter.com/projects/bad-elf/bad-elf-ads-b-affordable-weather-and-traffic-recei

ssokol, 2015-11-16:
Two years ago I talked with Canyon Pekham - former engineering lead at
SageTech and now a freelance ADS-B guru - about building a low-cost
receiver. His estimate for a complete design with several rounds of
prototypes was less than $80k including a $10k license for an FPGA
implementation from Mitre. If they already have a working prototype I don't
get the need for a $500k minimum.

I do like the look of the prototype - it would be great if we could get
Stratux down to that size / shape / heat profile. Anyone know how hard it
would be to cook up a Raspberry Pi shield / hat (or whatever they call Pi
daughterboards) that integrates the GPS, AHRS and SDR(s)? Do any of the IO
options on the GPIO header (SPI, I2C, etc.) support a high enough data rate
to handle the raw SDR output?

bradanlane, 2015-11-16:
If you assume somewhere in the range of $100K for all of the R&D then the $500K is probably their "slightly better than" break even for a manufacturing run of 2000 units. I would also assume that manufacturing 2000 units is close to their projected sales figures before they start to see price erosion from competition.

skypuppy, 2015-11-16:
I wouldn't mind having all the "parts' on a daughterboard (shield) but I
have concerns about RF and heat.  And, of course, many people desire a
different GPS and at least one person (me) wants to use a different
IMU.  How would one stack it while maintaining adequate cooling on the Pi2?

emuyshondt, 2015-11-16:
Looks to me like they want to pre-sell a certain number of units before committing to building them. They want to get a commitment from the market before ordering parts and starting up the production line. They give pre-orders a discount, and save themselves the margin they'd have to give to dealers. Their pricing is what they'd want from the market to make it worthwhile to manufacture, support and distribute the product.

The price doesn't look bad for a production product.

ssokol, 2015-11-16:
I thinks a good deal of the heat is specific to the Stratux architecture.
We're running both the radio dongle and one core of the Pi at very close to
full bore to decode UAT signals. An FPGA implementation would be much more
efficient. An ASIC implementation even more so. If an ADS-B-based
"sense-and-avoid" function becomes mandatory for drones we'll probably see
somebody step in with an affordable ASIC. At that point the Pi would just
handle the data management and delivery, much as it does for the GPS now.

rgutlon, 2015-11-16:
The FAQ section (towards the bottom of the KickStarter page) is really interesting.

* ForeFlight is not included in their list of compatible apps.
* They do not plan on including 1090 traffic
* Note the question *How is this different from DIY ADS-B receivers like Stratux?*

skypuppy, 2015-11-16:
Maybe this should be a new number but it is really an add-on to the
existing stratux, just in software.  How hard would it be to use the
stratux setup to catch transponder outputs and make the stratux have
TCAS (?) ability as well?  That is, get other plane's transponder
outputs and calculate distance and location, especially while ADS-B Out
is so rare.

jtremolo, 2015-11-16:
A shield isn't actually all that difficult to design.  As far as heat is concerned, if you provide adequate ventilation holes it'll be fine.  I have a shield designed with an on-board Li-Ion battery charger/controller, USB hub and header pins for RY835AI GPS module.  It still uses the current SDR dongles connected to the USB Hub since I don't have experience designing SDR's.  I'd be happy to share if anyone wants to work on it.

hilts50, 2015-11-16:
jtremolo could you provide a picture of your setup?

jtremolo, 2015-11-16:
Certainly, these were the 2 configurations I was considering.  The purple board in the middle is the shield.  I never had a prototype made due to issues getting the GPS module, so I stalled dev on it. Total cost estimate was ~$25 for the board, components and batteries, then add 2SDRs + GPS + RPi starter kit and it's setup.  Batteries capacity was about 12000 mAh.

http://imgur.com/a/ZfXyG

deandayton, 2015-11-16:
A mode C transponder only transmits its altitude, not it's position. Someone would have to design an antennae/receiver that allowed us to resolve the direction of reception and then somehow derive (or approximate) the distance. Zaon had a product that did this https://en.wikipedia.org/wiki/Portable_collision_avoidance_system but they are now out of business

[http://upload.wikimedia.org/wikipedia/commons/thumb/5/5e/PCAS_Xrx.jpg/220px-PCAS_Xrx.jpg]<https://en.wikipedia.org/wiki/Portable_collision_avoidance_system>

Portable collision avoidance system - Wikipedia, the free ...
A portable collision avoidance system (PCAS) is an aircraft collision avoidance system similar in function to traffic collision avoidance system (TCAS).
Read more...<https://en.wikipedia.org/wiki/Portable_collision_avoidance_system>

ggoodknight, 2015-11-16:
On 11/16/2015 09:22 AM, Steven Sokol wrote:
> I thinks a good deal of the heat is specific to the Stratux architecture.
> We're running both the radio dongle and one core of the Pi at very 
> close to
> full bore to decode UAT signals. 

Hi, I've been lurking (aka sitting on my hands) while I've been playing 
with 4r4 on the ground and maybe about 10 hours in the air, mostly 
between NorCal and South NV. Nice job, Mr. Young and friends.  Am using 
AVARE for the EFB on an 8" wifi android tablet.

Does any of the current code (libraries or stratux) use the GPU? If not, 
moving key pieces of the SDR front end to the GPU could pay some 
dividends. I've not written code for the GPU before and am not sure how 
to proceed in a reasonable straight path to a UAT/1090ES receiver front 
end that would play with stratux or other ADS-B projects. If someone 
else is doing it I'd be happy to learn about the path you are taking, or 
would take. Dreaming, if that works out an upshot (besides more headroom 
on the CPU core tasked with decoding chores) could even be a single 
RTL-SDR (or similar) dual band stratux on the RPi2.

Anyone here have knowledge of any announced/hinted new RPi2 computer 
module? All I've seen is a variation of probably someday.

-Greg good@nccn.net
(an MSEE either between jobs or retired)

bradanlane, 2015-11-16:
I've been buying the SDRs for under $7 each. I doubt there is much lower price possible unless the order is very large.

It's possible to build a base Stratux for about $40 with a 3D printed case, one SDR and integrated battery.

I like this configuration. I consider the next step up to be a USB GPS (rather than the second SDR). This setup is right within a dollar either side of $50.

ssokol, 2015-11-17:
@bradanlane - where are you getting SDRs for $7? Are they the newer R820T2
or the older R820T?

Curious: what's the advantage of the USB GPS? Need to place the Stratux
someplace cool and out of the way and have just the GPS on the glareshield?

ssokol, 2015-11-17:
Greg,

I've not see any news on the compute module in some time. I'm not really
sure how the original version did. It's a good idea but it needed some
additional love from the Sparkfuns and Adafruits of the world to be usable
by most mortals.

I've also wondered if the GPU could be used to offload much of the SDR
processing. It could certainly free up one of the cores, but at this point
with dual ADS-B, GPS, and AHRS I'm seeing something like 120% total usage -
out of a total of 400%. The effort of porting the processing to the GPU
would only be worthwhile if it reduced the heat. Perhaps if it had
optimized instructions for some of the most common tasks (FFTs, ???) it
might be a worthwhile endeavor.

Anyone out there both an SDR and a GPU guru?

Cheers,

-S

bradanlane, 2015-11-17:
Yes, the SDRs are the newer R820T2 devices. Strangely enough I'm getting them thru eBay as a Canadian transaction from a supplier in China.

Go figure.

bradanlane, 2015-11-17:
@ssokol, one of these days I want to build Stratux with profiling turn on.

I can't see why the Stratux code is using close to 100% of a core while the 1090 code only uses about 20%.

The reason I am so confused is that when I'm at my desk, my Stratux received 0 UAT messages and yet the gen_gdl90 process is 98-108% CPU all of the time.

cyoung, 2015-11-17:
This is great news - didn't think we'd see something like this come so soon.

Wonder if anyone is interested now in working on a daughterboard?

Well, things are only going to get cheaper quickly. I'm not sure $299 or even $250 is going to cut it.

cyoung, 2015-11-17:
The network code uses most of the CPU right now. That can be tweaked and it operates on a different thread from other vital functions like UAT capturing and parsing.

bradanlane, 2015-11-17:
By "network code" am I correct in that you are describing the communications between Stratux and the aviation app (eg ForeFlight)?

cyoung, 2015-11-17:
This: https://github.com/cyoung/stratux/blob/master/main/network.go#L186-L209

bradanlane, 2015-11-17:
Thanks. That is what I was trying to refer to. AKA the communications of all the new data to the connected device(s) running the aviation app(s).

skypuppy, 2015-11-17:
Wonder how much cpu is eaten by the polling?

jtremolo, 2015-11-17:
I'd be happy to share the schematics and board design for the board I was working on if anyone is interested in improving it.  It is designed in EAGLE (free for non commercial use)

cyoung, 2015-11-17:
@jtremolo - yes please, how far did you get?

jtremolo, 2015-11-17:
@cyoung It is essentially finished.  The battery charge/management IC is all configured, it has 4 battery level LEDs connected to a push button, push the button the light showing battery % remaining.  The RY835AI solders directly to the board via header pins and is wired to RPi2 header pins on the board. I also added a soft power-up/shutdown push-button.  It would require minor source updates that are described in the webpage link in the schematic.  Essentially, push the button it starts up and latches the power on like it does now, push the button again and Stratux can gracefully shutdown.  Hopefully that would prevent SD card corruption.  The only thing missing is a power in port for charging, I wasn't sure if it should be a USB mini, barrel-jack type, etc. but any 5V power in connector can be added to the board easily with 2 wires.  I never made it because I tried to order parts 3 times and the post office lost all 3 orders.  I got refunds, but it wasted 2.5 months and I gave-up.  I may have to try again soon.  Sorry for the long post, if you have more questions let me know.  Here are all the files:

Here are the files include board preview images.
https://www.dropbox.com/sh/3jw3qhcrzwgq2kr/AACkWLofYXfTD4b3T0QaEmfda?dl=0

jtremolo, 2015-11-18:
I made a few minor updates to the board.  It now has a min USB-B connector for charging.  The updated files are located at the dropbox link I already posted.

dmurray14, 2015-11-19:
That's really cool - you gotta build a couple! Would be a great option to
have.

egid, 2015-11-20:
@jtremolo awesome, the custom board stuff sounds interesting. What would you be using for the battery power source? This would basically put us on par with the Bad Elf / Stratus, it seems like.

Speaking of Bad Elf, their funding trend [doesn't seem too hot](https://www.kicktraq.com/projects/bad-elf/bad-elf-ads-b-affordable-weather-and-traffic-recei/#chart-exp-trend), which is sadly what I expected with a 500k goal for a specialty aviation device.

lynndixon, 2015-11-20:
I love this project :)  @jtremolo any plans on getting these designs to a producer for a small run of the boards?

lynndixon, 2015-11-20:
Love this project! @jtremolo any word on if you are planning on getting a small batch of this design out to some manufacturers for a short run?

jtremolo, 2015-11-20:
@egid I am planning on using these:
http://www.gearbest.com/batteries/pp_192844.html

Wired in parallel would yield a 3.7v x 12000mAh Li-Ion pack.  These are raw, unprotected batteries so they are cheaper.  However, the battery management circuit on the board handles the discharge protection and charge balancing.

@lynndixon I'm working on ordering a couple prototype boards and we will see were it goes.  I will be assembling the first boards myself since I have the equipment already...  I'm currently working with cyoung on a possible v2 of the board.

egid, 2015-11-21:
That sounds pretty cool. I'd be interested in buying an early-run board and helping out with testing, when you're looking for extra coverage.

jtremolo, 2015-11-21:
No problem! I'm doing a final check over the design and I'll be ordering some boards.  The board service I use requires prototype orders in multiples of 3. So, I'll have 2 spares if initial testing works as expected.  Do you want a blank board or an assembled board?

egid, 2015-11-21:
I'd probably be safest with an assembled board. I've got a soldering iron, and I can use it, but slapping wires on the RY835 was the first time I've soldered in probably 15 years and that was stressful enough :D

dmurray14, 2015-11-21:
Sign me up for an assembled board as well!

lynndixon, 2015-11-22:
@jtremolo  I'd be happy with either an assemlbed or non-assembled board :-)  I am pretty decent with a soldering iron so I am comfortable doing it.
I'd be willing to pay cost and shipping to you via paypal, bitcoin or whatever else method you prefer.

cyoung, 2016-01-10:
So what have we learned here?

bradanlane, 2016-01-10:
My reaction has been that any new player either needs to bring substantially more capability to the pilot or be at an even lower price point. 

I also suspect many people thought a Kickstarter offering didn't seem coherent with an established company.

There is also the question of just how big is the target base? Unlike smartphones, pilots do not see huge advantage to upgrade so the market is shrinking. 

I've been flying the same ADSB receiver for almost 4 years and see no motivator to upgrade.

cyoung, 2016-01-10:
So "universal compatibility" wasn't really that big of a selling point?

ssokol, 2016-01-11:
Not really. Once you've bought into an app and have become comfortable with
it, it's rather hard to justify changing. I got started with ForeFlight and
bought a Stratus. One of my friends has been using WingX for years and
bought a Dual XGPS170. Universal compatibility may mean something to those
who haven't invested in a receiver yet, but it probably won't sway somebody
who has.

I also think the price point for the Bad Elf unit was at an unattractive
mid-point between the incumbents and Stratux.

A VC friend of mine is always saying: 10x better or 10x cheaper if you want
to disrupt an existing, settled market. He's probably exaggerating a bit,
but only a bit.

Also, consider that this is an aviation-specific tool. Aviation has two
kinds of people - those who can afford anything and those who can't afford
anything. The former tend to go with Stratus, while the latter either go
with Stratux or go without.

cyoung, 2016-01-11:
>Aviation has two kinds of people - those who can afford anything and those who can't afford anything. The former tend to go with Stratus, while the latter either go with Stratux or go without.

Well, aren't iPad/tablets and portable receivers supposed to be affordable alternatives to certified receivers/displays? If we were to follow that logic, then the people actually using Stratus are the "can't afford"s also. I highly doubt that anyone that would get value out of FIS-B (fair amount of XC flying) couldn't afford $900 for a receiver. Ability to pay and willingness to pay are two different things.

Even so, in the bigger picture I think this has more to do with zero-risk bias than it does willingness to pay.

bradanlane, 2016-01-11:
The irony is that the sweet spot is ADSB weather receiver. Everything else is extra. I've had this conversation with several pilots.

A commercially available ADSB weather receiver for under $99 is the "10x statement".

cyoung, 2016-01-11:
http://www.qsl.net/dl4mea/fpgaadsb
http://wiki.modesbeast.com/Mode-S_Beast:Contents
http://wiki.modesbeast.com/Mode-S_Beast:Advantages_of_a_FPGA
http://miniadsb.web99.de/

Hmm, it looks like the campaign was claiming that FPGA solutions are actually more power hungry than their proposed solution. The DIY FPGA projects above don't go into that side of it much. All but the last guy is using an Altera Cyclone development board. I wonder what the proposed solution is such that it is so much more efficient than most FPGA solutions?

bradanlane, 2016-01-11:
when it comes to "power hungry" it's always hard to believe the vendors' claims. They typically use the "up to X hours". So, given that, here is one reference point:

The Garmin GDL 39 is a dual frequency receiver with GPS. It's battery is 7.4V @ 2100mah. That is 15,540 watts. They claim "up to 5 hours". Thus they are claiming a power consumption rate as low as 3,108 watts/hour. Truthfully, I found about 4 hours to be a close approximation. That places power consumption at 3,885 watts/hour.

My Stratux is pretty close in configuration and uses 1.02A at 5V when operating fully. That is a consumption rate of 5,100 watts/hour. My single SDR+GPS uses 0.75A at 5V. That is a consumption rate of 3,750 watts/hour.

cyoung, 2016-01-11:
So a dual band stratux consumes 31% more power than the GDL39.

Not such a large difference as I'd expect - especially with those SDRs.

skypuppy, 2016-01-11:
Makes me wonder why they think this kind of compute horsepower is
necessary, when an RPi2 can do what is necessary without all the hassle
of programming fpga's, their fickle power requirements, and so on. 
Maybe it's an advertising gimmick.

skypuppy, 2016-01-11:
If it was practical to move to low-power bluetooth instead of wi-fi, we
could save a fair amount on power consumption.  But, alas and alack, the
units we talk to usually want only wi-fi.

cyoung, 2016-01-12:
Looks like they cancelled it. In the meantime, we have actual support on all the apps that they've mentioned that they would have had support on (FlyQ, iFly, iFlightPlanner) - minus iFly which is not released. It is being worked on now and looks like will be out soon. Plus a few more apps.

http://i.imgur.com/qGRZhBK.png

So for everyone that tests the betas, contributes code, reports issues, or helps influence the decision of EFBs to support stratux - we are achieving the campaign's goal "of bringing an affordable, universal ADS-B receiver to market".

gitmoregrits, 2016-01-12:
Cyoung said: "So for everyone that tests the betas, contributes code, reports issues, or helps influence the decision of EFBs to support stratux - we are achieving the campaign's goal "of bringing an affordable, universal ADS-B receiver to market"."

:+1: You and others have done a fantastic job with this project.  I've learned a ton along the way about not only the implementation but about ADS-B itself.  A true wealth of knowledge here.  Your contributions are noticed and appreciated.  Thank you.

cyoung, 2016-01-26:
https://www.kickstarter.com/projects/251459606/flightbox-ads-b-receiver-kit

closed 2016-01-26 by cyoung

skypuppy, 2016-01-26:
Holy crap! 
With over $22k pledged, it still might come to $10k take home or better. 
Man, I am so conflicted over this.  :)

Skypuppy