cyoung/stratux #122: Test new hardware

cyoung, 2015-11-17:
It "looks" like it's the NooElec dongle - is it?

Anyone want to try the kit? Saves a few bucks.

http://amzn.com/B016QP0FXO

bradanlane, 2015-11-17:
I'm already heavy with hardware as I was experimenting with components for the "under $50" package (including a battery).

The bundle from that link doesn't include a battery so it's about a $10-$12 savings from the original shopping list.

rgutlon, 2015-11-17:
Certainly looks like the SDR is private labeled from NooElec. 
http://amzn.com/B017AFPHAK

cyoung, 2015-11-17:
Who knows - if it is then $10-12 is something.

bradanlane, 2015-11-17:
True. A buck is a buck and pilots are notoriously cheap :-) I should know. I am one ! 

Nearly all the savings can be had right here - http://amzn.to/1H5tGVj

They take 2-3 weeks to arrive but they are the same SDRs.

cyoung, 2015-11-17:
Have you been able to do a side-by-side with the NooElec?

I asked Vilros if it's the "same" as the NooElec dongle. They said "No, this is a Vilros brand RTL-SDR, basically the same as NooElec, but manufactured by Vilros."

bradanlane, 2015-11-17:
A "side by side" comparison would mean flying two identical systems which I do not have. 

However, a visual examination of the internals show they have the same boards with the same chips in the same places with the same PCB routes.

cyoung, 2015-11-17:
Fair enough, I'll order up one at some point and try it.

bradanlane, 2015-11-18:
I've got six of them :-)

cyoung, 2015-11-18:
So you have a NooElec and one of the cheap ones?

I just plug them in and do the rtl_sdr | dump978 command at the same time, send the output to diff files for each, and count the number of messages in each.

bradanlane, 2015-11-20:
Like I said, I visually compared these SDRs. I looked closer at all of the SDRs I have and I actually have units from three different vendors. They have different marking on the blue plastic case but the boards inside are identical.

The flying I do does not allow me to do inflight testing. (No safe place to have a Stratux that isn't completely self contained.)

bradanlane, 2015-11-20:
I did perform signal to noise comparisons, using an R820T as my base reference. All of the SDRs I have which are marked at R820T2 have better results that the R820T. I also tested against an E4000 SDR.

depletedcaliber, 2015-11-21:
i have been using this one: http://www.aliexpress.com/item/1pc-Wholesale-Mini-2-USB-TV-Stick-DVB-T-RTL-SDR-Digital-TV-Receiver-USB-RTL2832U/32333687109.html

working pretty well so far.

cyoung, 2015-11-21:
Great, but looking more for quantitative results as different setups can easily be evaluated with side-by-side message rate comparisons.

     rtl_sdr -d x -f 978000000 -s 2083334 -g 48 - | ./dump978 |

depletedcaliber, 2015-11-21:
I can pick up 2 towers from my 1090 antenna setup for piaware in my attic. 
I can only compare with an rtl-sdr blog dongle and the cheap aliexpress one as those are the only distinct models I have.

I can put a splitter(if i can find one) and try both dongles simultaneously and see if i can get some message rate metrics.  I don't have a nooelec tuner though.

depletedcaliber, 2015-11-22:
I ran the two commands, below, in two terminals.  It will count the number of lines printing out the average rate each second, with the label N. 

The rtlsdr-blog dongle is configured to put 5v bias tee on the feed up to a low noise amp, then antenna.
```
rtlsdr-blog     \___<two way splitter>_____________<LNA><ANTENNA>
blue dongle  /
```
```
rtl_sdr -f 978000000 -s 2083334 -g 48 -d 0 - | ./dump978 | pv -la -N RTLSDR-BLOG > /dev/null 
rtl_sdr -f 978000000 -s 2083334 -g 48 -d 1 - | ./dump978 | pv -la -N BLUE  > /dev/null
```
Average for both was about 3.61-3.91 lines per second. Blue seemed to be always head by 0.10 lines. I only really watched it for 5 or so minutes. Not entirely a large sample size.

Not completely scientific, but enough to see the blue $8 dongle was at least the same performance it seemed as the rtlsdr one.

cyoung, 2015-11-23:
Excellent, thank you.

heelix, 2015-12-08:
Not sure if we want to consider this new hardware or not - but I've been getting this to run on the pi zero.  Definitely in the spirit of driving the costs down.  Since it is the older ARM 6 core, rather than the 7 like the newer pi, it does require a few different hoops, however.  Use this thread for testing/progress on the zeros?

bradanlane, 2015-12-08:
@heelix, which powered USB hub did you go with?

heelix, 2015-12-08:
@bradanlane 

This is what I picked up from Amazon.
 
http://www.amazon.com/gp/product/B011K5XOCO

 I have a bunch of cable parts taking the slow boat from China where I should be able to do it for less, but don't have those in hand yet.

bradanlane, 2015-12-08:
We keep filling that "slow boat from China" :-) I'm running Stratux on one of these: 

http://m.aliexpress.com/item/32448079125.html

I haven't messed with an AHRS on it since I'm focused on one of two Stratux configurations - single SDR; and single SDR + GPS.

heelix, 2015-12-08:
I'm attempting to port to one of the $5 zeros, which does take power via micro-usb.

http://www.microcenter.com/product/457746/Raspberry_Pi_Zero

Other than the CPU being like the earlier generation, I can power at least one dongle without assistance.  I've plugged in two radios and seen both light up, but my Bride wanted to watch some TV before I had a chance to see if it was working.  (No power LED on the zeros, so a bit harder to tell if one is having issues or not)

Got a bag of micro-usb headers, etc. showing up in January.  I'm planning to solder up my own splitter/hub.

bradanlane, 2015-12-08:
Just an FYI: you'll need a "hub" chip. You can not just wire USB ports in parallel. (I tried)

heelix, 2015-12-08:
Got that coming too :) but the plan was more for power routing to each of
the USB devices than dealing with the data.  Looking to shim the USB with
the power going to a different power source than through the mainboard.

cyoung, 2015-12-09:
>Use this thread for testing/progress on the zeros?

Please do!

> Since it is the older ARM 6 core, rather than the 7 like the newer pi, it does require a few different hoops, however.

Probably should work fine just recompiling the binary. Maybe decrease replay buffer sizes or cap the max number of connected clients. What have you found so far?

skypuppy, 2015-12-09:
The older A6 core doesn't have hardware floats, IIRC.  But that's been a
while.  So calcs will take considerably longer for sin, cos, etc.

Skypuppy

jtremolo, 2015-12-23:
Any progress on using the RPi Zero?

heelix, 2015-12-23:
Yes.  Got it running, but did not have a chance to get in the air with it
before it was killed off by my crappy soldering skills.  I've not another
one I need to pick up, and will try again.

cyoung, 2016-01-26:
Vilros SDR works about the same as the NooElec and has a better antenna.

Pi Zero works fine @jtremolo. Used an external (powered) USB hub. There seemed to be some USB hub quirks to deal with, didn't try anything other the one that was one hand.

closed 2016-01-26 by cyoung