cyoung/stratux #124: Altitude is incorrect data type

depletedcaliber, 2015-11-21:
traffic.go:68
```go
      Alt uint32
```

Altitude is defined as an unsigned int.   This was pretty easy to spot when i saw planes listed in the traffic tab of the webpage at a some feet below UINT_MAX.  Avare properly showed the traffic at -25. (It was on the ground at Dulles the Airport taxiing)

Altitude should probably be a signed value.

AvSquirrel, 2015-11-21:
The altitude value that gets passed to the Ownship message has to be an offset uint per the GDL-90 spec (p. 20).

>The Altitude field "ddd" contains the pressure altitude (referenced to 29.92 inches Hg), encoded
using 25-foot resolution, offset by 1,000 feet. The 0xFFF value represents that the pressure
altitude is invalid. The minimum altitude that can be represented is -1,000 feet. The maximum
valid altitude is +101,350 feet.

>Examples:
```
    -1,000 feet        0x000
    0 feet        0x028
    +1000 feet        0x050
    +101,350 feet        0xFFE
    Invalid or unavailable        0xFFF 
```

However, the raw value from the pressure sensor is being cast as a uint *before* it's offset. I think that's what's creating the error. If you keep it a float until after the 1000' offset is subtracted, it should prevent the overflow:

```
	if isTempPressValid() {
                altf = float32(mySituation.Pressure_alt)
	} else {
                altf = float32(mySituation.Alt)  //Substitute GPS alt when pressure alt not available.
	}
        alt = uint16((altf + 1000) / 25)

	alt = alt & 0xFFF // Should fit in 12 bits.

```

cyoung, 2015-11-21:
@depletedcaliber - have any replay logs? Are you using UAT only?

skypuppy, 2015-11-21:
Signed?  Not if you want to see anything above 32,767 feet (or YOU are
up there.)  There are so few places in the US that are blow sea level, I
think an unsigned int is preferable.  The problem lies with GPS
algorithms and sometimes not having at least 4 sats in view.

cyoung, 2015-11-21:
It's a valid issue. See 3.5.1.4: http://www.faa.gov/nextgen/programs/adsb/wsa/media/GDL90_Public_ICD_RevA.PDF

>The minimum altitude that can be represented is -1,000 feet.  The maximum valid altitude is +101,350 feet.

skypuppy, 2015-11-21:
Thank you.  That does explain some things.

AvSquirrel, 2015-11-21:
To clarify, my comments were about a similar issue in ry835ai.go.

In traffic.go, there's a similar issue:
```
	//Altitude: OK
	//TODO: 0xFFF "invalid altitude."
	alt := uint16(ti.Alt)
	alt = (alt + 1000) / 25
	alt = alt & 0xFFF // Should fit in 12 bits
```
This can cast a negative pressure altitude as a uint before adding 1000 and dividing by to create the GDL90 format. Would this work?

```
	//Altitude: OK
	//TODO: 0xFFF "invalid altitude."
	alt32 := int32(ti.Alt)
	alt = uint16((alt32 + 1000) / 25)
	alt = alt & 0xFFF // Should fit in 12 bits
```

depletedcaliber, 2015-11-21:
This was UAT only in this instance. 

Doesn't look like i had logging on. I'll have to find/get a splitter so i can use my attic antenna and let it run for a bit with logging to see.

depletedcaliber, 2015-11-27:
The following downlink message causes the row listed in the traffic page to have an altitude near UINTMAX.  

```
-0b2789c637638991d6b201aa80597e100103605e3b32ec0beaa20300000000000000;rs=6;ss=78;
```

cyoung, 2015-12-02:
@depletedcaliber - tested with your example UAT string, looks good. Thanks.

closed 2015-12-02 by cyoung

depletedcaliber, 2015-12-05:
The Dark X X C c f'd x

AvSquirrel, 2015-12-11:
Traffic messages for altitudes above int16_max (i.e. 32775' or higher) are offset high by 36875' due to integer overflow in traffic.go. Change `int16` to `int32` to resolve.

`	alt := int16(math.Max(float64(ti.Alt), float64(-1000.00)))`
https://github.com/cyoung/stratux/blob/master/main/traffic.go#L143

jpoirier, 2015-12-20:
fixed via this pull request https://github.com/cyoung/stratux/pull/154

and an example https://play.golang.org/p/AQ0fpDudvi

commit a8dd42ef (2015-12-02): Fixes #124.