cyoung/stratux #125: USB device that turns on/off another USB device?

cyoung, 2015-11-21:
These radio dongles are taking 100mA even in idle state, would be great to be able to switch them off when there's no activity.

cyoung, 2015-11-22:
I guess just a transistor hooked to a GPIO pin can work: https://www.raspberrypi.org/forums/viewtopic.php?f=44&t=91635

bradanlane, 2015-11-22:
There's mixed reports that the USB ports can be toggled via software ...

https://github.com/codazoda/hub-ctrl.c

cyoung, 2015-11-22:
root@raspberrypi:~# ./hub-ctrl -b 001 -d 002 -p 0

I don't know what it turned off, but the power consumption went down by 50mA. Everything plugged into USB was still working. Can't seem to get it to turn off the dongle.

jpoirier, 2015-11-24:
Given a newer linux kernel, run the following as su:

$ lusb

```
Bus 003 Device 002: ID 8087:8000 Intel Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 003: ID 0bc2:ab24 Seagate RSS LLC 
Bus 002 Device 002: ID 2109:0812 VIA Labs, Inc. VL812 Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 8087:07dc Intel Corp. 
Bus 001 Device 005: ID 0bda:5752 Realtek Semiconductor Corp. 
Bus 001 Device 003: ID 06cb:0af8 Synaptics, Inc. 
Bus 001 Device 013: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
Bus 001 Device 006: ID 04d9:0169 Holtek Semiconductor, Inc. 
Bus 001 Device 004: ID 046d:c016 Logitech, Inc. Optical Wheel Mouse
Bus 001 Device 002: ID 2109:2812 VIA Labs, Inc. VL812 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Realtek DVB-T is at Bus 01 Device 13 

$ lusb -t

```
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/8p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
        |__ Port 4: Dev 3, If 0, Class=Mass Storage, Driver=uas, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/9p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 2: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 3: Dev 6, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 3: Dev 6, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 2: Dev 13, If 0, Class=Vendor Specific Class, Driver=, 480M
    |__ Port 2: Dev 13, If 1, Class=Vendor Specific Class, Driver=, 480M
    |__ Port 3: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 5: Dev 5, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 5: Dev 5, If 1, Class=Video, Driver=uvcvideo, 480M
    |__ Port 6: Dev 7, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 6: Dev 7, If 1, Class=Wireless, Driver=btusb, 12M
```

Follow Bus 01 down to Dev 13 gives port 2. To shut it off:

$ echo '1-2' > /sys/bus/usb/drivers/usb/unbind

Check for the device at Bus 01 Port 2

$ lusb -t

```
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/8p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
        |__ Port 4: Dev 3, If 0, Class=Mass Storage, Driver=uas, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/9p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 2: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 3: Dev 6, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 3: Dev 6, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 3: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 5: Dev 5, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 5: Dev 5, If 1, Class=Video, Driver=uvcvideo, 480M
    |__ Port 6: Dev 7, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 6: Dev 7, If 1, Class=Wireless, Driver=btusb, 12M
```

To re-enable the device

$ echo '1-2' > /sys/bus/usb/drivers/usb/bind

Check that the device is back at Bus 01 Port 2

$ lsusb -t

```
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/8p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
        |__ Port 4: Dev 3, If 0, Class=Mass Storage, Driver=uas, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/9p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 2: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 3: Dev 6, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 3: Dev 6, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 2: Dev 13, If 0, Class=Vendor Specific Class, Driver=, 480M
    |__ Port 2: Dev 13, If 1, Class=Vendor Specific Class, Driver=, 480M
    |__ Port 3: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 5: Dev 5, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 5: Dev 5, If 1, Class=Video, Driver=uvcvideo, 480M
    |__ Port 6: Dev 7, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 6: Dev 7, If 1, Class=Wireless, Driver=btusb, 12M
```

kdknigga, 2015-11-24:
Binding and unbinding a driver to a device requires a newer kernel?  Is
this a USB driver specific thing?  I do this with the ahci driver on rhel 6
(2.6.32) fairly regularly.

jpoirier, 2015-11-24:
Okay, maybe I should have said "it might not work for really old kernels." :/

jpoirier, 2015-11-24:
...or "really, really, old kernels." :)

kdknigga, 2015-11-24:
:)

kdknigga, 2015-11-24:
I really wasn't trying to be a dick; I wouldn't be surprised if individual
driver support is required and support in the USB driver is newish.  I
really was wondering if you knew that to be the case.

jpoirier, 2015-11-24:
Np. I forget sometimes that the 2.6 kernel isn't "newer" anymore.

cyoung, 2015-11-24:
without tuner

     root@raspberrypi:~# lsusb -t
     /:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
         |__ Port 1: Dev 2, If 0, Class=hub, Driver=hub/5p, 480M
             |__ Port 1: Dev 3, If 0, Class=vend., Driver=smsc95xx, 480M
             |__ Port 2: Dev 4, If 0, Class=vend., Driver=rtl8192cu, 480M
     
with tuner
     
     root@raspberrypi:~# lsusb -t
     1-1.5:1.1: No such file or directory
     /:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
         |__ Port 1: Dev 2, If 0, Class=hub, Driver=hub/5p, 480M
             |__ Port 1: Dev 3, If 0, Class=vend., Driver=smsc95xx, 480M
             |__ Port 2: Dev 4, If 0, Class=vend., Driver=rtl8192cu, 480M
             |__ Port 5: Dev 6, If 0, Class=vend., Driver=usbfs, 480M
             |__ Port 5: Dev 6, If 1, Class=vend., Driver=, 480M
     root@raspberrypi:~#

cyoung, 2015-11-24:
root@raspberrypi:~# echo "1-1.5" >/sys/bus/usb/drivers/usb/unbind

Takes it off the USB devices list. Doesn't turn the LED off on the NooElec. Will hook it up to the power supply to see what power savings is.

cyoung, 2015-12-02:
Okay, the NooElec dongle doesn't shut down properly so it uses the same amount of power unbound as it does bound (and idle) - which is about 100-120mA.

Since it uses the RTL2832 for USB, guessing that none of the RTL-SDR that use this chipset can be used in this way to save power.

We know what to do now when power consumption becomes a higher priority.

cyoung, 2016-01-26:
Should be looking at some newer SDRs soon that won't have these issues.

closed 2016-01-26 by cyoung