cyoung/stratux #128: Debugging ahrs

RobertGary1, 2015-11-25:
My ahrs just stopped working. I'm using the RY835AI. It had been working but today the Stratux GPS/ahrs page shows 0 for p alt, pitch, roll and heading. The power light is on on the RY835AI. The wiring looks very secure. Is there anything I can look at to see what the issue could be? Can I snoop the packets to the device?  I'm using Stratux 4r4 but I also tried 5 without change.

skypuppy, 2015-11-25:
Try using the i2c tools.
from the command line:
  apt-get update.
  apt-get install i2ctools.
etc.
if you don't see any numbers at the imu address, the chip might be fried.

RobertGary1, 2015-11-25:
I'm not sure what good addresses are but here is what I get...

pi@raspberrypi ~ $ sudo i2cget -y 1 0X03
Error: Read failed

-Robert

cyoung, 2015-12-02:
0x68

closed 2015-12-03 by cyoung