cyoung/stratux #129: Question on /getSituation

Bagarre, 2015-11-25:
I made a pretty simple EFIS to run on a Raspberry Pi and a 3.5" hat.
https://github.com/Bagarre/Pi2D2
It's been a fun project so far.

It's currently runs on the same Pi that my Stratux is on and I'm naively making ajax calls to  /getSituation about 10 times a second to update the screen. <- Not ideal.

Is there a better way on the Pi to get this data?

RobertGary1, 2015-11-25:
If WingX eventually supports our ahrs I was planning on using an old iPhone 5 running WingX to achieve a similar display. Stratux support multiple wifi clients so the iPhone can be a dedicated backup attitude and an iPad can be a handheld mfd. 

-Robert

bradanlane, 2015-11-25:
I'd need to test the performance impact. Assuming it's not a detriment, the solution is to change from a polling model for getSituation, to a websocket - similar to what is used for status, traffic, and weather.

Bagarre, 2015-11-25:
A websocket would be wonderful but as you mentioned, my concern would be the impact on Stratux.

bradanlane, 2015-11-25:
An easy way to get a "ball park" idea is to start up Stratux, let it get running and stable, and monitor the CPU ("top" command from an ssh terminal) for 30 seconds. 

Then call "getSituation()" 1/sec, 10/sec, and 100/sec - running each test for 30 seconds and see what effect there is on the CPU. 

The AJAX interface has some extra overhead that a websocket avoids so the impact on CPU in the test should be slightly worse than what a websocket interface would generate.

cyoung, 2015-12-02:
>my concern would be the impact on Stratux

Very little. All it does is encode a struct and send it over a socket.

@Bagarre  - Made a branch for you: https://github.com/cyoung/stratux/tree/Bagarre. Change this if you think it's needed: https://github.com/cyoung/stratux/blob/Bagarre/main/managementinterface.go#L105

cyoung, 2015-12-02:
@Bagarre - great work so far, going to play around with it a bit. Closing this issue unless you find some real issue with using the website or even hitting the webserver with 10 requests/sec, which is really nothing in terms of load.

closed 2015-12-02 by cyoung

Bagarre, 2015-12-02:
(y) thanks.