cyoung/stratux #130: Feature request - Port Number

Axtel4, 2015-11-28:
Please consider adding an option to configure the port number from the Wed UI or allow more than one port “out of the box” for different Apps.  I know this was broached in Issue 61 and was subsequently closed, but I think it may be worth looking at it again.

In instances where a Stratux device is deployed with multiple users (i.e. within a flying club), there is an incompatibility with some apps at the default port settings. This requires the user to manually change the config file prior to using Stratux. This could lead to someone inadvertently disabling Stratux for a subsequent user.

For example App A works out of the box with Port 4000, however App B requires Port 43211. User 1 has App A and User 2 has App B. User 1 will be able to use Stratux “out of the box” with the default setting. Whereas, User 2 would need to edit the config file for Port 43211 prior to using Stratux for their flight. This forces user 2 to know how to manipulate Linux and edit the files. In addition, there is a risk that if User 2 does not return Stratux to Port 4000 when completed with their flight and when User 3 comes along to use Stratux with their App C they find that Statux no long works.

skypuppy, 2015-11-29:
Just set software switches, and let them choose which hardware (by name)
that they want to emulate.  They don't have to know anything about
computers to use them.  "Just choose the box on this page," with the
default setup as what everyone else uses.

Easy peasy.

cyoung, 2015-11-29:
Which app accepts GDL90 on port 43211 but not 4000?

Axtel4, 2015-11-30:
DroidEFB. I have only gotten it to work with {"Conn":null,"Ip":"","Port":43211,"Capability":5} or with {"Conn":null,"Ip":"","Port":43211,"Capability":1}. When tried with "Port":4000, Droid EFB shows "ADSB NO TRAFFIC TARGETS" on the chart page and no Targets are displayed. Switching back to "Port":43211 DroidEFB shows "ADSB TRAFFIC DETECTED" with Targets being displayed.

cyoung, 2015-12-02:
Going to call this a DroidEFB bug, then. It's the only app that I know of currently using 43211 and not 4000.

cyoung, 2016-01-10:
@Axtel4 - still nothing back from DroidEFB. Not going to do this because 1) there is only one app that I know of that is using 4000 only, 2) they probably will support 4000 + 43211 at some point in the near future, 3) you can easily fix it with a config change below (and not ever have to change between the two).

The drawback to fixing with the config is that reduces the number of clients you can connect... so maybe 10 clients instead of 20 :)

You can make stratux work on more than one port by just adding more than one `NetworkOutputs` entry. Try this (change the 1090/978 lines for your config).

```JSON

{"UAT_Enabled":true,
"ES_Enabled":true,
"GPS_Enabled":true,
"NetworkOutputs":
[{"Conn":null,"Ip":"","Port":4000,"Capability":5},{"Conn":null,"Ip":"","Port":43211,"Capability":5},{"Conn":null,"Ip":"","Port":49002,"Capability":2}],
"AHRS_Enabled":true,
"DEBUG":false,
"ReplayLog":false,
"PPM":0,
"OwnshipModeS":"F00000",
"WatchList":""}
```

closed 2016-01-10 by cyoung