cyoung/stratux #136: Wanted: GPS pseudo-attitude calculations

bradanlane, 2015-12-08:
Anyone happen to have stumbled across code that does GPS pseudo-attitude calculations? Thus far, all I can find are academic papers on the subject and they can be tough to decipher.

GPS pseudo-attitude has been around for more than 15 years and is how Garmin and other vendors depict a "six pack" on a portable GPS.

I'd like to tinker with implementing a GPS based version of @bagarre Pi2D2.

Nokomis449, 2015-12-08:
Not sure if this will help as it's way over my head, but check out the source code they mention here:
http://diydrones.com/profiles/blogs/705844:BlogPost:16708
It uses AHRS as well as GPS and I think you're looking for an all-GPS solution, but this code is supposedly well documented so maybe it can give some ideas.

AvSquirrel, 2015-12-16:
Thinking about the Garmin six-pack, we're halfway there.

Airspeed: Use GPS ground speed.
Altimeter: Use GPS altitude.
Heading indicator: Use GPS track.

For the rest of the instruments, turn coordinator and VSI are straightforward. Pseudo-attitude would be a function of those two. The challenge, if there is one, is ensuring that the GPS data is fast enough and noise-free enough to have useable inputs. I'll look at writing and testing an implementation this weekend.

**Turn coordinator:** Turn rate is time derivative of heading. GPS heading is stable and we want to react quickly to changes, so I'd calculate the slope of heading vs time over no more than 3-5 points... so two seconds (assuming 1 Hz GPS sampling), or the last second (if sample rate is 5 Hz or higher). 

**VSI:** Vertical speed is time derivative of altitude. GPS altitude (especially non-WAAS) can drift a few meters around the actual altitude, so filtering would be needed, especially since this will be fed into the pseudo pitch calculation. I'd try to use at least 5, preferably 10 points depending on GPS refresh rate. Report only if GS is greater than 30 knots (?) 

**Attitude indicator - pitch:** Approximate as the flight path, so this simplifies to atan(VS/GS) as long as vertical speed and ground speed are converted to the same units. Report only if GS is greater than 30 (?) knots.

**Attitude indicator - roll:** Assuming coordinated flight, turn radius is a function of true airspeed and bank angle. Rate of turn is a function of true airspeed and turn radius. I'd have to write out and substitute the equations / convert units, but this means bank angle should be be atan(c*turn rate * GS), where c is some constant I need to solve for. Report only if GS is greater than 30 (?) knots.

AvSquirrel, 2016-01-12:
Proof of concept (along with a few other enhancements) : https://github.com/AvSquirrel/stratux/commit/bf8e3763e8df81dfbca4ca9f8921575ac0dd858a ([Release notes](https://github.com/AvSquirrel/stratux/releases/tag/v0.6a99))

Hope I don't have any showstopper bugs -- haven't had a chance to test in a moving vehicle, let alone in flight. Not planning to spin an image, but feel free to 'git clone' and give it a whirl.

AvSquirrel, 2016-01-18:
It needs a little work, but this has promise. 

Data below is an 11-second excerpt from a ground-based test (45-55 MPH on a curvy suburban highway). Reception was only fair, with 8 satellites in solution and no WAAS.

The raw output of simple dh/dt calculation (blue diamonds) is pretty bouncy at low speeds -- at 50 knots, it only takes about 0.4° apparent course change in 0.1 seconds to swing indicated roll by ± 10°. However, I've been toying with a couple of simple filtering algorithms that would greatly reduce this noise without introducing lag.

These can be seen as the four line fits through the diamonds. The thin black line is a 1-second moving average, which I'm considering to be a baseline. It has fairly good noise rejection (±2°, vs ±15-20° in the unfiltered data) during the left turn, but lags by about 0.5 seconds.

The gold line is a complementary filter. I'm ruling this one out since it takes a relatively low gain (0.1) to get noise below the level of the moving average filter, and this pushes the lag up to about 1 second behind the heading change.

Two methods of local linear regression both look very promising. These methods involve for each time t_i, find the slope and intercept of roll vs time over the past 1.0 seconds, then calculate filtered roll = m*t_1+b. The orange line takes this data, and applies a moving average filter to the last 0.3 seconds of regression-filtered roll. The green line ("double filtered") instead of applying a moving average filter, runs the local linear regression on the regression-filtered roll.

Both of these methods seem to have about as much noise as the 1-sec moving average filter, but they both converge almost instantly to changes in direction.

[image]

As soon as I implement a least squares method to calculate m and b, I'll roll this into the proof-of-concept code.

bradanlane, 2016-01-18:
Impressive work! I would expect the more filtered result of the green line is still quite valuable and not misleading.

AvSquirrel, 2016-03-07:
I'm close to a pull request on this. Adding locally-weighted linear regression (first iteratation: https://github.com/AvSquirrel/stratux/commit/68f8d5e88e9aa084dbefd8f5bd5ac9e4456b044a) really smoothed out the bouncing I saw on my January prototype. This is an excerpt from a 15-minute drive I took tonight. Even at 30-35 miles per hour, pitch and roll tracked smoothly and instantly to turns, and even tracked hill climbs and descents.

[image]

I've since made three other commits for cleanup, and to add support for SIRF (not tested yet). I also need to flight test this to validate the physics in the air. Latest and greatest here: https://github.com/AvSquirrel/stratux/tree/gpsattitude

bradanlane, 2016-03-07:
What would you like for flight testing? I have a test-setup with GPS (no-AHRS). I'm currently running Avare on a Nexus 7 for testing.

AvSquirrel, 2016-03-08:
I need to verify that the synthetic attitude matches reality. E.g.:
- Climb at Vx and see a 8-12° pitch up
- Descend in the pattern and see 3-4° pitch down
- Roll angle matches what the wings are doing in coordinated turns across a wide range (from less than standard rate, up to 60° bank)
- Output is smooth and responsive, with minimal lag or noise

I'm hoping to fly it at some point this week, but external feedback would be helpful.

I've already quashed one showstopper bug, and new commits submitted this evening to solve a random index out-of-bound panic. Commit a3f0c40b or later should be stable.

AvSquirrel, 2016-03-13:
45-minute flight test performed wonderfully this evening -- standard rate, 30°, 45°, and 60° turns all indicated properly on WingX. Vy climbouts showed 8-10° nose up, and descents in the pattern were 3-5° nose down.

This excerpt shows performance through a pair of clearing turns, then a series of 360° turns. In sequence: 45° bank left, 45° bank right, 55-60° bank left, 55-60° bank right, and standard rate left.

[image]

Current commit is now 2f56c8a to incorporate a 1090 TIS-B bugfix. Update package: [update-stratux-v0.8r1-2f56c8ac73.sh.zip](https://github.com/cyoung/stratux/files/170788/update-stratux-v0.8r1-2f56c8ac73.sh.zip)

AvSquirrel, 2016-03-18:
Pull request #340 submitted at commit AvSquirrel@f7a9479.

This version includes detection of GPS measurement rate and adapting the regression window to fit the sample rate. In other words, it filters / averages over a longer period of time if the GPS is reporting at 1 Hz than if it was reporting at 10 Hz.

skypuppy, 2016-03-18:
I hope these results are not being passed to the EFB's since some of 
them use their own smoothing methods.  Compounding errors and all that.

Skypuppy

AvSquirrel, 2016-03-21:
@skypuppy, the method I used is basically a reverse-looking Lowess filter. *Much* less lag than a moving average filter of the same smoothness. Its responsiveness is similar to the green line in my plot from 1/17, and both ForeFlight and WingX indicate quick roll entries with a couple tenths of a second.

I really should put together a side-by-side video showing how this method, a Stratus 2, a vacuum AI, and the actual outside horizon all compare.

cyoung, 2016-12-01:
@bradanlane - try out http://updates.stratux.me/builds/update-stratux-v1.0r1-ahrs-ebfb671142.sh.

GPS attitude is calculated using @AvSquirrel's implementation automatically when a GPS device is detected. It's looking good.

closed 2018-01-17 by cyoung