cyoung/stratux #139: stratux process fails to start

shayfrendt, 2015-12-10:
I'm experiencing a somewhat difficult to debug scenario wherein the end result is that I *can* connect to the `stratux` wifi network, but *cannot* see a supported device connected in ForeFlight:

[image]

Here are the device components and versions I'm running:

| Component  | Version      |
| :----------| :----------- |
| Raspberry Pi | Model B Revision 2.0 |
| NooElec    | NESDR Mini 2 |
| Edimax     | EW-7811Un    |
| iPad       | A1416        |
| iOS        | 9.1          |
| stratux    | [v0.5b1](https://github.com/cyoung/stratux/releases/tag/v0.5b1)
| ForeFlight | 7.5          |

When I ssh'd into the device, I was able to get a decent ways down the call stack, but not far enough to know what line of code was causing the root issue:

```shellsession
$ ssh pi@192.168.10.1
$ ps aux | grep stratux
pi        2313  0.0  0.3   3548  1596 pts/0    S+   00:06   0:00 grep --color=auto stratux

$ stratux
$ echo $?
0

$ cat /usr/bin/stratux
#!/bin/bash

screen -S stratux -d -m /usr/bin/start_uat

$ /usr/bin/start_uat
/usr/bin/start_uat: line 3:  2305 Illegal instruction     /usr/bin/gen_gdl90

$ /usr/bin/gen_gdl90
Illegal instruction
```

One thing I noticed above is that although the `start_uat` script exits with a non-zero status code, the `stratux` script seems to be swallowing that error and failing silently. Perhaps it's the way the screen session is being invoked?

I dropped into a `gdb` session to see if I could get any more info:

```shellsession
$ gdb /usr/bin/gen_gdl90
(gdb) run
Starting program: /usr/bin/gen_gdl90
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".

Program received signal SIGILL, Illegal instruction.
runtime.check () at /root/go/src/runtime/runtime1.go:146
146	/root/go/src/runtime/runtime1.go: No such file or directory.
```

I'm definitely not a Go engineer: is this indicative of any deeper issue? Or is there perhaps something I'm misconfiguring with my setup? Any help is appreciated!

Oh, and thanks for this awesome project! :zap: :airplane: :metal:

cyoung, 2015-12-10:
Hi Shay, thanks for starting to troubleshoot and including the info that you did in this ticket.

Could you paste the output of "cat /proc/cpuinfo", please?

shayfrendt, 2015-12-10:
No problemo! :bow:

```shellsession
$ cat /proc/cpuinfo
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 2.00
Features	: half thumb fastmult vfp edsp java tls
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2708
Revision	: 000e
Serial		: 000000005c257558
```

I'm using the wrong hardware, aren't I...

I totally missed that there were like tail numbers on frequency:
- `Raspberry Pi Model B Revision 2.0`
- `Raspberry Pi 2 Mobel B`

I stared at the `Pi 2` requirement on the README for like 5 minutes, too. But was like "yeah I totally have an old raspberry pi layin around".

cyoung, 2015-12-10:
There you go :+1: 

You can try "make; make install" in /root/stratux to recompile - the older RPi are ARMv6 procs. Should work alright, but the RPi 2 is only supported now. Recommend getting one of those.

closed 2015-12-10 by cyoung

dalfry, 2015-12-14:
Running the same hardware. R-Pi B+.

Tried a make in /root/stratux, it fails with -

Makefile:8: recipe for target 'all' failed
make: *** [all] Illegal instruction

cyoung, 2015-12-14:
Not too sure what that one is, you might want to hook it up to ethernet and apt-get update / apt-get upgrade. Or just get a new Pi2 ... they are quite a bit faster.