cyoung/stratux #141: Edimax Wifi green power settings

duecedriver, 2015-12-13:
Are we taking advantage of low power settings... is this internal to the hardware or is it software control.. with the unit and iPad separated by a max of 10ft we could hard code the unit to its lowest power output I would imagine to save heat and voltage drop on the pi

Thoughts?

Supports Green WLAN
EW-7811Un has adapted a clever protocol  in smart transmission power control.  The adapter smartly adjust transmission output by distance and CPU offload to help reduce  power  consumption when wireless is idle. With the green WLAN technology, the power consumption can be reduced up to 20%  ~ 50%.  The adapter does not just provide you with the best wireless technology in the world, it also  care for  the environment by using less energy and your pocket by saving up on your electric bills.

cyoung, 2015-12-13:
No, we're turning off the power saving features of the Edimax. As far as I know, this is the same power saving feature that is managed with "rtw_power_mgnt".

If you want to do some testing, change /etc/modprobe.d/8192cu.conf. Currently the settings are:

     options 8192cu rtw_power_mgnt=0 rtw_enusbss=0

More info here:
https://www.raspberrypi.org/forums/viewtopic.php?t=61665

cyoung, 2015-12-22:
WiFi adapter uses less than 10% of total power, could be improved but SDR and RPi CPU cuts are a lower hanging fruit. The adapter is also finicky to any changes, so it would require extensive testing. If you want to give it a try and have new results, please reopen.

closed 2015-12-22 by cyoung

skypuppy, 2015-12-22:
I had an LB-Link (B-Link?) wifi lying around, I forget where it came
from.  I'm using it with zero wifi problems with stratux.