cyoung/stratux #150: Stratux AHRS Status Message

HiltonG, 2015-12-17:
Hi,

The purpose of this document is to propose through collaboration and agreement a status message for the Stratux ADS-B receiver.  This is a draft.  Provide all feedback through this thread.  The intent is to finalize this status message on or before December 31, 2015.  Please contribute to the discussion as soon as possible so that your ideas are heard early enough for discussions.

http://hiltonsoftware.com/stratux

Thanks,

Hilton

duecedriver, 2015-12-17:
@HiltonG 

I have read the .pdf and I believe that we are off to a good start.

I would like to propose, since this is open source, that whatever is agreed upon here remains an open stratux message that will be available for any developer to incorporate stratux into their hardware or software.  And that stratux remains open source.. not tied into a manufacturer.. otherwise I will just go back to getting paid for this.

If this project gains traction it will become more popular and at least one thing in aviation should be above corporate and government greed and that is safety.

that said, while we certainly need the message standard I have a couple questions and comments.

information is great provided its not for informations sake.  IE what are we doing with the information?
How will it be presented (does it need to be presented or just looked at behind the scenes), will it even be presented or will it just cause pop up warnings etc?  And what is appropriate for stratux to process vs just giving raw data and leave the presentation to the EFB software.

TRAFFIC 

Traffic.. while having a messages/sec message might say something.. what is it saying?  I would like to see (no really need) to ensure whatever traffic is displayed on the screen is vetted and current.  I mentioned this somewhere else already but currently traffic shows up in WingX and others that is over a minute old without update and there is no differentiation.. that is no good.  I proposed at least for wingX since you place a circle around it anyway (why yellow) that the color of the circle mean something.  If traffic is within 5 miles / 1 min co-alt it needs to get my attention(flash/blink/do a jig).. if its close and its updates are more than a couple seconds old it needs to scream TARGET LOST.. blink/ change shape (like plane for current, diamond for lost/estimated) or something.  audio could be incorporated as well since people plug in their devices to the audio panel.  if any target is more than a coupe seconds old and updates stop.. just plain changing the color of its ring would be fine.. you could continue to move the target along its previous track estimating position using its speed.. but it needs to be different than a target that has a regular update.  Colored rings and blinking could be used, in wingX when the data changes, change between ID and seconds since updated for targets lost or something.  I cant specifically comment on details since I am on a non disclosure, but do some research on standard TCAS presentation standards and follow it)  Predictive collision would be a nice feature.. i.e. compare your projected flight track and the projected traffic and calculate a time to nearest point of collision.. set a circular error probable and if you predictively get within a mile ID the traffic with a predictive collision and time to NPC.  obviously it only needs to be done for targets that stands a chance of getting anywhere near you so most if not all targets would be thrown out without even getting a calc.

Stratux could possibly do the calcs, but I dont know if the GDL90 spec would allow or has the capability of changing the target status from active to estimated.. so most of this needs to be done in the EFB software... comparing message times since the iPad vs pi timestamp might be way off.

Rule number one in any product is bad information is worse than no information at all. Information overload causes distraction, distraction causes accidents.  My best friend in the U-2 was a single bezel button that would change the display from like 10 overlays back to a basic PFD/HSI presentation so that when shit got weird I could loose all the distraction and get the aircraft back under control.

Weather... again, while ADS-B message/sec is an indication of activity.. how much is garbled.. what is actually happening ... one of you competitors, with a single tap.. provides not just the status of how many towers and atcivity, but a tower radar view, hold old each FIS-B data group was.. i.e. TAF <1min, METAR>5min, Radar > 15 min.. it gives actual times since a valid message updated the weather database... that is useful info.

right now in wingx it take 2 taps just to see if the status page and it only displays indicators that something is attached... 

my biggest complaint in WingX with weather is that when you tap an airport and bring up the wx screen.. again 2 or more tap.. while your competitor you just tap the aiprorts VFR/IFR on the map and it brings up the weather.. you get to the weather.. in big red letters it tells you that the internet weather is 2 days old.. who gives a shit if it was updated 5 minutes ago by FIS-B.. you have to look at each product to see if it as a small ADS-B marker and an even smaller update time.. 

Airport status and other products should not be visible on the map unless they are getting regular updates.. otherwise I could be looking a a green VFR button on my destination and in reality it could have gone IFR but ADS-B didnt update it.. if its not current within 2 cycles of the FAA FIS-B update timeline.. dont display it.. make me drill to the weather page to see its status and that way I see its age...

WORST part is that any updates over ADS-B do not permanantly  change the data in WingX.. if the iPad crashes, I have to restart the program, etc.. it reverts all data back to the internet data which could be days old.. that is really bad...
 
[... cut ...]  of the airspace system without it... keeping safety products from pilots for leverage is reckless and I hope someone gets sued in the future.

I am looking for a developer that want to make something really extraordinary that increases GA safety an order of magnitude.. we are still trying to fit this new tech into the old paradigms when we could shed the old way of doing it and move fully into the digital age.

Hilton, if you want to talk more offline, let me know.. I am getting carpal tunnel!!

skypuppy, 2015-12-17:
@deucedriver  I believe you're preaching to the choir, deuce.  All your
concerns lie in the EFB (tablet app), it seems to me.  To fully address
them, we'd need to create our own flight app (boy, we could make stratux
functionality SOLID then!)  I'm not opposed to that but it is a rather
large effort.  Avare comes to mine but I simply cannot stand their
current user interface. Talk about user unfriendly...
Some of your desires are met in iFlyGPs.  I think they have the 30-day
free trial, too, but not certain.  I know Garmin's does but Garmin tries
to make *everything* sole-source so that only their gear works with
their software, another thing I absolutely hate and will not buy into.

Now, at the risk of being flamed, how practical is it and how far can we
go with some of this data being displayed on the webui from stratux?  In
flight, that means minimizing the aviation app and pulling up the webui
window, then going back to the aviation app; a non-trivial effort during
flight.  (Thank you Google Android and Apple.)

Skypuppy

cyoung, 2015-12-17:
> Now, at the risk of being flamed, how practical is it and how far can we go with some of this data being displayed on the webui from stratux? In flight, that means minimizing the aviation app and [...]

Practical - all of the data that passes through stratux can be displayed however you want. You get text updates and traffic on the UI if that's what you want. The question is if we want to use the webui to manage a weather receiver, or we want to start doing (what one would think of as) EFB functions?

I started work on having the Pi plug into my audio panel to read new METARs and adverse weather conditions to me as they are newly received. There's a lot that can be done with it - question is what is the most useful.

duecedriver, 2015-12-17:
that is what I am talking about.. if in an  EFB settings menu you tell it you have it hooked up to audio.. than when you go to a weather page for an airport, you could push a button and have it read off whatever it is you pushed.. that is thinking out of the box...

skypuppy, 2015-12-17:
@cyoung  How important is the impedance mismatch between the aviation
audio panel and the rest of the world, including audio out from the Pi2?

cyoung, 2015-12-17:
No clue, it wouldn't be an issue with my panel specifically because it has an audio input which hooks up to just about anything. If you mean hooking it up to a mic input, not sure. Doesn't seem like it would be a huge issue to deal with in any case.

duecedriver, 2015-12-17:
@skypuppy 

yeah.. not to preach.. BUT this is the problem I have seen with all these products, HARDWARE and SOFTWARE outside of certified equipment... the development process is .. lets throw shit at the wall to see what sticks.. in a open source hobby environment .. well that is that way it is.. hard to have a coordinated, well planned and thought out attack when you have your developers strewn across the globe, speaking different languages and they have a day job.. it is NOT appropriate for a company that is taking money for a product.

so.. since we have this effort to really think about a standard stratux message stream..

 the idea that we sit down and do some really in-depth thought and use a system integration approach seems prudent..  you have to figure out what you want and how you want it before you can decide on the flow of information.

I think that sticking to a standard is a great idea. like GDL-90 or better GDL-88 (faster sat update I think) since it may make it more compatible once avionics manufacturers begin getting approval to allow portable info into their boxes.. like aspens connected cockpit

the part 23 re-write will help fast track cheaper avionics meant for GA soon too.. or at least that is my hope.

duecedriver, 2015-12-17:
skypup.. impeedance is very important.. we just to hook up audio players to military aircraft intercom since it get boring flying 10hrs across the ocean alone at night.. we made impedance matching patch cables otherwise the signal would be so weak you couldn't here it .  the mil impedance was diff than GA audio panels too.. so we had our avionics techs make them for us...

bradanlane, 2015-12-17:
@Skypuppy - early on there was a lengthy discussion of webUI "features". At that time I recommended we define the scope of the Stratux WebUI as "configuration and operational verification".

This has gone a long way to avoid huge amounts of EFB functionality creeping into Stratux. 

This has also left the door open to other projects implementing EFB-like features. A great example is RPi2D2.

I still believe the driving focus on Stratux is to work with as many EFBs as possible. To this end, the WebUI is there to insure Stratux is configured and functioning and - where necessary - assist with interoperability diagnostics.

duecedriver, 2015-12-17:
cyoung is correct, most audio panels have an aux in which is impedance matched for consumer music sources.. having it hooked up though a passthrough to the pilots headset cord would allow playback to only the pilot of clearances and such and record as well... so you could do a replay if you miss a clearance.. if your panel has vox..only issue is when you key the mike.. if the iPad screams something the controller will get an earful..  we had one aircraft that had bad blead from the intercom to the transmit bus and a pilot was listening to some music that basically had a phrase .. "gotta smack that ass" and he had keyed the mike at just the wrong time and center said it better not be mine that you want to smack.. i just gave you direct.. then a couple airliners chimed in.. I was on freq too and just about pissed my pants it just got that funny...

skypuppy, 2015-12-17:
LOL, wish I could have heard that!

skypuppy, 2015-12-17:
the dreaded **feature creep!**

duecedriver, 2015-12-17:
another good one.. a 4 ship of T-38s.. took off out houston .. had the callsign gaze... houston center gaze 11 flight of 4 passing 3000 direct amarillo.... next thing you hear is .. what.. this is texas boy we dont allow no gaze in amarillo... what, there are 4 gays here... this is american 401.. it isn't us.. bla bla.. it lated more than 2 minutes before center said enough, he had to actually you know.. control traffic.. I had a hard time staying in formation...

skypuppy, 2015-12-17:
pilots do tend to cut up a lot.  :)  Thank God.

duecedriver, 2015-12-17:
no feature creap.. we are designing a box.. that box has info in it.. that info has to get out to the programmers.. they design the feature around what the box can provide.. 

the data stream needs to be compatible (i.e. of some standard), any additional info over the standard must not break the standard. and the the stream must be future proof.. the stream must be open source.. 

thinking about it ahead of time saves programmers from having to re-write their code  from scratch down the line..

duecedriver, 2015-12-17:
a web ui would allow us to use data that a particular EFB program has yet to implement.. IE if you want to know the sat constellation status but your EFB just gives you a green light saying its attached.. switch to the web page to see your estimated error or waas status.. whatever..

duecedriver, 2015-12-17:
2 different things though.. what the data stream needs to output vs what we could drill into via web page.. not everything the box can get at needs to be in the data stream.. 

we also need to decide if it will be a bi directional standard to allow the EFB to config the hardware or just config the hardware via web page

cyoung, 2015-12-17:
EFB config the hardware - there's pause/unpause of a data stream when the EFB knows it will become unable to accept data (going into background or closing). What other config is there?

duecedriver, 2015-12-17:
the other thing I am trying to get across.. is take for example traffic...

if the target is lost.. is stratux still keeping it in the data stream at its last known position.. or does it drop in immediately..

if we are keeping it in the stream.. that it has to have a time tag so that the EFB can see the target has not been updated in xx seconds because it can compare the target last known time to gps message time.. or internal time.. whatever.. but we need to have that standard too.. who has the master time and how is it set to ensure everything is talking the same time.. 

this is basic system design

Bagarre, 2015-12-17:
"we also need to decide if it will be a bi directional standard to allow the EFB to config the hardware or just config the hardware via web page"
Your config web page would (should) be hitting your own api so the EFB developers would be able to hit/display the same data...if they chose to do that.

skypuppy, 2015-12-17:
I think there is already a standard for traffic data and at least ifly
implements it.  Can't remember what the nomenclature for that standard
is, though.

duecedriver, 2015-12-17:
@cyoung.

well ... one radio or 2.. what are the ppm offsets.. enable or disable ahrs, shutdown stratux.., re-cage the ahrs,  etc etc.. 

set system time... especially for internal iPad gps or no gps.. you dont need a gps position to use a chart program.. its a lot more helpful but I didnt have a dot moving across my paper chart and weather and traffic can still be put on a map without knowing where the users aircraft is.. but I think I have made my point...

cyoung, 2015-12-17:
>if the target is lost.. is stratux still keeping it in the data stream at its last known position.. or does it drop in immediately..

Yes, for 60 seconds. https://github.com/cyoung/stratux/blob/master/main/traffic.go#L86-L92

skypuppy, 2015-12-17:
Getting all those flight app vendors to comply is most likely impossible.
Would love to write my own flight app but who has the time or resources?

Maybe spruce up Avare...

AvSquirrel, 2015-12-17:
@HiltonG - GNSS satellites in view would be a useful addition. For each satellite in view, and without bit packing, you'd need 5 bytes. You could squeeze this to 4 bytes by packing 9 bits of azimuth with 7 bits of elevation. 

The format below is the same order as NMEA GxGSV sentences.

* SV NMEA number: 1 byte // satellite ID -- see table
* Elevation angle: 1 byte  // 0-90 deg
* Azimuth: 2 bytes // 0-360 deg
* SNR: 1 byte // 0-99 dB

Mapping of NMEA IDs below is from http://www.catb.org/gpsd/NMEA.txt if you wanted to mark GPS / GLONASS / SBAS differently in a data table or status display. 

```
|====================================================
| 1 - 32    | GPS
| 33 - 54   | Various SBAS systems (EGNOS, WAAS, SDCM, GAGAN, MSAS)
| 55 - 64   | not used (might be assigned to further SBAS systems)
| 65 - 88   | GLONASS
| 89 - 96   | GLONASS (future extensions?)
| 97 - 119  | not used
| 120 - 151 | not used (SBAS PRNs occupy this range)
| 152 - 192 | not used
| 193 - 195 | QZSS
| 196 - 200 | QZSS (future extensions?)
| 201 - 235 | BeiDou
|====================================================
```
```
|====================================================
|System     |Satellite       |PRN            |NMEA-ID
|EGNOS      |AOR-E           |120            |33
|EGNOS      |Artemis         |124            |37
|EGNOS      |IOR-W           |126            |39
|MSAS       |MTSAT-1         |129            |42
|EGNOS      |IOR-E           |131            |44
|WAAS       |AMR             |133            |46
|WAAS       |PanAm           |135            |48
|MSAS       |MTSAT-2         |137            |50
|WAAS       |Anik            |138            |51
|====================================================
```

duecedriver, 2015-12-17:
@HiltonG 

yes.. it would be helpful information if you plan to implement it in the EFB software with a skyview page or you wanted to compare external gps error vs iPad error and either flag the issue or blend the solution.. 

but if the info is not going to be used by developers inside the program.. its just garbage on the bus... and could be displayed by accessing a webpage in the box

duecedriver, 2015-12-17:
you dont get the vendors to comply... you publish what your box does and how it presents it.. they developers either support your box or they loose customers.. its that simple...

duecedriver, 2015-12-17:
if this little box can become capable.. it will canabilze the handheld ADB-B in market to such an extent that at some point it may be greater that 50%.. but lets face facts.. lots of people are having basic hardware issues right now.. we dont really know if the SDRs are getting week or the signal is drifting so badly in flight after getting hot that they are missing things, the GPS is flakey on lots of peoples boxes over gpio.. likely due to noise and assisted startup info that we need to provide to the gps, and the ahrs hardware gives us nothing useful in accelerated flight yet.. 

am I missing something... we still have lots of work to do... and this goes back to my post of probably first baselining the hardware that we can get to work before we can tell with any certainty what data will be available to this new data stream..

the stream needs at its core
GPS position and time
Trafic position and its ability to filter out old / degraded traffic
ahrs.. useable attidude in accelerated flight with roll, pitch, coordinated flight (for a ball) speed, heading and vvi 

everything else is eyewash

skypuppy, 2015-12-17:
Don't forget weather.  My primary drivers for using stratux are weather,
traffic, and WAAS.

Ergonomicmike, 2015-12-17:
@skypuppy "I think there is already a standard for traffic data and at least ifly
implements it. "

Yes. There were some complaints posted in the iFly forum about the colors used for various targets. (Yellow targets hard to see on a Sectional.) IFly responded that they were following a standard.

cyoung, 2015-12-17:
I think we should arrive in this decade at some point and just let the EFBs hit the webserver, grab a JSON structure, and get data from that - easier for everyone, is it not? I can add new stats in a matter of minutes with this and have literally done so when @bradanlane was working on the interface.

Having a discussion about what bits go where and how to save a bit or two on the protocol is kind of silly. GDL90 was designed for 9600 baud serial connections. Our data rate with likely never reach even 50% of what it can be.

duecedriver, 2015-12-17:
my oversight... 

same with the weather products.. get the FULL FIS-B message stream.. I think that it should be the EFB developer that should parse it?  otherwise your have to change your data stream format each time the FAA adds a product which then has to flow to the EFB developers who then have to change their code..

just let the EFB maker take the full FIS-B and decode it.. they update their software regularly and work with the FAA proactively so that it can be implement fast.

Same with TIS... just output the FAA standard message and let the developers parse and display it.. 

if you want to have a way for the web page to access it too that is possible.. but the DATA STREAM that is constantly flowing from the box into the EFB should have the full FIS-B full TIS-B and probably custom ahrs stream unless someone knows of an existing industry standard that is not closed and proprietary.. I am sure there is an ARINC standard... since almost all certified systems get their data over the bus from a remote INU box and not internal to the display..

duecedriver, 2015-12-17:
in the stream along with FULL TIS-B and FIS-B, and our ahrs stream, a stratux box heartbeat, a GPS status indictor for 3d position, status of a WAAS position, a hardware temp/failure bit (just a warning that would tell the user to go to the web page.. i dont think we should add a lot of specfic hardware status in the data stream)

what else.. lets build the list of what the box  MUST provide.. and then let Hilton run with it.. read his .pdf  FIRST .. its pretty comprehensive.. but I do feel that the box should put out the RAW FAA PRODUCTS and let the EFB parse it...

duecedriver, 2015-12-17:
dont forget we are talking the DATA stream.. not what we can get from the box with our own custom interface..

cyoung, 2015-12-17:
All EFBs get the full FIS-B message stream currently. It is currently the EFBs responsibility to parse them.

> otherwise your have to change your data stream format each time the FAA adds a product 

Really isn't that much of an issue unless you go with a protocol that is highly obfuscated.

>which then has to flow to the EFB developers who then have to change their code

Which happens anyways, except both the decoding and the "other" jobs are part of the EFB maintainer now. In which case:

>the development process is .. lets throw shit at the wall to see what sticks

duecedriver, 2015-12-17:
right.. but we cant force the EFB developer to use what is in the stream regardless.. so providing them with the RAW data gives them everything that the box is getting from the FAA without any additional work on the hardware side when things change.. 

the good EFB developers work with the FAA and know when the data stream is getting changed and can code for it.. we cant make the EFB programmers better.. hence the reason most people use foreflight since they have the best product... then I think FlyQ is second.. if I remember the marketing stats right.. 

the better the software the more customers they will have... right now HARDWARE makes a lot of people buy the forflight and garmin products because they are relatively cheap and do a good job.. 

when stratux gets the hardware right and our output stream gives the EFB developers everything they need to compete in a clean well thought out stream and other accessible data that we have in a published standards doc like hilton has on draft here.. they WILL USE IT, I guarantee it..

we are the EGG.. the chicken comes later in this case.. build it and they will come

duecedriver, 2015-12-17:
along with FULL TIS-B are we adding in 1090 targets?  and if so I assume GDL-90 specs that.. and the EFB has to decide if the same target is coming from the ground via TIS or over 1090 direct to the radio
so 2 targets dont show up with a ghosting

Ergonomicmike, 2015-12-17:
All this stuff is over my head. But I once asked thr iFly team in their forum if their app could annunciate when WAAS data was being received from Stratux (as their dedicated box, with its own built in GPS, already does). I didn't get a response. I'm guessing there has to be an agreed upon flag from Stratux so that iFly could announce WAAS. Is this the time & place to decide these kind of things? (I suppose the Stratux UI could report WAAS., if it doesn't already. But as others have said already, the UI is more a diagnostic tool. I typically don't look at it in flight unless I have to.)

cyoung, 2015-12-17:
Yeah, they get the raw data stream in any case.

We should get back on track - the status message.

And yes, TIS-B and 1090 targets are combined by ICAO identifier.

cyoung, 2015-12-17:
The stratux UI does report WAAS - build the latest master branch. @AvSquirrel added it (https://github.com/cyoung/stratux/pull/127).

Ergonomicmike, 2015-12-17:
@cyoung "The stratux UI does report WAAS - build the latest master branch"

Ah, okay. Just found that in the commits. Guess I'll see it in the next beta. Is the WAAS indicator something that iFly could snag if it wanted to, to display a WAAS Annunciator on their EFB app?

duecedriver, 2015-12-17:
that data stream should be minimal containing only the RAW FAA data as it comes in, the GPS position stream, the AHRS stream(roll, pitch,yaw(slip),), and a couple status bits(on/off) for waas status, hardware status (overheat and other warnings to be investigated), heartbeat, 

for ahrs .. speed, heading (mag or track made good i.e.  gps) vvi etc should be in the stream somewhere either with the GPS data .... or wherever the GDL standard puts it..

remember .. it should hold to whatever standard we want to follow.. adding only what is necessary and used constantly.. any little used info like constellation info can be directly accessed on the box via our web tool or the developer can query the box directly.

duecedriver, 2015-12-17:
@HiltonG 

take charge here.. since you have built these data streams before and you have researched and recommended the GDL-90 standard.. we just need to format the current data to these standards and add in any status bits.. again I would not add it a whole bunch of fluf that is not used at least once per second.

the data stream is data used all the time.. specifics can address the hardware directly for specialty data

duecedriver, 2015-12-17:
@HiltonG 

the other thing I have notices is that you app does not run in the background.. and does not recapture the heartbeat reliably .. like 1 out of 10.. 

we also need a very clean implementation of cashing the data in a wait mode to burst it to the unit/software when it wakes.. I know you have some of this coded now.. so its probably already done.. but

cyoung, 2015-12-17:
@duecedriver - buffering of data while the app is asleep is done on the receiver side. I don't think it can be done on the app side, and probably don't want it to. AFAIK, we're the only receiver that does this in such a general manner that it works with all apps (that are on the list, at least).

Re: heartbeat, is this a stratux specific issue with WingX?

And the app "not working" in the background - not sure what that means but almost all apps going into the background will have some functions suspended.

duecedriver, 2015-12-17:
personally I dont think this needs to be a full community pile on.. cyoung and hilton need to take this offline and get on a call together.. 

we are not designing an EFB.. we are not reinventing the wheel.. everything that stratux should be broadcasting into the stream is currently being implemented by certified and non certified developers and the standards exist... 

RAW FAA FIS and TIS, ahrs etc is all spelled out in ARINC and perhaps Garmin standards.. since the G1000 uses off board attitude as well and I am betting they follow pretty close if not exactly to ARINC standards to make certification easy... 

just decide on what additional STATUS bits to include and call it a day.. 

we can add developer hooks into the box for other data later.. this get the 99% of the must have data into the EFB developers hands as a STANDARD that they can start using... later they can add a stratux hardware page to their apps that will have the additional hardware and GPS detailed info..

my humble opinion... of course...

cyoung, 2015-12-17:
Well, parts of this discussion have been useful for working on compatibility - we need to address some things like heartbeats not being registered on WingX.

HiltonG, 2015-12-17:
Hi all,

Wow, did this thread get off track...  OK, let me clarify a few thing.  We are talking about a Status message - the status message does not include traffic data, does not include weather data etc, it is simply to tell the app (for example) if GPS is WAAS or not - the GPS data is already defined in other messages.  Same with traffic - those come in using other well-defined messages - apps already display traffic from Stratux.  Implementing this simple status message will allow EFB apps to display a WAAS flag.  How an EFB developer displays this information is up to them.

So let's not talk about specific apps or the actual data (in this thread) - this proposed message is a Status message; i.e. how many targets, WAAS, is AHRS enabled, is the RY835AI board connected etc.

I'd strongly suggest that other topics such as audio, developing another app etc get moved to a different thread otherwise the on-message conversation about the proposed Status message will be lost.

Yes, this will be open, I don't own this in any way.  The reason I said that I would/should be the only one editing the document (and hence modifying the format) is so that we don't end up with 10 different versions.  Let's all agree on one format, I'll update the format/doc, we all agree, I make it final and a few days later EFB apps will be able to display a WAAS flag, show the Stratux firmware version, what hardware it is running on, etc.

Hilton

duecedriver, 2015-12-17:
again.. dont think specifically about a particular app.. WE/YOU decide what you want the box to output.. its up to the EFB  developers to use it.. not the other way around.. 

in reality you could come up with your own data stream and IF stratux gets popular enough.. developers would be forced to add it to their products or loose business to customers who move on to a product that does..

or you could EMULATE someones elses custom box that is well represented in most EFBs ..but you run the risk of having to change every time they do.. 

keeping it to an open industry standard allows any EFB or system to parse the messages... 

personally, I think any EFB maker that has its own hardware is going to lock you out... regardless of what you do.. until they loose enough sales that they turn back around and have to support it or die...

HiltonG, 2015-12-17:
"personally I dont think this needs to be a full community pile on.. cyoung and hilton need to take this offline and get on a call together.."

While that would be expedient, I didn't want anyone to claim that I was being biased (towards WingX Pro7) in any way, hence the openness.  Now if cyoung wants to take this offline with me - we could probably agree to a format in one day and be done.  Again, I'm trying to coordinate, not dictate.

Hilton

cyoung, 2015-12-17:
Hilton, thanks for the time you've put into this.

What would be your thoughts on (instead of using GDL90), we do something like this:

     $ curl -vvvv http://192.168.1.146/getStatus
     *   Trying 192.168.1.146...
     * Connected to 192.168.1.146 (192.168.1.146) port 80 (#0)
     > GET /getStatus HTTP/1.1
     > Host: 192.168.1.146
     > User-Agent: curl/7.45.0
     > Accept: */*
     > 
     < HTTP/1.1 200 OK
     < Access-Control-Allow-Origin: *
     < Content-Type: application/json
     < Date: Thu, 17 Dec 2015 16:52:14 GMT
     < Content-Length: 275
     < 
     {"Version":"v0.5b1","Devices":0,"Connected_Users":1,"UAT_messages_last_minute":0,"UAT_messages_max":0,"ES_messages_last_minute":0,"ES_messages_max":0,"GPS_satellites_locked":0,"GPS_connected":false,"GPS_solution":"","RY835AI_connected":false,"Uptime":23091,"CPUTemp":28.786}
     * Connection #0 to host 192.168.1.146 left intact
     

(those are the currently implemented fields, as mentioned I can add other ones in a matter of minutes).

The benefits of doing this is 1) quick/easy additions on both ends, 2) two way communications, WingX can pause message flows actively just like FF/Stratus do.

duecedriver, 2015-12-17:
@HiltonG 

I agree. however for this project to really get traction and start moving we need a couple things

1> we nee to investigate and baseline some hardware that is going to WORK and code it.. AHRS I am looking mostly at you.. but GPS also has some hits and misses and the RY does both but in some cases is not doing well for either... gotta have the hardware working

2> once the hardware is working you can decide what info is now available and STANDARDIZE how to get it out.. documentation of what information is available and how to access it will be KEY to get developers to look at it.. they dont have the time to dick around trying to figure out how your box works.. it has to tell them how to easily use that data that your box delivers... 

3> any web based tools or displays to allow users of stratux that use an EFB that does not fully support stratux the ability to get at the box to see gps status, temps, whatever...

we are not talking bias if its open sourced and documented so everyone can use it.. if you pulled all the EFBs into  a room I bet none of them are doing status messages the same way... data they may be very similar on except for those that have their own hardware... 

there is the foreflight box and its closed
you have garmins boxes.. closed and encrypted...

as long as stratux provides reliable data and is open source.. it will be added to EFB software just like ilevel and the other manufacturers

skypuppy, 2015-12-17:
Don't get me wrong: I'm all for feature creep as long as it makes sense.

Skypuppy

duecedriver, 2015-12-17:
@cyoung 

i dont see any reason that the stratux status messages cant be custom.. but if the GDL passes waas status just let the DGL data stream carry it... if you want to have other status messages that a certified or garmin systems doesn't have.. then ping it directly..

the high frequency data stream really should just be GDL formatted ADS-B,GPS, AHRS data and whatever statuses for that data is already supported..

skypuppy, 2015-12-17:
IIRC, iFly shows a * in the GPS box (where they show relative signal
strength/number of sats) when they have WAAS or equivalent.

duecedriver, 2015-12-17:
PASS-THROUGH REPORTS
When the pass-through interface is in use, the GDL 90 will output the received ADS-B messages as-is without further processing. Targets are selected for output based on proximity to the ownship position, and the amount of data interface bandwidth available. No traffic alerting service is provided when this interface is in use. No report extrapolation or coasting is performed.
There are two Pass-Through report messages; one for the Basic UAT message (see Table 13), and one for the Long UAT message (see Table 14).

duecedriver, 2015-12-17:
the GDL status heatbeat message has just about everything near as I can tell

1  message ID heartbeat
2 STATUS BYTE 2
Status Byte 1
Bit 7: GPS Pos Valid Bit 6: Maint Req'd Bit 5: IDENT
Bit 4: Addr Type
Bit 3: GPS Batt Low Bit 2: RATCS
Bit 1: reserved
Bit 0: UAT Initialized

3 Status Byte 2 
Bit 7: Time Stamp
(MS bit)
Bit 6: CSA Requested
Bit 5: CSA Not Available Bit 4: reserved
Bit 3: reserved
Bit 2: reserved
Bit 1: reserved Bit 0: UTC OK

Time Stamp

Message Counts

you could hijack a reserved byte but then you stray from standard

https://www.faa.gov/nextgen/programs/adsb/wsa/media/GDL90_Public_ICD_RevA.PDF

duecedriver, 2015-12-17:
aspen avionics connected cockpit is supposed to be an open format language.. its designed for wireless and panel certified communications and just got FAA approval

Hilton as a developer you could get their tech specs and see if its usable for our needs as well

HiltonG, 2015-12-17:
@duecedriver WingX Pro7 has been compatible with Aspen Avionics system since almost day 1 (it's been a while, I'd need to go back and look at the inner workings).

Regarding status messages, ignoring Stratus and Garmin, all ADS-B receivers use a status message like the one I propose; i.e. small, defined fast, small, easy-to-read bytes (not JSON).  I think there are one or two ADS-B receivers that do not have a status message, but the Levils, Clarity, etc all have a status message specific to their hardware very similar to the one I proposed - that's why I did it like that; i.e. because that is the standard way (and a super efficient way) of passing status data.  We cannot use another manufacturers status message because it wouldn't include a flag to show if the RY835AI is connected (for example), perhaps Stratux will support other hardware in the future.  You'll see that I included a version number on the Status message, this allows us to add additional fields (e.g. sat azimuth) at a later date without breaking currently implementations.

Hilton

duecedriver, 2015-12-17:
if the gps is providing a valid 3d fix.. do I care if its a RY835AI?

results based warning info... 

think of them no as equipment heartbeat, think of them as data validity heartbeat..

valid 3d gps position set a bit high
valid WAAS set its bit high
valid statux heartbeat etc

I dont care what is hooked up.. that status bits in the GDL heartbeat are condition based.. not if its plugged in...

[76 more replies]

closed 2015-12-18 by cyoung