cyoung/stratux #151: tcas standard symbology

duecedriver, 2015-12-17:
￼￼Own-aircraft. Airplane- like symbol, in white or cyan.

￼￼Other Traffic, altitude unknown. Unfilled diamond in white or cyan

Proximate Traffic, 1100 feet above and descending. Filled diamond in white or cyan

Traffic Advisory (TA), 900 feet below and level. Filled yellow/amber circle.

Resolution Advisory (RA), 500 feet below and climbing. Filled red square.

TIS is a bit different on Garmin ..

224XS, 2015-12-18:
At present Stratux provides a data stream to an independent App on a display device. It has no control over how the data is displayed on the screen

duecedriver, 2015-12-18:
yeah.. some folks were asking about it.. just threw is somewhere for them to reference.. 

only gripe with stratux is if we are keeping traffic in the stream that is behind on updates for too long.. without somehow changing the tags or target shape/color.. its too easy for somebody to briefly look down and see traffic in one place but really its not there and a collision hazard..

cyoung, 2015-12-22:
Closing, reopen @duecedriver if you find any actionable / authoritative information on how we should change traffic handling. As far as I know there is no guidance on this, and apps like FF will remove a target from the screen after the newest location update is something like 5 seconds old.

closed 2015-12-22 by cyoung

duecedriver, 2015-12-22:
@cyoung 

so is that 5 seconds after you drop if from your stream where you are holding it for 60 seconds?

if the stratux is going to ghost a target for more than 1 or 2 traffic cycles.. and my understanding of the FAA spec is the longest traffic cycle is like 7 seconds.. than we should decide how to highlight it as a non updated / ghost target.

that is why I think that having traffic time tags that are meaningful is important.. so that software developers know that the traffic is either old or being updated regularly.. and handle it their specific way.

bottom line is that there should not be traffic displayed on my iPad for more that a couple minutes as a definitive target if we have no idea where it is.. its just poor form.. .and bad data is worse than no data...

cyoung, 2015-12-22:
Yes, if you want to suggest to any EFB provider to connect to stratux and retrieve traffic information - please do so.

It has all of the following information:

https://github.com/cyoung/stratux/blob/master/main/traffic.go#L57-L80

Note that it has "last seen".

GDL90 doesn't have a timestamp or age field to communicate this to an EFB.

Unless you find otherwise, I operate on the assumption that GDL90 leaves the decision of what traffic to show and when to show it to the receiver.

FF afaik deletes traffic from the display if no update has been received for ~5 secs. So that means from the last position update on traffic with stratux, it takes 1m5s to stop displaying.

If any EFB wants more granular control of when to display what, they can just use the above-mentioned interface and drop targets whenever they decide they've become stale.

cyoung, 2015-12-22:
Traffic stream (bottom of doc) - just added: https://github.com/cyoung/stratux/blob/master/notes/app-vendor-integration.md

duecedriver, 2015-12-22:
than my suggestion would be to do one of the following

1> change callsign to LOST during that 1 minute ghosting

or 

2> just not ghost traffic of more than a couple seconds just to see if the receiver missed an update… not 1 minute… that way if the developers are leaving traffic on screen using their own algorithms you won’t be adding 1 minute to bad position data..

mooneyguy, 2015-12-22:
With the v0.5b2 version, I cannot see the shutdown or reboot feature in
ForeFlight. Am I supposed to see it? Right now, my master switch shuts off
the power to the unit.

 

Thanks for the great job you guys are doing! Have a happy and safe holiday
season.

 

https://github.com/cyoung/stratux/blob/master/notes/app-vendor-integration.m
d

-
<https://github.com/cyoung/stratux/issues/151#issuecomment-166735321> .
<https://github.com/notifications/beacon/APEqprh0y9wo3q9GX-1ChSIVVMWXCObjks5
pSblhgaJpZM4G3kfM.gif>