cyoung/stratux #153: Altitudes showing wrong in Foreflight in latest head build

braddward, 2015-12-20:
[image]
[image]
Attached are a couple of screenshots showing the issue.  This is from the ground so it's 1090 data.  

I installed earlier production release and it shows up correctly in Foreflight.

AvSquirrel, 2015-12-20:
Noted in #124.

Traffic messages for altitudes above int16_max (i.e. 32775' or higher) are offset high by 36875' due to integer overflow in traffic.go. Change int16 to int32 to resolve.

alt := int16(math.Max(float64(ti.Alt), float64(-1000.00)))
https://github.com/cyoung/stratux/blob/master/main/traffic.go#L143

cyoung, 2015-12-22:
Fixed https://github.com/cyoung/stratux/commit/373c0fa9567792e50f1a22721d8712736b3806ad.

This change was made and effective in v0.5b2, give that one a try @braddward.

braddward, 2015-12-22:
Confirmed that this is resolved.  Thanks!

closed 2015-12-22 by braddward