cyoung/stratux #155: Pre release terminology

duecedriver, 2015-12-21:
I would anotate the new stratux status message as open not for wingx. 

You can list apps that have chosen to implement but you need to push that you now have a message and heartbeat standard published and encourage developers to use it in their code to keep their customers happy.

duecedriver, 2015-12-21:
@cyoung 

Taking it a step further. Now that stratux is in the mainstream, use it as a platform to promote

Open standards for avionics

Proof that inexpensive development can provide safety and bennifit to the aviation community so speed up the approval of the part 23 rewrite that they are in violation of federal law for not yet having approved

Etc

ScraboTower, 2015-12-21:
I assume the new beta is currently incompatible with the current  release of WingX which is 8.6.2.1, until @hiltonG latest update hits the App Store.

HiltonG, 2015-12-21:
The added status msg will just be ignored, so Stratux and WingX Pro7 (and other apps) will work fine together.

Thanks,

Hilton

Ergonomicmike, 2015-12-22:
Building on @duecedriver 's comment, one word: Soundblaster. (Or am I showing my age?)

In the old days, before there were standards for the sound interface for PC's, Soundblaster took it upon itself to develop it's own drivers. They became the de facto standard for a long time.

So I agree with @duecedriver here. I see an opportunity for the Stratux deveopers to take the reigns and set some good standards that will help the industry.

bradanlane, 2015-12-22:
I do not have the time, money, or desire to take on a "defacto standard initiative". I'll let stouter souls take the lead.

Ergonomicmike, 2015-12-22:
@bradanlane Don't think of it that way. This is like creating history. You don't even know you're doing it. Just keep plugging along with the good job you all are doing, and the industry will follow. (Per my updated post above about the major EFB manufacturer soon to recognize Stratux. Who knows, they might even annunciate WAAS on screen, now that you've set a "standard" for it. See? You didn't even know you had set a standard.)

duecedriver, 2015-12-22:
yep.. I agree @Ergonomicmike .. you become a standard simply because you will be the only box maker that publishes the standards you are using and will allow.. actually encourage others to use it too... it takes the commercialism out of it and because there will be thousands that will likely build one.. they either have to accept that fact or loose all those customers that will gravitate to a EFB maker that their box will work with..

garmin unfortunately became a deficacto in GPS but unfortunately.. they never left their handheld origins when they had to build a device while walking and try to take that same 6 button approach to panel mounted software... their GUI and logic suck and from a human factors standpoint is a dinosaur.. thats why I believe avadine will come on strong with 430 530 replacement slide in solutions that have much better logic... 

anyway.. keep up the good work.. keep sucking the collective brain dry to incorporate the best the industry and the contributors here have to offer to make this project the best that it can be.. 

and as I said in the other thread.. if you build it... they will come...

duecedriver, 2015-12-22:
where I would like to see the GA panel go in the future is much like where the automotive industry landed... very few people want nav in their cars these days.. the databases are expensive, the displays get an order of magnitude better by the time you get it paid off.. so everyone uses their phones.. that are connected, do real time traffic, and are updates often.. 

well a car is replaced every couple years while planes.. not so much.. how many people do you know driving a car built in the 60s.. the avionics need to be front ended by external devices.. the box takes care of the processing and signals while the external device is the display and user interface.. which can be updated and improved on a much greater frequency than replacing the avionic stack....

duecedriver, 2015-12-22:
anyway ... back on point... it would be great if the word got out that stratux is not proprietary .. that open standards are used and here is where to find them so you can get it working with your application.. 

not say here is the latest version with a protocol / status message for WingX .. which its not.. it was a standard put together for everyone.. so your notes should reflect that!

cyoung, 2015-12-22:
Well, it's all open - that should be assumed from the start.

It's important to remember that until there's a "stratux EFB", we need to be cognizant of the fact that EFBs control the display of the data that we collect. So it is important to recognize their efforts to improve integration between their apps and stratux. That was really the intention.

That said, open standard - yes. "the" open standard - well, we can't call it that yet because WingX is the only one that has implemented it.

In fact, just today a dev from FltPlanGo contacted me about doing this same thing - getting status messages from stratux. They are putting a lot of effort into their app, which will be free by the way.
So there were two routes, I sent them the PDF sheet that Hilton drew up and info that I wrote up documenting already-existing interfaces that stratux has available to EFBs:
https://github.com/cyoung/stratux/blob/master/notes/app-vendor-integration.md
They chose the HTTP/JSON route, pulling /getStatus to get the same thing.

But you're right, the concept of "proprietary" is so entrenched in pilots' minds that you have to actually advertise the fact that something is "open" - even if that fact is right in front of their face.

I vote duecedriver to be our new marketing director!

Ergonomicmike, 2015-12-22:
I expect that the major EFB manufacturer of which I spoke earier (of whom I expect will offer dedicated Stratux connectivity in thier next version) is already looking at your new status spec to make their product play w/ yours. They might not have contacted you (perhaps for NDA reasons). But that doesn't mean that they're not looking at your newly created 'standard.' As I said before, you're making history - although I don't think anyone (like Steve Jobs?) knows it while they're doing it.

closed 2015-12-22 by cyoung