cyoung/stratux #157: stratux updates

jpoirier, 2015-12-22:
It would be convenient to know when it's okay to update the stratux software via command line (e.g. go get..., mv, reboot, etc...) and when it's recommended to write and use a new image.

cyoung, 2015-12-22:
Unless there is a change with a system conf or spindle scripts, you can update from command line.

The majority of users will need a new image to update, even so.

Guess I could put it in the release notes from now on, fairly certain if I do then it will cause some confusion for at least some people.

From v0.5b1 to v0.5b2, it's safe to update from the command line.

jpoirier, 2015-12-22:
Maybe make a small note of a sys or spindle config change in the merge note
and/or release notes.

duecedriver, 2015-12-23:
On the web tools page you could on the Stratux systems setting page have a current version number and a update available button that could be pressed to auto update...

duecedriver, 2015-12-23:
I would still like to see 2 different formats for the webpage.. 

one for the iPhone or using the pullout feature.. IE the current compact multi tab

another that is more a all-in-one page for a larger iPad so that in flight you dont have to do lots of tabbing.. just fast app switch from efb to safari which is already on the stratux webpage in the background and everything important is right there.. stats, shutdown, reboot, critical options.. etc...

bradanlane, 2015-12-23:
@DeuceDriver- interesting suggestion for devices with tablet sized screens. You should go ahead and implement it and submit a pull request when you feel it's ready. I'd be happy to help with testing it on my Android tablet.

duecedriver, 2015-12-23:
I would if I were a coder still… I have not coded since the early 90’s and none of the languages are used any more.. my last code was ADA and I helped and I rewrote DOD-STD-2167A software devel milspecs… I left software development as the officer in charge of Independent validation and verification quality assurance sections on a 100 million dollar development project.

but then I became a pilot and finished 16 of my 21 in the cockpit.. I assisted in the development and training for 2 glass cockpit conversions, the kc-135 and U-2 as more of a systems analyst and human factors expert, instructor pilot.  I was also a graduate of the USAF School of Aerospace Medicine as the top graduate in my physiologist class.  I was the usaf expert on high altitude physiology as the operations officer of the west coast physiology program.

I know how how it should look and how it should work.. not how to code it…..

ssokol, 2015-12-23:
@bradanlane - any idea how much effort would be required on the web app in
Stratux to support file uploads? Ideally we will want to get to the point
where somebody can download a zip or tar update file then upload it to the
Stratux from the web UI.

bradanlane, 2015-12-23:
The challenge with a file upload to Stratux is the overall workflow. Off the top of my head it looks something like this. 

1. mobile device is connected to Internet
2. mobile device determines a new file is available on the Internet
3. mobile device downloads file
4. where does mobile device store file ? (iOS doesn't let a file be stored to the file system)
5. iOS devices need a 3rd party storage solution with installed App (eg Dropbox)
6. how does the Stratux WebUI get access to the local file stored in the 3rd party App?
7. WebUI somehow has access to the file data and stream uploads using a newly developed AJAX interface on Stratux. 
8. Stratux code needs to store file in a known location
9. Stratux does a specialized application 'stop' which will automatically be followed by a script execution
10. script executes to install update and must be 100% sure it installed or be able to rollback.
11. Script ends by restarting Stratux

Figure out steps 4 thru 7 and we might have a chance.

Alternately, Stratux needs to be able to 

1. Stratux application 'stop';
2. toggle from AP mode to client mode; 
3. connect focus Internet; 
4. check for update; 
5. download, install, and potentially rollback update; 
6. toggle back to AP mode; 
7. restart

RobertGary1, 2015-12-23:
Pulling out the SD card and downloading a new image on it is how Garmin works. I think most pilots will find that the easiest. 

-Robert

cyoung, 2015-12-23:
Well, we should start figuring out some update mechanism with pre-programmed kits hitting Amazon now. Even if it's not perfect, perhaps a modem/router firmware style update - download the update file to laptop/desktop, switch wireless networks, upload to web interface. That doesn't seem to bad to me as a first step.

bradanlane, 2015-12-23:
If the "FAT" partition is big enough, then it would be possible for the user to copy an "update file" to that space. Then on boot, Stratux sees if sn update is there, installs it, and removes (or marks) the update file.

Ergonomicmike, 2015-12-23:
@duecedriver If I have it right, the Web interface is all HTML and CSS. Especially CSS. I leraned FORTRAN in college and was never very good at it. But I taught myself CSS in a week. Might be JavaScript in there too (to check for screen size). JS never made sense to me. But you can probably figure it out. One trick is to find an Interface you like and look at the source.

cyoung, 2015-12-23:
95% of the image isn't changed from update to update - the updates would likely be under 20MB. There is about 1GB of built-in empty space which each device is "guaranteed" to have. Eventually we'd probably want to change the startup script to automatically expand the image to give more extra space.

ssokol, 2015-12-23:
cyoung's "router firmware" example is probably the easiest thing to do:

From your home laptop / desktop:
1. Download the update files
2. Fire up your Stratux
3. Connect to the Stratux from the laptop / desktop
4. Use the upload function to push the update to Stratux
5. Stratux (gen_gdl90) fires off a script that verifies, unpacks, and
installs the update

If you want to try to do it from the web UI on a mobile device, you could
try something for 4 - 7 like:

4. iOS and Android store the update into LocalStorage or IndexedDB.

http://caniuse.com/#feat=namevalue-storage
https://hacks.mozilla.org/2012/02/saving-images-and-files-in-localstorage/

http://caniuse.com/#feat=indexeddb
https://hacks.mozilla.org/2012/02/storing-images-and-files-in-indexeddb/

5. I don't think the web app could get at data stored in DropBox, etc.
Things are pretty well partitioned in Apple-land.

6. Not necessary if the data is simply a blob in LocalStorage or IndexedDB.

7. Pull the data from the storage and post it to Stratux.

Honestly, it might be easier to build native or pseudo-native (PhoneGap)
apps just to do updates.

bradanlane, 2015-12-23:
"localStorage" on mobile has a default limit of 5MB which can be increased by user permission to 10MB. This may not always be enough.

Loading from the desktop will require the user have additional computer skills. This may or may not be an issue for the target audience.

If the FAT partition on the SD card were big enough, this would be the most basic solution for the users as it requires no additional skills above what they likely already have regardless of their computer (Windows or OS-X).

jpoirier, 2015-12-23:
btw - vendoring has been turned on by default as of Go v1.6 and, imho, it
should be in place prior to using any update process/procedure.

ssokol, 2015-12-23:
I may be in the minority here, but I really don't like having to remove the
uSD card from the Pi. To do so requires that the case be engineered to
allow for easy access to the card. Those cards can and do vibrate loose if
there isn't something to keep it from backing out of the sleeve. This makes
the case design more complicated. You have to have the uSD adapter. You
have to be able to figure out which partition to drop the update file onto.

If you have to take out the card, you might as well reimage it.

I think a download-then-upload process is well within the technical skill
range of anyone who's capable of using Stratux.

My $0.02. Feel free to disagree.

-S

skypuppy, 2015-12-23:
what's vendoring?  Installing "company info"?

jpoirier, 2015-12-23:
Using a specific tagged/committed/versioned package and/or library. 

by default, the go tool looks for packages rooted in GOPATH, or relative paths, but with go 1.6 it'll now look for a vendor folder within the package that's being built for fully qualified imports first.

There's a bit more to it but that's the gist. 

Why is it important? E.g. Right now stratux pulls gortlsdr directly from my master branch but I make no guarantee that master works (but i do make every attempt to keep master clean and only push to the development branch) or is tested whereas the tagged releases I do, etc...

skypuppy, 2015-12-23:
That hurt my head.

Skypuppy

egid, 2015-12-24:
@duecedriver off topic, but do you know Doc Jensen? Ex USAF flight surgeon, runs Aeromedical / Human Factors courses at UND for the last decade or so.

duecedriver, 2015-12-24:
yep.. although we only briefly crossed paths.. I did send him a space suit once for a presentation and he sent me a UND hat… have it still…

cyoung, 2015-12-28:
v0.5b3 can be updated with a pull and make.

Axtel4, 2015-12-28:
Already pulled and made.  Looks good so far.

cyoung, 2016-01-10:
Anyone know of a package builder / deployment system that would work well for updates in our case? At some point in the very near future we will go with a base image and just start sending a packaged update that only includes the changed files.

duecedriver, 2016-01-10:
it would be cool if we could put it into client mode and let it download updates from the web when on a wireless network…

bradanlane, 2016-01-10:
Possibly the simplest and least complicated would be to detect a USB thumb drive with a specific file. This is the same as what most commercial EFIS systems are doing.

1-prop-head, 2016-01-13:
Only obliquely related to this but really not worth making a new issue. When I did the make for stratux in image v0.5b5 it failed because of some necessary updates of go().  error was (error server certificate verification failed)
This was easily worked around by adding the line "GIT_SSL_NO_VERIFY=1" to the bottom of the .bashrc file in /root

ssokol, 2016-01-13:
I think we should assume that the device will not be able to connect to the
Internet. I also think that requiring the end user to open the case and
remove the SD card will lead to Bad Things (tm) happening. For
consistency's sake, it would probably be best to make .deb files of
gen_gdl90, the Stratux web interface, the customized hostapd, and any other
packages we use. Then...

Developer side:

1) Build .deb file(s) for packages to update.
2) Create an update.sh file that installs the .deb files using dpkg and
does any other housekeeping necessary.
3) Zip or tar the .debs and the update.sh file into something like
"stratux_update_0.6b2.tar.gz" (i.e. to keep it separate from a complete
system image of 0.6b2)
4) Post that as an update on Github or elsewhere

End user side:

1) Download the update file onto your computer.
2) Power up the Stratux
3) Connect your computer to the Stratux wifi network.
4) Point your browser to 192.168.10.1 (or, stratux.local if you install
avahi ??).
5) Go to the Update tab.
6) Drag / drop the update file onto the update tab.

Stratux system side:

1) After the file uploads, the server-side web app unzips the uploaded file.
2) It then runs the update.sh script
3) When it's done, it deletes the update file and reboots the Pi

The only real concern is what happens if you have dependent updates - i.e.
your update to v1.9.9 requires that you have v1.9.8 already installed? It
might be easier to require all builds include the latest / greatest version
of all Stratux-specific packages.

RobertGary1, 2016-01-13:
I guess there are lots of use cases. I would rather not have to pull the Stratux out of the plane where it's zip tied and all. It's much easier to do it Garmin style and just walk out with a fresh SD card. 

-Robert

1-prop-head, 2016-01-13:
As a retired Unix developer I might ask:
What about using the extra USB slot to attach a SD card reader and when it is inserted the (ex vFat) file system is mounted? 
A gross oversimplification follows because this stuff is not simple or easy.
At each reboot you look for that USB file system and if you find it you look for some special file (stratux.zip) which you open and if it is a real update file (check quid,version,etc) and not already applied use the contents to freshen the system.  You could also put things on that drive pre update that might be useful post update depending upon success or failure.

ssokol, 2016-01-13:
@RobertGary1 - that's a fair point, but the Pi isn't as friendly for that
kind of update as the Garmin and King units. They have their operating
systems on another non-volatile storage device (EEPROM?) and the data card
is used for updates and data. Those systems are also mechanically designed
to make it safe and easy to insert / remove the data card. The Pi uses the
SD card as its boot / root, which means it CAN'T be removed while the
system is running. Sot the cases I'm designing for the Pi are intended to
keep the SD card as securely in place as possible. This makes it quite a
pain to get the card out.

@1-prop-head - also a good idea, but many of us are using all four USB
ports: UAT SDR, 1090-ES SDR, GPS, Wifi.

Ultimately, I think we want apps for iOS / Android that can download the
update over Wifi (at home) or LTE (anywhere) and then push the update onto
the Pi - no computer needed.

egid, 2016-01-13:
>Ultimately, I think we want apps for iOS / Android that can download the
update over Wifi (at home) or LTE (anywhere) and then push the update onto
the Pi - no computer needed.

Yeah, this feels like the right approach. It's not that different from how Appareo set up the Stratus units, although I think they get the new firmware right out of ForeFlight.

bradanlane, 2016-01-13:
We were not suggesting pulling the active SD card. As mentioned a few days back, the suggestion is inserting a USB storage device. A process on the RPi, detects the USB and looks for an update file. It performs the update and restarts the Stratux software.

skypuppy, 2016-01-14:
Wouldn't that require static libraries and bloat the resultant code
size?  Nah, all the items that stratux currently uses, it needs all the
time anyway, which should result in a smaller overall fingerprint in
active memory.  Which are you guys compiling with now?  (I haven't
looked that deeply into the current memory/"disk"  footprint.)

skypuppy, 2016-01-14:
Or USB drive, maybe. 
It would be easiest all around if the RPi2 just downloaded the upgrade
via the Internet, -- except for those that leave it in their plane all
the time.  <sigh>

For non-tech pilots (are there even any of those???) just downloading a
full image to a USB drive and dd'ing it to the [forever in-place] SD
card has it's own consideration -- what if the guy has and needs a
larger SD than 4 Gb?  When the RPi2 boots up with no way to ask whether
to grow the filesystem, something will break.  It is conceivable to
include boot software that searches the SD card size and grows it all on
it's own, then reboots.  Kind of a pain, though.

Personally, I like having the RPi2 get on the web and get an update,
which requires removal of the RPi2 board to a networked area, and the
pilot would have to have a power supply independent of ship's supply.

For any automatic updates, might be prudent to only allow *stable*
images/software.  If the pilot wants any beta versions, he will have to
know what he's doing anyway.

Skypuppy
<who is probably rambling today>

cyoung, 2016-01-14:
I'll just make a deb packager for releases and then we can do whatever we want with it - auto detect it on a USB or upload/download and install it.

cyoung, 2016-02-15:
https://github.com/cyoung/stratux/commit/f8b2ebd519ffe999e9dab7e181664870ec61b3e2

cyoung, 2016-02-15:
@AvSquirrel @Ergonomicmike and others that are beta testing a lot lately, can upgrade to the equivalent of the latest beta release (assuming you have v0.6r1):

`service stratux stop ; wget -O - https://github.com/cyoung/stratux/releases/download/v0.7b1/update-v0.7b1-02bf2251d8.sh | sh ; reboot`

Then after this by using the web interface.

closed 2016-02-15 by cyoung

bradanlane, 2016-02-16:
De we want to move the webUI to a separate http service so the updates can be driven directly from the web interface going forward?

cyoung, 2016-02-16:
I've added a simple upload field on the Settings page and a handler on the HTTP listener that does the update.

https://github.com/cyoung/stratux/blob/master/main/managementinterface.go#L275-L295
https://github.com/cyoung/stratux/blob/master/web/plates/settings.html#L108-L120

If we want to break off HTTP serving functions from gen_gdl90 then it might be a good idea as it facilitates web dev and addition of other projects to one's Stratux - like Pi2D2. But it's not a priority at the moment (for me) as it adds extra services on the device to maintain and there isn't any immediate need calling for it.