cyoung/stratux #158: First boot - randomize WiFi SSID?

cyoung, 2015-12-22:
https://www.reddit.com/r/stratux/comments/3xwbic/stratux_wifi/

Anyone see any downsides to it?

duecedriver, 2015-12-23:
more to the point what is the upside?

if I want to set my iPad to only connect to known networks.. that will be broken

apps won't be able to determine its stratux that is connected by ssid like its published in your docs. (that can be fixed)

if you are doing if for security I would much rather see it locked up with a trusted certificate or a just plain old WPA-2 but again.. if you change the ssid I would have to enter credentials each time we fire up the box or a reboot, power interruption, etc

again... what do we gain...?

cyoung, 2015-12-23:
>if I want to set my iPad to only connect to known networks.. that will be broken

Good point.

Probably will just base it off of the MAC address of the wifi, which means for any given unit it would be the same (even with future software updates) but between any two units they'll likely have different SSIDs.

duecedriver, 2015-12-23:
or call it stratuxx with the last 2 of of the wifi mac address of the dongle..

ScraboTower, 2015-12-23:
I change it to my the tail number just because I CAN !!!!

cyoung, 2015-12-23:
Isn't it ScraboStratux?

Ergonomicmike, 2015-12-23:
I like the idea of StratuxXX, because the typical user will be looking for sometthing with the name "stratux" in it. (Altho the typical user will not know his MAC a priori. Would have to learn it on first boot without other Stratux's nearby and make a note of it.) But suggest using  the last 3 digits from the WiFi's MAC, just to decrease the odds of SSID collisions. (Even tho having more than one Stratux near by is an outlier scenario.) And perhaps Stratux_XXX just to make it pretty. (Underscore or period.)

JohnOCFII, 2015-12-23:
The Appareo Stratus uses the format: StratusXXXXXX where X = some string of numbers.  I forget if it is the serial number, or the MAC, but I would suggest using the last 3 octets of the MAC., So for example, Stratux8afabe.

Personally, I'd capitalize Stratux (which I do with my SSID, because my Mom was an English major, and it just feels like a proper name).  :)

I'm in a flying club of 60 members.  We've probably got about 12 Appareo Stratus units, and at least 5 Stratux units already.  We in the club are already using individualized SSIDs, but I think this is a feature that should be in the software as released.

John

skypuppy, 2015-12-23:
Show the last 3 in the WebUI.  Of course, that doesn't help if they
can't connect in the first place.  So, keep the default and if conflicts
ever do arise, they can look on their WebUI, see what the numbers are,
then select a slider to automagically install the 3 as part of the
stratux id.
Maybe.

Skyuppy

RobertGary1, 2015-12-23:
If the issue is that some people have several than it seems like the solution is to optionally allow some to configure the ssid in the web UI. Randomizing it doesn't seem beneficial. Putting security on it may but it's not something I'm worried about as I power up before taxi. 

-Robert

skypuppy, 2015-12-23:
You have to take the guesswork out of it, though, for those that just
turn 'em on and turn 'em off.

duecedriver, 2015-12-23:
Whatever we do and whatever number of octet groups you decided to add... these conditions are a must.

In must not change so that it can auto connect upon a power interruption or restart and remain a known network  keeping it the same allows printing the StratuxXXXXXX whatever on the outside of the case for those robber barons here that have a fleet of them operating...

It must not interfere with the current simplicity

It must, of course, be published in your Stratux standards documents so that developers know what to expect.

Now might be a good time to put a developers section in the WIKI with all the docs and notes they need on the details of how to get the beeps and squeeks out of the box .. they dont care how its deleloped.. just how to use it...

JohnOCFII, 2015-12-23:
If you use part of the MAC address, then it won’t be random, and the WebUI doesn’t need to be involved.  Sure, the first time you use Stratux, you’ll have to select the network (unless you have auto-join selected — which I don’t), but that is no different than today.  I’d suggest not re-inventing the wheel.  The way Stratus does it (and my WiFi Thermostat, and my WiFi printer, etc.) works just fine.

John O'Shaughnessy
osh@cloud7.org

jtremolo, 2015-12-23:
I agree with John,  I think the best approach is to have the format be Stratux_XXXXXX.  The X's are the last 2 or 3 octets of the mac address.  This will eliminate conflicts between devices, be constant for a device allowing auto-reconnect on wifi dropout and easily recognized by the user when a new device is powered up.

duecedriver, 2015-12-23:
there is no harm having the stratux box assess as its name in the internal web page as a confirmation you are changing the right box… that way if your iPad is configured to connect automatically to all known networks you dont end up changing the wrong one…

but whatever we use, it needs to remain static once set, until the wifi adapter changes of course..

jpoirier, 2015-12-23:
Given the short effective range of the wifi module I find it difficult to
believe there's any real (airborne) issue at the moment, unless of course
people are flying in tight formations. :/

Adding four or so octets from the mac id would be a simple enough
precaution that should be straight forward to implement. Hopefully it'll be
enough to quell the fud.

jpoirier, 2015-12-23:
On Wed, Dec 23, 2015 at 11:59 AM, Joseph Poirier <jdpoirier@gmail.com>
wrote:

> Given the short effective range of the wifi module I find it difficult to
> believe there's any real (airborne) issue at the moment, unless of course
> people are flying in tight formations. :/
>
> Adding [...]

...and with that said, I wouldn't change it yet.

Nokomis449, 2015-12-23:
"I'm just a simple caveman; your world frightens me."  Adding octets seems awfully impersonal for a system I built myself out of Amazon parts and some popsicle sticks.
Since the Web interface can already set the PPM and weather stations, I would prefer the Stratux image to default to "stratux" and for the minority of users who have a conflict or want to personalize it, allow it to be changed on the Web page with the note that a reboot is required.  Some linux-savvy people are going to change it anyway; why not give everyone that option?

JohnOCFII, 2015-12-23:
> Some linux-savvy people are going to change it anyway; why not give everyone that option?

Adding the MAC address fragment to the word Stratux doesn't preclude any savvy person from modifying the SSID in the same way they can do it now.  In our flying club we are already seeing the conflict situation, with 60 members and 6 aircraft in two adjacent hangars.  Next month I'll be including some basic Stratux tips in a winter seminar, so we'll only have more conflict sooner rather than later, at least in my neck of the woods.

Nokomis449, 2015-12-23:
We're on the same side of the fence here and my emphasis was on the 2nd half of that sentence.  I feel that a homemade ADS-B should have the option of a personalized SSID rather than an unpronouncable one, and most users would need an easy way to make that happen.  So even if it starts out as an octet-enhanced SSID, I think most users would appreciate a way to personalize it.

skypuppy, 2015-12-23:
For only $49999.95, ship them all to me and I'll fix them for ya.

Skypuppy
<tongue in cheek>

Nokomis449, 2015-12-24:
You wouldnt be the first!  My SkyRadar DX has an SSID of "Skyradar26".  I contacted SkyRadar and asked if it could be changed, and they responded Sure!  Just send it to us postage paid with a check for some ungodly amount and they would change it for me.  Uh, no thanks.  It's still Skyradar26.

ScraboTower, 2015-12-24:
>> You wouldnt be the first! My SkyRadar DX has an SSID of "Skyradar26".

I have the link to download their program that allows you to change the SSID for SkyRadar, let me know if you are interested

Nokomis449, 2015-12-24:
Yes please!

ScraboTower, 2015-12-24:
You can try to do it with SSIDSwitcher utility: www.skyradar.net/sofrware/SSIDSwitcher.zip

You will need windows PC and USB cable connected to SkyRadar.

You do need to be very quick when changing it as the program will refresh the SSID from the unit if you are not quick !!!

ScraboTower, 2015-12-24:
@Nokomis449 - did it work for you ?

Nokomis449, 2015-12-24:
Haven't tried it yet - the SkyRadar is in the Mooney so I'll have to wait.  But I believe that's the same software I tried before.  I just reviewed my email exchanges with SkyRadar and need to correct myself.  SkyRadar wanted $120 to upgrade my old SkyRadar L model from an ad-hoc signal to an infrastructure signal.  At that time the L model was already discontinued, so I thought the price was too high.  Later, I purchased the DX model and asked if the SSID could be changed.  Alexey Zaparovanny's response was that it could be changed in their lab, because "it requires low level reprogramming of wifi module".  They did not give a price for that, and I did not persue it further.  I'm pretty sure I tried that software with the DX, but I'll  try again when I bring home the SkyRadar.

heelix, 2015-12-25:
I'd prefer the initial default to remain consistent.  I was going to try and get the android wrapper to automagically switch to the stratus network, then connect to the web view.

I can see value in having the end user being able to change it without modifying code.  I've got a couple on my desk that I had to tweek to know which one I was connecting to.  Be nice if that were an option, however, and not the default.

skypuppy, 2015-12-25:
When you have two separate computing devices connected only with a
wireless signal, it is best to have a default that works for most
situations.  The ability needs to exist to change a (very) few settings
for those that don't meet the 'standard' situation.

What's great about an open-source like stratux is that you can have it
both ways!  Much better than a product like, say, #armin and some
others, where things only work with the same brand -- and even then, not
all versions of that brand.  Lord, how I **despise** that.  (That's why
I turned away from Apple back when Scully took over.  Ok, rant over.)

Trouble is, one must code the UI so that people who don't know the
technology can just turn it on and it works, while at the same time, the
ability to fix things, for those that can.  And especially when there is
no "service group" ready to charge you a fortune for a 2 minute fix like
some (all?) brands.  This is why I fret over the non-graceful
power-downs.  It doesn't corrupt the filesystem every time, but you'll
never know when it does.  The corruption could be subtle or could be
catastrophic.  Guaranteed it will cause problems in the long run.  Could
even turn into a safety of flight issue, most obvious in the traffic regime.

Skypuppy

cyoung, 2015-12-28:
Scrapping this idea and we'll either go with a 1) first boot setup page, 2) "WiFi login" pop for first device to connect, or 3) some app or external program that does this.

closed 2015-12-28 by cyoung