cyoung/stratux #160: Latest build (0.5r2)

Axtel4, 2015-12-23:
I haven't been able to figure out what I may be doing wrong.  I pulled the latest version of Stratux through commit 5423190.  When running the Make command I go the following error:

pi@raspberrypi ~/stratux $ make
Makefile:6: *** GOROOT is not set!.  Stop.

After some experimenting, I commented out line 6 in the Makefile. . . 

 # $(if $(GOROOT),,$(error GOROOT is not set!))

. . .and was able to successfully compile the latest build.

The build appears to be successful.

What am I doing wrong to get the error? 

Also, here is my go environment:
pi@raspberrypi ~ $ go env $GOROOT
GOARCH="arm"
GOBIN=""
GOEXE=""
GOHOSTARCH="arm"
GOHOSTOS="linux"
GOOS="linux"
GOPATH=""
GORACE=""
GOROOT="/home/pi/go"
GOTOOLDIR="/home/pi/go/pkg/tool/linux_arm"
GO15VENDOREXPERIMENT=""
CC="gcc"
GOGCCFLAGS="-fPIC -marm -pthread -fmessage-length=0"
CXX="g++"
CGO_ENABLED="1"

Thanks.

cyoung, 2015-12-23:
Try doing it as root, "sudo su -"

There's something in /root/.bashrc you can copy to the "pi" user bashrc

Just a change in the build process for circleci

Axtel4, 2015-12-23:
No joy. Using the sudo command I get: 
/bin/sh: 1: go: not found
Makefile:7: recipe for target 'all' failed
make: *** [all] Error 127

I also only have one .bashrc and it is in /home/pi

cyoung, 2015-12-23:
Were you compiling as root before? All of the Go binaries are in /root/go, wonder how it worked before?
In either case, just get root privileges and 

     export PATH=/root/go/bin:${PATH} 
     export GOROOT=/root/go
     export GOPATH=/root/go_path

That should do it.

Axtel4, 2015-12-24:
To be honest, I don't know for sure. I log into Stratux using the default log in credentials (pi / raspberry)and go from there.  I am able to use the sudo and sudo su commands so I am assuming I have Root privileges.

It has been months since I set it up, but I installed GO on the PI per instructions from here; https://www.reddit.com/r/stratux/comments/3ko8zs/build_instructions_for_compiling_stratux_from/ and here; https://github.com/ssokol/stratux/wiki/Compiling-Stratux-on-the-Raspberry-Pi-2.  However, I could have had it messed up from the beginning.

Here are my paths as set in .profile:
export PATH=$PATH:$HOME/go:$HOME/go/bin:/usr/local/go/bin

My GOROOT directory is located here:
pi@raspberrypi / $ cd $GOROOT
pi@raspberrypi ~/go $ pwd
/home/pi/go

Here is my build sequence:
Fetch / pull the latest Stratux version from Github.
Export the GO Path: export GOPATH=~/stratux
Switch to the stratux directory: cd stratux
Invoke the make command: make
Invoke the make install command: sudo make install
Reboot the R-Pi
Refresh Browser to clear the cache and load the new features.
Start using the new version of Stratux.

I tried your export suggestions above and there were all sorts of errors when I invoked the make command.

Thanks again for helping this coding novice.

cyoung, 2015-12-24:
Start from login:

     sudo su -
     rm -rf stratux
     git clone https://github.com/cyoung/stratux
     cd stratux
     export PATH=/root/go/bin:${PATH} 
     export GOROOT=/root/go
     export GOPATH=/root/go_path
     make
     make install

Axtel4, 2015-12-26:
Thank you for the information. 
It looks like I have two copies of Stratux on my R-Pi. One copy is located in /root/stratux and the other copy is located in /homepi/startux.  I now have my make process pointed to the correct locations for Startux and Go and I have it compiling without any errors by entering the following:

sudo su –
cd /home/pi/stratux
export PATH=/home/pi/go/bin:${PATH} 
export GOROOT=/home/pi/go
export GOPATH=/home/pi/stratux
make
make install

Question: 
Other than only having one copy of Stratux on the R-Pi, what is the advantage of deleting my current Stratux cloane and having Stratux located in /root/stratux as opposed to also being located in /_**user**_/stratux?

cyoung, 2015-12-27:
You can put it wherever you like, I just do everything as root because permissions don't really matter on the device. Since I set up things with root, a lot of the tools (like the go binaries) are in /root. I just use /root/stratux for convenience, and would recommend that because that's how it comes set up. But set it up however you like, really.

cyoung, 2015-12-28:
Reopen if you run into any other issues.

closed 2015-12-28 by cyoung

Axtel4, 2015-12-28:
I think I am good. I pulled 0.5r3 and built it this evening and it went smoothly.  Thanks again for your assistance.

McNugget6750, 2022-08-04:
I have the same issue:
**/bin/sh: 1: go: not found
make[1]: *** [Makefile:31: gen_gdl90] Error 127
make[1]: Leaving directory '/root/stratux'
make: *** [Makefile:28: all] Error 2**

However:
**root@stratux:~/stratux# ls -la /root/go/bin/
total 16512
drwxr-xr-x  2 root root     4096 Apr  8  2020 .
drwxr-xr-x 10 root root     4096 Apr  8  2020 ..
-rwxr-xr-x  1 root root 13577060 Apr  8  2020 go
-rwxr-xr-x  1 root root  3321590 Apr  8  2020 gofmt**

So I'm unsure what's going on here.

McNugget6750, 2022-08-04:
Following this https://github.com/cyoung/stratux/issues/787

I got a little further:
sudo apt-get install -y golang
export GOPATH=/home/pi/go
export GOROOT=/root/go
export PATH=$GOPATH/bin:$PATH

ERRORs:

**go: downloading github.com/kidoman/embd v0.0.0-20170508013040-d3d8c0c5c68d
go: downloading github.com/b3nn0/goflying v0.0.0-20210424141101-d83dac8fdc36
go: downloading golang.org/x/net v0.0.0-20210326220855-61e056675ecf
go: downloading github.com/tarm/serial v0.0.0-20180830185346-98f6abe2eb07
go: downloading github.com/mattn/go-sqlite3 v1.14.6
go: downloading golang.org/x/sys v0.0.0-20210324051608-47abb6519492
go: downloading github.com/BertoldVdb/go-ais v0.1.0
go: downloading github.com/ricochet2200/go-disk-usage v0.0.0-20150921141558-f0d1b743428f
go: downloading github.com/gorilla/websocket v1.4.2
go: downloading github.com/uavionix/serial v0.0.0-19700101022104-87f23b1d3198
go: downloading github.com/golang/glog v0.0.0-20160126235308-23def4e6c14b
go: downloading github.com/dustin/go-humanize v1.0.0
go: downloading github.com/skelterjohn/go.matrix v0.0.0-20130517144113-daa59528eefd
go: downloading github.com/jpoirier/gortlsdr v2.10.0+incompatible
go: downloading github.com/stratux/serial v0.0.0-19700101022104-87f23b1d3198
go: downloading github.com/adrianmo/go-nmea v1.3.0
go tool: no such tool "compile"
go tool: no such tool "compile"
go tool: no such tool "compile"
make[1]: *** [Makefile:31: gen_gdl90] Error 2
make[1]: Leaving directory '/root/stratux'
make: *** [Makefile:28: all] Error 2**