cyoung/stratux #161: RY835A GNSS light on

tysonr0, 2015-12-24:
I'm assembling Stratux with RY835A. The GPS works correctly if connected to the Pi via USB.  i.e The GNSS light blinks and the appropriate position info is sent to the Pi.  If I disconnect the USB and rely on the pin connections (TXD to #10, RXD to #8) the GNSS light is constantly lit (no blinking) and no position data is sent to the Pi.   Any suggestions?  Do I need a configuration change to tell Stratux to expect position info from the pins rather than USB?

Ergonomicmike, 2015-12-24:
Lots about this already answered at www.reddit.com/r/stratux/

Nokomis449, 2015-12-24:
Wouldn't it also need power and ground?

tysonr0, 2015-12-24:
I couldn't find an answer to this specific issue at /r/stratux and posted the question there as well.   Regarding power, vdd is connected to #2 and gnd to #6. Power LED is lit.

duecedriver, 2015-12-24:
check that you have tx connected to rx and rx connected to tx....

tysonr0, 2015-12-24:
I have pi pin10 to txd and pi pin8 to rxd.

cyoung, 2015-12-28:
Check your solder work, sounds like you have the wires going to the right places. Try adding another ground or something to one of the many ground pins.

duecedriver, 2015-12-29:
@cyoung 

did you change the code to reduce sat updates to 5hz instead of 10hz while having gps/glonass ?>  someone pointed out 10hz is too fast while using both per the mfg

jpoirier, 2015-12-29:
> @cyoung <https://github.com/cyoung>
>
> did you change the code to reduce sat updates to 5hz instead of 10hz while
> having gps/glonass ?> someone pointed out 10hz is too fast while using both
> per the mfg
That would be me. I've seen ongoing gps lock problems with the 10Hz setting
that didn't occur with the manufacturer's recommended 5Hz setting but I've
not done any significant testing so I've not pushed for the change.

duecedriver, 2015-12-29:
if the spec sheet says 5hz with the configuration we prefer with gps/glonass as a 5hz max.. I say push it.. its a single line change and 5hz is the update spec of the the GDL-90 anyway isn’t it.. if not.. its plenty for a moving map.. we are not driving the autopilot with it…

speaking of autopilot.. have you seen that xavaion (of which I am not a fan) is doing some kind of external gpss autopilot steering via ipad.. interesting to say the least.. for experimental only obviously.. check it out…

back on point.. I say lets set the unit up for 5hz and see what that does.. should use less juice that way as well.

cyoung, 2015-12-29:
The 5Hz limitation was on a GNSS fusion solution with the M8, GLONASS was removed. Apparently GLONASS is removed from the solutions whenever a WAAS solution is used.

jpoirier, 2015-12-29:
10Hz is restricted to GPS only where WAAS and GPS are listed as two different SBAS values, IIRC.

skypuppy, 2015-12-29:
Autopilot is in my plans...  :)  Probably not linked to stratux gdl-90,
but getting data from those sensors.  Thanks, deucedriver, for the info
about someone else already trying it!
Next is ILS function, but in 3D, rather than a simple 1d line.
After that is TCAS -maybe-.
All from a $35 computer and about $100 worth of goodies.

Ongoing is engine sensors monitoring.

Maybe then think about our own free version of EFB.  With all the bells
and whistles, including synthetic vision.  I'm only one guy who moves
very slowly, but hey, I can dream, can't i?

These efforts began in 2001.

David

duecedriver, 2015-12-30:
Sounds good. Perhaps have the engine probes connected to a Bluetooth transmitter the 
Bt into the pi.

tysonr0, 2015-12-30:
I added a second ground path and tried 1, 5, & 10 Hz. In all cases it works on USB but not on pins.  At this point I'm just looking for a short very flexible M-M USB to mini-USB connector. 

BTW The latest version of Foreflight removed the ability to simultaneously receive on Freeflight and Sim so I no longer have AHRS display in Foreflight (interesting upgrade path, i.e. removing capabilities) . At least for now I may just go with a small USB GPS and /or perhaps look at alternatives to FF.  I read that AHRS works in WingX. 

ILS and open EFB would of course be awesome!

skypuppy, 2015-12-30:
Don't know of any that could handle the heat in an engine compartment.

skypuppy, 2015-12-30:
iFly has promised to "look into it."  They're pretty good about such
things if they think the market is there -- BUT they would NOT help me
at all with that when I asked them as an individual last year.
Stratux having the "big name" now could be a strong enough push for them.

jzeevi, 2015-12-30:
I think Ambiqmicro makes one. It's red to at least 60 C. They're CPU part is being qualified at 85 C.

duecedriver, 2015-12-30:
no I mean the wires go into the cabin into a BT module but I guess you could just jack them into the pi direct?  any who.. we have bigger fish to fry right now!! hehe

duecedriver, 2015-12-30:
we had this thread a couple weeks ago,  when the EFB makers that dont make hardware realise that many folks will gravitate to stratux .. they will fall over themselves to add support.. and perhaps want to add modules or features to the stratux that the others dont to fight for market share.

those companies that do make hardware are going to start locking out stratux to try and save their market share… becoming less relevant and perhaps out of business… foreflight right now is considered the standard among EFBs… they might have the staying power and be able to survive it.. everyone else will likely build a stratux.. the elfs and other cheap ADS-b device days are numbered.. clarity does have a very good ahrs and phased array antenna that provides best in class reception without an external antenna and will likely continue to build boxes.. probably selling less though..

stratux needs to stay focused, get documented, and set the open standard..

skypuppy, 2015-12-30:
We're way off topic here, but I won't buy a box ads-b unit at their
prices unless I'm required to at gunpoint.  Even then, I'd contest the
credit card charge at the end of the month.

Skypuppy

shayfrendt, 2015-12-30:
> stratux needs to stay focused, get documented, and set the open standard

:metal:

closed 2016-03-08 by cyoung