cyoung/stratux #163: Avare I/O plugin listening problem

mooneyflier, 2015-12-25:
Since the v0.5b2 beta release the Avare I/O Plugin will not start listening on 40000. Without that, I can't establish GPS or 1090 traffic capabilities on the Avare app.  Is anyone else experiencing this? I'm using a Samsung Galaxy Tab S and Lollipop.  It worked on previous versions.

andy51edge, 2015-12-25:
I don't know if port 40,000 is a typographical error or not but the correct port is 4,000.

mooneyflier, 2015-12-25:
That is a typo. It won't listen on either 43211 or 4000. Another change I've noticed with Avare is they have filtered the traffic out to a max of 100,000 ft. Not that that makes that much difference. I would just like to continue to use Avare and without the plugin working correctly, it makes it much less functional trying to use it with Stratux. Andy51edge, are u having any problems with the plugin??

deandayton, 2015-12-25:
I used Avare last weekend. For some reason it lost its configuration and defaulted to 43211. I had to reconfigure it for port 4000

-------- Original message --------
From: mooneyflier
Date:12/25/2015 6:03 PM (GMT-05:00)
To: cyoung/stratux
Subject: Re: [stratux] Avare I/O plugin listening problem (#163)

That is a typo. It won't listen on either 43211 or 4000. Another change I've noticed with Avare is they have filtered the traffic out to a max of 100,000 ft. Not that that makes that much difference. I would just like to continue to use Avare and without the plugin working correctly, it makes it much less functional trying to use it with Stratux. Andy51edge, are u having any problems with the plugin??

-

andy51edge, 2015-12-25:
I ran into that too. I currently run v0.5b1 but in a few hours i will try v0.5b2 and see if i can duplicate that problem on my end.

Ergonomicmike, 2015-12-26:
Port 4000 works for me. (V0.5b2)

AvSquirrel, 2015-12-26:
With a fresh 0.5b2 image, I can't replicate this with the latest version of Avare (7.1.1) and Avare IO (3.0.6) on Android Lollipop 5.1.1. Heartbeat and traffic both show up on port 4000.

Axtel4, 2015-12-26:
It works for me. Avare 7.1.1, Avare I/O 3.0.6 on my phone Lollipop) and on my tablet (Kit Kat) on both put 4000 and Port 43211.

mooneyflier, 2015-12-26:
Since several of you said it worked, I kept trying. After several tablet and Stratux reboots and numerous re-entries of the correct listening port number it finally works. For some reason either 43211 or yes 40000 defaulted in the port entry cell. I've just never had that much trouble getting it to work before. Thanks for your input and this issue can be closed if that's what needs to be done.

andy51edge, 2015-12-26:
Glad that it is working again. Happy flying

cyoung, 2015-12-26:
The interface is a little finicky, I think if you click the "listen" before you change the port then it actually doesn't update. Also don't click "Save" unless you really want to save debug logs.

closed 2015-12-26 by cyoung

pophop99, 2018-07-13:
I had to disable battery saving mode for it to work