cyoung/stratux #164: Webpage cache needs flushed

Nokomis449, 2015-12-26:
When running 2 different Stratuxes with 2 different versions and switching between the two, the version number does not update correctly when switching between the boxes.

Ergonomicmike, 2015-12-26:
I don't know how to look at the Source for the UI on my tablet, and I don't have wi-fi on my desktop to look. But perhaps the source should have

Cache-Control: no-cache, no-store, must-revalidate
Pragma: no-cache
Expires: 0

If it doesn't already?

http://stackoverflow.com/questions/49547/making-sure-a-web-page-is-not-cached-across-all-browsers

closed 2015-12-27 by cyoung

bradanlane, 2015-12-27:
We should test the performance impact of this. The total web resource download is not insignificant and "no-cache" is a bit like using a sledge hammer on a finish nail.

cyoung, 2015-12-27:
I think the only thing we really needed to fix this was https://github.com/cyoung/stratux/commit/1c231dd301c48c579250097f219dff382b98ad32.

commit 1c231dd3 (2015-12-27): Fix websocket reconnect bug.

Fixes #164.
commit 4f56a4c7 (2015-12-27): Set "no cache" on all served pages.

Fixes #164.