cyoung/stratux #165: AHRS shows inverted at 0 AGL or lower

laluzdiy, 2015-12-26:
I noticed the AHRS ground and sky is reversed when running a simulation on Wingx if the altitude is 0 or below ground level.
[image]

HiltonG, 2015-12-28:
Hi,

The performance of the RY835AI + Stratux is really good, responsive etc (good work Chris).  What you're seeing here is not an issue with Stratux at all.  WingX Pro7 does not check for 0 AGL so this is your being underground and the earth's surface above you.  Something we need to fix.  It is easy to verify this, simply slide your finger down the altitude slider on the right to change altitude and you will magically popup above the earth; i.e. nothing wrong with Stratux or the communication (and hence data stream) between Stratux and WingX Pro7 and nothing being inverted.  The easiest way to solve the problem is to tap on the altitude at the very top, you will see that Altitude Calibration popdown, just add a few feet (and/or correctly calibrate the altitude).

Hilton

laluzdiy, 2015-12-28:
Thanks for responding,
I later realized I could duplicate this without even connecting to stratux so it had to be specific to app.

closed 2015-12-28 by laluzdiy