cyoung/stratux #168: Roll reversed in v0.5b3 vs. v0.5b2 and v0.5b1

laluzdiy, 2015-12-28:
The roll direction seems reversed in v0.5b3 vs. b1 or b2.  When viewing the heading in the web interface it is very stable in v0.5b3 as opposed to b1 or b2 but the roll appears to be reversed.

hclark, 2015-12-28:
I just downloaded v0.5b3 and found the same - both pitch and roll seem to be reversed. At least in the webUI.

AvSquirrel, 2015-12-31:
Confirmed.

Roll direction can be fixed with a sign change in mpu6050.go. Pitch direction is a webUI issue only. Possible fixes below.

https://github.com/AvSquirrel/stratux/commit/8410819ae2ef2454bdbdd68a2b9cce36c877a146
https://github.com/AvSquirrel/stratux/commit/2d7e213af2cf3c8a7328f8625612d2a199cf9a6a

hclark, 2015-12-31:
Thanks. 
I'm not fluent in Unix, so I have no idea how to implement those fixes.
Until such time as the AHRS info. is available to fpGO, FlyQ or FF, it's not an issue for me. I'm sure it will be corrected in the next release.

cyoung, 2015-12-31:
Fixed by Pull Request #175.

closed 2015-12-31 by cyoung