cyoung/stratux #171: Stratux displaying altitude of certain aircaft as "zero"

Ergonomicmike, 2015-12-29:
I'll make it official, based on Scrabo Tower's test with a SkyRadar and a Stratux in the Phoenix area:
"Yeah it's Stratux , N971PA just passed by, SkyRadar showed altitude, Stratux was zero."

https://www.reddit.com/r/stratux/comments/3yld3s/video_of_flight_school_ac_not_displaying_altitude/cyfcux3

short video of the issue at https://vid.me/gzLT

cyoung, 2015-12-29:
I looked at your 0002-es.log.gz and there's only messages from UAL1640. Have you got replay logs for N971PA?

Axtel4, 2015-12-29:
Here is the log I downloaded from his Wiki link.  It has two of the "PA" aircraft; N926PA (Mode S = ACD5BA) and N925PA (Mode S = ACD203). There are also 3 other aircraft that report an altitude of 0 ft.

[0002-es.zip](https://github.com/cyoung/stratux/files/74394/0002-es.zip)

cyoung, 2015-12-29:
```
$ cat 0002-es.log | grep MSG,3 | cut -d, -f2-19  | egrep -v '(ACD5BA|ACD203)'
MSG,3,,,A73DB6,,,,,,,35000,,,,,,
MSG,3,,,A73DB6,,,,,,,35000,,,,,,
MSG,3,,,A73DB6,,,,,,,35000,,,33.30645,-111.80327,,
MSG,3,,,A73DB6,,,,,,,35025,,,33.31256,-111.77715,,
MSG,3,,,A73DB6,,,,,,,35025,,,33.31256,-111.77715,,
MSG,3,,,A73DB6,,,,,,,35025,,,33.31256,-111.77715,,
MSG,3,,,A73DB6,,,,,,,35025,,,33.31256,-111.77715,,
MSG,3,,,A73DB6,,,,,,,35000,,,33.32153,-111.73970,,
MSG,3,,,A73DB6,,,,,,,35000,,,33.32185,-111.73700,,
MSG,3,,,A73DB6,,,,,,,35025,,,33.32222,-111.73689,,
MSG,3,,,A73DB6,,,,,,,35025,,,33.32293,-111.73124,,
MSG,3,,,A73DB6,,,,,,,35025,,,33.32350,-111.73123,,
MSG,3,,,A73DB6,,,,,,,35025,,,33.32386,-111.72815,,
MSG,3,,,A73DB6,,,,,,,35025,,,33.32419,-111.72832,,
MSG,3,,,A73DB6,,,,,,,35000,,,33.32483,-111.72535,,
MSG,3,,,A73DB6,,,,,,,35000,,,33.32554,-111.71975,,
MSG,3,,,A73DB6,,,,,,,35000,,,33.32623,-111.71672,,
MSG,3,,,A73DB6,,,,,,,35000,,,33.32810,-111.71106,,
MSG,3,,,AB6339,,,,,,,4800,,,,,,
MSG,3,,,AB6339,,,,,,,4950,,,,,,
MSG,3,,,AB6339,,,,,,,5175,,,33.47790,-112.12103,,
MSG,3,,,AB6339,,,,,,,6100,,,33.47790,-112.12103,,
MSG,3,,,AB6339,,,,,,,6250,,,33.47790,-112.12103,,
MSG,3,,,AB6339,,,,,,,6275,,,33.49829,-112.14199,,
MSG,3,,,AB6339,,,,,,,6450,,,33.50099,-112.14474,,
MSG,3,,,AB6339,,,,,,,6575,,,33.50332,-112.14709,,
MSG,3,,,AB6339,,,,,,,6700,,,33.50569,-112.14945,,
MSG,3,,,AB6339,,,,,,,6900,,,33.50569,-112.14945,,
MSG,3,,,AB6339,,,,,,,6975,,,33.50569,-112.14945,,
MSG,3,,,AB6339,,,,,,,7000,,,33.50569,-112.14945,,
MSG,3,,,AB6339,,,,,,,7075,,,33.51297,-112.15673,,
MSG,3,,,AB6339,,,,,,,7550,,,33.51297,-112.15673,,
MSG,3,,,AB6339,,,,,,,7775,,,33.52627,-112.16828,,
MSG,3,,,AB6339,,,,,,,7900,,,33.52876,-112.16997,,
MSG,3,,,AB6339,,,,,,,7925,,,33.52922,-112.17030,,
MSG,3,,,AB6339,,,,,,,7950,,,33.52972,-112.17057,,
MSG,3,,,AB6339,,,,,,,8350,,,33.53660,-112.17439,,
MSG,3,,,AB6339,,,,,,,8400,,,33.53783,-112.17491,,
MSG,3,,,AB6339,,,,,,,8500,,,33.53954,-112.17574,,
MSG,3,,,AB6339,,,,,,,8550,,,33.53954,-112.17574,,
MSG,3,,,AB6339,,,,,,,8600,,,33.54126,-112.17646,,
MSG,3,,,AB6339,,,,,,,8650,,,33.54252,-112.17693,,
MSG,3,,,AB6339,,,,,,,8700,,,33.54298,-112.17710,,
MSG,3,,,AB6339,,,,,,,8775,,,33.54474,-112.17781,,
MSG,3,,,AB6339,,,,,,,8800,,,33.54524,-112.17798,,
MSG,3,,,AB6339,,,,,,,8875,,,33.54652,-112.17842,,
MSG,3,,,AB6339,,,,,,,8900,,,33.54726,-112.17865,,
MSG,3,,,A1C534,,,,,,,3675,,,,,,
MSG,3,,,AB6339,,,,,,,8925,,,33.54776,-112.17882,,
MSG,3,,,AB6339,,,,,,,8950,,,33.54852,-112.17911,,
MSG,3,,,AB6339,,,,,,,9000,,,33.54955,-112.17945,,
MSG,3,,,AB6339,,,,,,,9075,,,33.55104,-112.17991,,
MSG,3,,,AB6339,,,,,,,9100,,,33.55155,-112.18008,,
MSG,3,,,AB6339,,,,,,,9125,,,33.55206,-112.18025,,
MSG,3,,,AB6339,,,,,,,9475,,,33.56044,-112.18291,,
MSG,3,,,AB6339,,,,,,,9575,,,33.56300,-112.18375,,
MSG,3,,,AB6339,,,,,,,9625,,,33.56401,-112.18409,,
MSG,3,,,AB6339,,,,,,,9700,,,33.56528,-112.18449,,
MSG,3,,,AB6339,,,,,,,10075,,,33.57312,-112.18706,,
MSG,3,,,AB6339,,,,,,,10250,,,33.57669,-112.18818,,
MSG,3,,,AB6339,,,,,,,10325,,,33.57794,-112.18861,,
MSG,3,,,AB6339,,,,,,,10350,,,33.57870,-112.18885,,
MSG,3,,,AB6339,,,,,,,10400,,,33.57994,-112.18924,,
MSG,3,,,AB6339,,,,,,,10450,,,33.58046,-112.18941,,
MSG,3,,,AB6339,,,,,,,10475,,,33.58122,-112.18969,,
MSG,3,,,AB6339,,,,,,,10650,,,33.58553,-112.19107,,
MSG,3,,,AB6339,,,,,,,10675,,,33.58681,-112.19149,,
MSG,3,,,AB6339,,,,,,,10700,,,33.58731,-112.19165,,
[... 579 lines cut ...]
MSG,3,,,ACD203,,,,,,,0,,,33.47566,-111.98535,,
MSG,3,,,ACD203,,,,,,,0,,,33.47516,-111.98516,,
MSG,3,,,ACD203,,,,,,,0,,,33.47516,-111.98516,,
MSG,3,,,ACD203,,,,,,,0,,,33.47516,-111.98516,,
MSG,3,,,ACD203,,,,,,,0,,,33.47516,-111.98516,,
MSG,3,,,ACD203,,,,,,,0,,,33.47516,-111.98516,,
MSG,3,,,ACD203,,,,,,,0,,,33.47516,-111.98516,,
MSG,3,,,ACD203,,,,,,,0,,,33.45649,-111.97933,,
MSG,3,,,ACD203,,,,,,,0,,,33.45602,-111.97930,,
MSG,3,,,ACD203,,,,,,,0,,,33.45575,-111.97930,,
MSG,3,,,ACD203,,,,,,,0,,,33.45392,-111.97930,,
MSG,3,,,ACD203,,,,,,,0,,,33.45374,-111.97927,,
MSG,3,,,ACD203,,,,,,,0,,,33.45300,-111.97925,,
MSG,3,,,ACD203,,,,,,,0,,,33.45250,-111.97925,,
MSG,3,,,ACD203,,,,,,,0,,,33.45149,-111.97919,,
MSG,3,,,ACD203,,,,,,,0,,,33.44998,-111.97914,,
MSG,3,,,ACD203,,,,,,,0,,,33.44998,-111.97914,,
MSG,3,,,ACD203,,,,,,,0,,,33.42009,-111.97826,,
MSG,3,,,ACD203,,,,,,,0,,,33.41687,-111.97821,,
MSG,3,,,ACD203,,,,,,,0,,,33.41515,-111.97815,,
MSG,3,,,ACD203,,,,,,,0,,,33.41235,-111.97798,,

```

ACD5BA and ACD203 are the only ones in the logfile for which dump1090 is decoding 0 altitude.

cyoung, 2015-12-29:
https://github.com/antirez/dump1090/issues/53

Looks like a bug in dump1090. Since all the info we have recorded from dump1090 is this replay log, would you be able to set up this version of dump1090? https://github.com/MalcolmRobb/dump1090

Do the following:

1. sudo su -
2. cd /tmp
3. git clone https://github.com/MalcolmRobb/dump1090
4. cd dump1090/
5. make
6. cp dump1090 /usr/bin/dump1090

Then restart the device.

Should go something like:

```
root@raspberrypi:~# cd /tmp
root@raspberrypi:/tmp# git clone https://github.com/MalcolmRobb/dump1090
Cloning into 'dump1090'...
remote: Counting objects: 1401, done.
remote: Total 1401 (delta 0), reused 0 (delta 0), pack-reused 1401
Receiving objects: 100% (1401/1401), 4.14 MiB | 3.62 MiB/s, done.
Resolving deltas: 100% (850/850), done.
root@raspberrypi:/tmp# 
root@raspberrypi:/tmp# cd dump1090/
root@raspberrypi:/tmp/dump1090# make
gcc -O2 -g -Wall -W `pkg-config --cflags librtlsdr`  -c dump1090.c
gcc -O2 -g -Wall -W `pkg-config --cflags librtlsdr`  -c anet.c
gcc -O2 -g -Wall -W `pkg-config --cflags librtlsdr`  -c interactive.c
gcc -O2 -g -Wall -W `pkg-config --cflags librtlsdr`  -c mode_ac.c
gcc -O2 -g -Wall -W `pkg-config --cflags librtlsdr`  -c mode_s.c
gcc -O2 -g -Wall -W `pkg-config --cflags librtlsdr`  -c net_io.c
gcc -g -o dump1090 dump1090.o anet.o interactive.o mode_ac.o mode_s.o net_io.o `pkg-config --libs librtlsdr` -lpthread -lm 
gcc -O2 -g -Wall -W `pkg-config --cflags librtlsdr`  -c view1090.c
gcc -g -o view1090 view1090.o anet.o interactive.o mode_ac.o mode_s.o net_io.o `pkg-config --libs librtlsdr` -lpthread -lm 
root@raspberrypi:/tmp/dump1090# cp dump1090 /usr/bin/dump1090
```

Axtel4, 2015-12-29:
I see those in addition to A1B1C4, A51BF5, and AD6E27 with 0 feet.

I also found this discussion about a 0 ft Mode C reported issue with FR24.
http://forum.flightradar24.com/threads/7356-Wrong-altitude-readings-is-it-a-receiver-problem

skypuppy, 2015-12-29:
I've seen the zero altitude as well.  I figure it has something to do
with the messages coming from their airplanes not conforming to spec
-or- some obscure spec in stratux that we have missed.

David

skypuppy, 2015-12-29:
Some airplanes don't have mode C, after all.

Ergonomicmike, 2015-12-29:
At step 6 I got an error message saying -
cp: missing destination file operand after `dump1090/usr/bin/dump1090'
Try `cp --help' for more information.

Ergonomicmike, 2015-12-29:
Never mind - looked up the cp command. I missed the space between the file names. Rebooting now.

Ergonomicmike, 2015-12-29:
Rebooted. No joy. Still seeing "zero" for N418PA. Interestingly, before that Flight tag painted, there was a different number showing in the tag. The zero altitude problem couldn't possibly have anything to do with the "PA" tail number, could it? (Since that is one thing they all have in common here.)

I'm going to reimage and do your make one more time just to be doubly sure I'm not the problem.

Axtel4, 2015-12-29:
Actually Mode C is interleaved with Mode S and Mode A.  Depending on the pulse spacing of the interrogator pulse, the Mode S transponder will reply with Mode A, Mode C, or Mode S.

When the transponder receives an P1 - P3 8 microsecond spaced pulse on 1030 MHz it will reply with the Mode A code, with a P1 - P3 21 microsecond pulse it will reply with the Mode C altitude information, if it receives a P6 pulse, it will respond with the Modes S reply.

Axtel4, 2015-12-29:
ErgomonicMike, I think what Chris is after is a different set of logs.  If I was following him correctly, the version of dump1090 used with Stratux has an issue with logging whereas  the proposed version doesn't.

Ergonomicmike, 2015-12-29:
Hi Axtel4 - Ah, okay. I thought this was a fix. Turns out I'm on my third retry. I got a permission error on the cp command last time. I'm such a noob.

Axtel4, 2015-12-30:
Here is an example of the data I recorded.  In addition to other items, notice it contains a date and timestamp while the other log file didn't.
 
4009538731,MSG,8,111,11111,A3F4D7,111111,2015/12/29,17:01:42.198,2015/12/29,17:01:42.197,,,,,,,,,,,,
4009785656,MSG,4,111,11111,C035CC,111111,2015/12/29,17:01:42.200,2015/12/29,17:01:42.198,,,463,321,,,0,,,,,0
4009859249,MSG,8,111,11111,AD00E3,111111,2015/12/29,17:01:42.213,2015/12/29,17:01:42.199,,,,,,,,,,,,0
4009914509,MSG,7,111,11111,A2BE96,111111,2015/12/29,17:01:42.214,2015/12/29,17:01:42.199,,38000,,,,,,,,,,0
4010658825,MSG,4,111,11111,A7818F,111111,2015/12/29,17:01:42.234,2015/12/29,17:01:42.201,,,474,310,,,0,,,,,0

Ergonomicmike, 2015-12-30:
I'm about to try Chris's instructions again. Do I need to kill some process so that the dump file isn't protected?

Ergonomicmike, 2015-12-30:
And I do I need to have first run the sudo raspi-config to resize a partition?

AvSquirrel, 2015-12-30:
I'll bet this has to do with the lack of "Q" bit handling in the antirez branch of dump1090.

Bits 41-52 of the 1090ES position messages include a 12-bit altitude field. One of these bits (bit 48, the "Q" bit) determines whether altitude is encoded in 25-foot or 100-foot increments:
* If Q = 1, the remaining 11 bits are decoded as an integer "N", where altitude = N*25 - 1000. Valid altitude range is -1000 to 50,175'.
* If Q = 0, the remaining 11 bits are decoded as [Mode C Gillham code](http://www.airsport-corp.com/modecascii.txt)

The dump1090 implementation we're using only handles Q = 1 -- [it returns zero if Q = 0.](https://github.com/cyoung/stratux/blob/master/dump1090/dump1090.c#L850-L866)

The MalcomRobb branch @cyoung references [appears to handle the Q = 0 case correctly](https://github.com/MalcolmRobb/dump1090/blob/master/mode_s.c#L701-L722).

Axtel4, 2015-12-30:
@Ergonomicmike I didn't, I just did it as cyoung had listed. However, you may want to kill the Stratux  process before you start do: 

sudo pkill screen

That will stop the Stratux processes.  To verify run: top and you should not see any Stratux processes.  Follow that with <CTRL> C exit top.

After the cp command, restart Stratux and you should be good to go.  Also, after restart, go to settings and turn on the logs.

Ergonomicmike, 2015-12-30:
I ended up disabling the 1090 SDR - that allowed me to cp the file. 

I think it's working. I rebooted and enabled logging. Am waiting for a PA to fly by.

Just got one - it's showing altitude!

Dump file coming soon.

cyoung, 2015-12-30:
Yep, good find.

Axtel4, 2015-12-30:
@AvSquirrel I was just starting to research the "Q" bit.  From DO-181:
Variable Direct Data
Variable direct data characterize the flight condition of the aircraft and are therefore subject to dynamic changes. These changes are made either automatically, based on inputs received from sensors in the aircraft, or manually by the flight crew.

a. Pressure Altitude
When operated in conjunction with a pressure-altitude encoder (digitizer) or an air data system, the transponder shall have the capability for pressure-altitude transmission up to its designed maximum altitude. Pressure altitude reports shall be referenced to a standard pressure setting of 29.92 inches Hg (1013.25 hectopascals).

(1) ATCRBS – When operated in the ATCRBS system, the altitude shall be transmitted in response to a Mode C interrogation, encoded in accordance with ICAO Annex 10. If altitude information is not available to the transponder, only the framing pulses required for a Mode C response shall be transmitted.

(2) Mode S – When operated in the Mode S system, the 13-bit AC field (bits 20-32 transmitted in the short and long special surveillance reply, the altitude surveillance reply, and the altitude Comm-B reply) shall contain the encoded altitude of the aircraft. For aircraft with 25-foot or better pressure altitude sources, pressure altitude-information should be reported in 25-foot increments. Pressure altitude data obtained from a source with larger than 25-foot resolution shall not be reported using 25-
foot increments. The altitude shall be encoded as follows:

(a) Bit 26 shall be designated as the M bit and shall be ZERO if the altitude is reported in feet. M equals ONE shall be reserved for possible future use to indicate that the altitude reporting is in metric
units.

Note: Use of the M bit as defined here does not alter any conventions regarding the X bit in ATCRBS replies.

(b) If M equals ZERO, bit 28 shall be designated as the Q bit. Q equals ZERO shall be used to indicate that the Mode S altitude is reported in 100-foot increments as defined in c below. Q equals ONE shall be used to indicate that the altitude is reported in 25-ft increments as defined in d below.

Note: Bit 28 (Q) corresponds to the D1 pulse of a Mode C reply and is
not used in the ATCRBS altitude code.

(c) If M and Q both equal ZERO, the altitude shall be coded according to the pattern for Mode C replies specified in ICAO Annex 10. Starting with bit 20, the sequence shall be C1, A1, C2, A2, C4, A4, ZERO, B1, ZERO, B2, D2, B4, D4.

(d) If M equals ZERO and Q equals ONE, the 11-bit field represented by bits 20 to 25, 27, and 29 to 32 shall represent a binary-coded field whose least significant bit has a value of 25 feet. The binary value of the decimal number N shall be used to report pressure altitudes in the range (25 x N - 1000 ±12.5) feet.

Note: The most significant bit of this field is bit 20 as required by subparagraph 2.2.1.4.2. This code is able to provide code values only between -1000 ft and +50,175 ft. The coding used for Mode
C replies in c above must be used to report pressure altitudes greater than 50,175 ft.

(e) ZERO shall be transmitted in each of the 13 bits of the AC field if altitude information is not available or if the altitude has been determined to be invalid.

Ergonomicmike, 2015-12-30:
Well, this might be moot now, but here's the latest log file. Look for N926PA.

wikisend.com/download/554958/0004-es.log.gz

cyoung, 2015-12-30:
That file is 10 bytes

closed 2015-12-30 by cyoung

Ergonomicmike, 2015-12-30:
Sorry. Good help is so hard to find. (But I can shoot an ILS partial panel to ATP standards.) 

wikisend.com/download/591642/0005-es.log

Look for N416PA

cyoung, 2015-12-30:
```
$ cat 0005-es.log  | grep A4EBB0 | grep MSG,3
1558409500809,MSG,3,111,11111,A4EBB0,111111,2015/12/30,00:43:07.628,2015/12/30,00:43:07.619,,4400,,,33.41505,-111.99370,,,,,,0
1563718296745,MSG,3,111,11111,A4EBB0,111111,2015/12/30,00:43:12.940,2015/12/30,00:43:12.928,,4400,,,33.41776,-111.99320,,,,,,0
1582787079654,MSG,3,111,11111,A4EBB0,111111,2015/12/30,00:43:32.008,2015/12/30,00:43:31.997,,4500,,,33.42732,-111.99194,,,,,,0
1584623207674,MSG,3,111,11111,A4EBB0,111111,2015/12/30,00:43:33.854,2015/12/30,00:43:33.833,,4500,,,33.42823,-111.99211,,,,,,0
```

http://flightaware.com/live/flight/N416PA/history/20151229/2351Z/tracklog

~0043Z showing it at 4400-4500 ft.

Ergonomicmike, 2015-12-30:
So, is it a confirmed kill?

cyoung, 2015-12-30:
Yeap, have it fixed now in the current master. Will be included in the next release (probably tonight or tomorrow).

Ergonomicmike, 2015-12-30:
Excellent. Gotta love the synergy of the Internet. Lots of really smart people here. (I'm obviously not one.)

ScraboTower, 2015-12-30:
Thanks @cyoung and @Ergonomicmike for following thru on this, I thought it was just me with those "PA" aircraft showing zero and just never realized until @Ergonomicmike ask me to compare the target against a SkyRadar unit.

commit 17c43dbb (2015-12-30): Update build process to use new dump1090.

Fixes #171.