cyoung/stratux #174: RTL error messages when no UAT traffic ? - "failed with -4"

bradanlane, 2015-12-31:
Are the following messages "normal" with the new power savings mode (when not receiving UAT) ?

The timestamped lines make sense but all of the "failed with -4" are worrisome.
````
2015/12/31 15:58:07  - GPS vertical velocity: 0.00 ft/sec; GPS vertical accuracy: 0 m
2015/12/31 15:58:07 Pausing UAT listening for 50 seconds - none received.
2015/12/31 15:58:07 Entered UAT shutdown() ...
2015/12/31 15:58:07 UAT shutdown(): calling uat_wg.Wait() ...
2015/12/31 15:58:07 UAT read(): shutdown msg received...
2015/12/31 15:58:07 UAT shutdown(): uat_wg.Wait() returned...
2015/12/31 15:58:07 UAT shutdown(): closing device ...
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
r82xx_write: i2c wr failed=-4 reg=06 len=1
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_write_reg failed with -4
````

cyoung, 2015-12-31:
Not normal, no. So .Close() on the device is throwing these errors, previous to 15:58:07 were any of these errors appearing?

bradanlane, 2015-12-31:
I just fetched code from github (probable two weeks worth of changes); build; install; restart Stratux. That is when I saw the errors.

cyoung, 2015-12-31:
I mean that do the errors show up on opening/initializing the SDR or just after "closing device ..."?

bradanlane, 2015-12-31:
(sorry) - startup all looks good. Then, I get that block of messages each time it cycles the SDR.

cyoung, 2015-12-31:
Hmm, do you have another dongle that you can use for UAT to see if it does this uniformly with different SDRs in your case?

bradanlane, 2015-12-31:
I don't know what is going on with that code but its hammering the CPU. I see a 10C temp spike.

jpoirier, 2015-12-31:
I just noticed the uat duty cycle code changes. Any idea if the problem
existed before those changes were made?

bradanlane, 2015-12-31:
@jpoirier - I cant say for certain if the problem occurred "just prior" to the duty cylce change. I can say I was fine as of about two weeks ago (the last time I had time to work on the Stratux UI).

Something from then to now chanced. It could be code or it could be hardware. I'm going to start with a clean image and work forward from there.

bradanlane, 2015-12-31:
it appears the starting and stopping of the SDR is CPU intensive. I have an enclosure on my Stratux which is not well vented. I see the CPU temp just 10C when the SDR is restarted. This was enough to cause a number of problems and ultimately throw a fault code. I added a heatsink which lowered the initial temperature so the spike was within limits.

My only guess is that with the old code, the spike happened when the system was relatively cold and thus the 10C peak was tolerated.

I'll comment out the SDR power-down and see where my system temp stabilizes.

bradanlane, 2016-01-01:
If it helps, the "failed with -4" errors do not appear in the logs. I only see them when I run "gen_gdl90" directly on the command prompt.

bradanlane, 2016-01-01:
it would appear I have trashed one SDR. I swapped in an SDR from another setup and I no longer get the error.

cyoung, 2016-01-01:
Interested to know if the newest code caused the error to surface somehow? Would you be able to test with v0.4r4? git checkoout tags/v0.4r4

bradanlane, 2016-01-01:
I'll have to setup my system to pull the old code. While the "failed with -4" errors seem to have disappeared with the replacement SDR, I'm still crashing out of gen_gdl90 after 15-30 minutes.

cyoung, 2016-01-01:
Do you debug output for the crash?

bradanlane, 2016-01-01:
I have the crash trace dump. I first verified my build. Next I verified the hardware. When I have time I will look at 0.4r4 and I will look at the dump. I have limited time each day to poke the bear.

bradanlane, 2016-01-02:
It would appear to be a timing issue. Possibly an protected common resource? I am running V0.4r4 and get the same issue as with the current code. I was able to get Stratux to start up if I unplugged the SDR, started Stratux and then plugged the SDR in.

Here is one dump of the crash - in case any of this makes sense to somone:

````
root@stratuxpi:~# /usr/bin/gen_gdl90
1970/01/01 00:02:54 Stratux v0.4r4 (d0d466896ee0b0efbfb66868e695327dd3021553) starting.
1970/01/01 00:02:54 can't read settings /etc/stratux.conf: open /etc/stratux.conf: no such file or directory
1970/01/01 00:02:55 ===== UAT Device name: Generic RTL2832U OEM =====
Found Rafael Micro R820T tuner
1970/01/01 00:02:56 	GetUsbStrings - Realtek RTL2838UHIDIR 00000001
1970/01/01 00:02:56 	GetTunerType: RTLSDR_TUNER_R820T
1970/01/01 00:02:56 	SetTunerGainMode Successful
1970/01/01 00:02:56 	SetTunerGain Successful
Exact sample rate is: 2083334.141630 Hz
[R82XX] PLL not locked!
1970/01/01 00:02:56 	SetSampleRate - rate: 2083334
1970/01/01 00:02:56 	GetSampleRate: 2083334
1970/01/01 00:02:56 	GetXtalFreq - Rtl: 28800000, Tuner: 28800000
1970/01/01 00:02:56 	SetXtalFreq - Center freq: 28800000, Tuner freq: 28800000
1970/01/01 00:02:56 	SetCenterFreq 978MHz Successful
1970/01/01 00:02:56 	GetCenterFreq: 978000000
1970/01/01 00:02:56 	Setting Bandwidth: 1000000
1970/01/01 00:02:57 	SetTunerBw 1000000 Successful
1970/01/01 00:02:57 	ResetBuffer Successful
1970/01/01 00:02:57 	GetFreqCorrection: 0
1970/01/01 00:02:57 	SetFreqCorrection 0 Failed, error: invalid parameter(s)
1970/01/01 00:02:57 	ReadSync Failed - error: overflow
SIGILL: illegal instruction
PC=0x6e386 m=5
signal arrived during cgo execution

goroutine 36 [syscall, locked to thread]:
runtime.cgocall(0x13338, 0x10800ce4, 0x0)
	/root/go/src/runtime/cgocall.go:120 +0x11c fp=0x10800cc8 sp=0x10800cb0
github.com/jpoirier/gortlsdr._Cfunc_rtlsdr_close(0xb3000468, 0x0)
	github.com/jpoirier/gortlsdr/_obj/_cgo_gotypes.go:128 +0x34 fp=0x10800ce0 sp=0x10800cc8
github.com/jpoirier/gortlsdr.(*Context).Close(0x1076a2b8, 0x0, 0x0)
	/root/go_path/src/github.com/jpoirier/gortlsdr/rtlsdr.go:228 +0x30 fp=0x10800d1c sp=0x10800ce0
main.sdrReader()
	/root/stratux/main/sdr.go:127 +0xeb4 fp=0x10800fd4 sp=0x10800d1c
runtime.goexit()
	/root/go/src/runtime/asm_arm.s:1036 +0x4 fp=0x10800fd4 sp=0x10800fd4
created by main.sdrWatcher
	/root/stratux/main/sdr.go:154 +0xd8

goroutine 1 [syscall]:
syscall.Syscall(0x3, 0x0, 0x107e8000, 0x1000, 0x0, 0x0, 0x0)
	/root/go/src/syscall/asm_linux_arm.s:17 +0x8
syscall.read(0x0, 0x107e8000, 0x1000, 0x1000, 0x0, 0x0, 0x0)
	/root/go/src/syscall/zsyscall_linux_arm.go:783 +0x78
syscall.Read(0x0, 0x107e8000, 0x1000, 0x1000, 0xc81a0, 0x0, 0x0)
	/root/go/src/syscall/syscall_unix.go:160 +0x4c
os.(*File).read(0x1076a0e8, 0x107e8000, 0x1000, 0x1000, 0xc5dd8, 0x0, 0x0)
	/root/go/src/os/file_unix.go:211 +0x54
os.(*File).Read(0x1076a0e8, 0x107e8000, 0x1000, 0x1000, 0xc7284, 0x0, 0x0)
	/root/go/src/os/file.go:95 +0x7c
bufio.(*Reader).fill(0x10745f44)
	/root/go/src/bufio/bufio.go:97 +0x1c4
bufio.(*Reader).ReadSlice(0x10745f44, 0x3b00a, 0x0, 0x0, 0x0, 0x0, 0x0)
	/root/go/src/bufio/bufio.go:328 +0x264
bufio.(*Reader).ReadBytes(0x10745f44, 0x2f310a, 0x0, 0x0, 0x0, 0x0, 0x0)
	/root/go/src/bufio/bufio.go:406 +0x7c
bufio.(*Reader).ReadString(0x10745f44, 0x10745f0a, 0x0, 0x0, 0x0, 0x0)
	/root/go/src/bufio/bufio.go:446 +0x44
main.main()
	/root/stratux/main/gen_gdl90.go:906 +0x5e8

goroutine 17 [syscall, locked to thread]:
runtime.goexit()
[... 107 lines cut ...]

goroutine 29 [chan receive]:
main.printStats()
	/root/stratux/main/gen_gdl90.go:814 +0x8c
created by main.main
	/root/stratux/main/gen_gdl90.go:898 +0x504

goroutine 30 [chan receive]:
main.cpuTempMonitor()
	/root/stratux/main/gen_gdl90.go:442 +0x4c
created by main.main
	/root/stratux/main/gen_gdl90.go:901 +0x518

goroutine 35 [chan receive]:
main.(*uibroadcaster).writer(0x107be020)
	/root/stratux/main/uibroadcast.go:32 +0x48
created by main.NewUIBroadcaster
	/root/stratux/main/uibroadcast.go:18 +0xf4

goroutine 50 [chan receive]:
main.icmpEchoSender(0x107cc080)
	/root/stratux/main/network.go:233 +0x70
created by main.sleepMonitor
	/root/stratux/main/network.go:270 +0x134
````

bradanlane, 2016-01-02:
We may be picking up changes from external sources as well (such as the various Go repositories) and possibly changes from dump1090, dump978, and rtlsdr ?

jpoirier, 2016-01-02:
The error below indicates that an old version of gortlsdr and/or librtsdr
is being used. librtlsdr's rtlsdr_set_center_freq function returns an error
when you send a freq correct that's already set. gortlsdr was initial
changed to intercept the returned error and change it to a non-error return
then we forked the librtlsdr code and made the changes their.

 "SetFreqCorrection 0 Failed, error: invalid parameter(s)

jpoirier, 2016-01-02:
Actually, it's the two parts GetFreqCorrection 0 then SetFreqCorrection 0
which is the indicator...

bradanlane, 2016-01-02:
This is deliberately older code: V0.4r4. What we are trying to track down is the crash.

jpoirier, 2016-01-02:
Sorry, didn't catch that. Then obviously you can ignore that error message.

jpoirier, 2016-01-02:
btw, what version of Go are you compiling with?

bradanlane, 2016-01-02:
Since this is a backup image from November it has GO 1.5.1 (IIRC)

Ergonomicmike, 2016-01-02:
Don't know if this is related: am flying v0.5b3 now w/ the MalcolomRobb 1090dump. Had to reboot the Stratux four times in the Phx area when on 978. Kept losing heartbeat. Even had to pull SDR out 'cause it wouldn't boot. After out of Phx and no traffic, didn't have to reboot anymore. Wasn't logging - Sorry.

cyoung, 2016-01-02:
@Ergonomicmike - I assume you mean v0.5b3 plus those commands to update to MalcolmRobb, or the most current master branch?

@jpoirier - do you think we should use submodules for critical parts like gortlsdr and keep it at a specific commit?

Ergonomicmike, 2016-01-02:
@cyoung Yes, the special MR code you talked me thru. Otherwise stock v0.5b3.

I'll turn on logging for next two flights. Meantime looks like someone opened a similar issue.

skypuppy, 2016-01-02:
Have you by chance done a lot of power-offs without shutdown?  Missing
files points in that direction.  Try a fresh image with a fresh SD
Card.  If that still happens, then I would suspect hardware failure on
the Pi2.

bradanlane, 2016-01-02:
This is my development system so it always gets a systematic shutdown via SSH. This unit had never been flown.

jpoirier, 2016-01-02:
It seems the targets went from mostly stationary to erratically moving in
an uncomfortably short period of time. :(

@cyoung - as far as Go code is concerned, the Go compiler version 1.6 (in
beta at the moment but usable) supports vendoring natively and that's what
I'd recommend as opposed to submodules.

jpoirier, 2016-01-03:
Was looking for justification to buy either a BladeRF (http://nuand.com/)
or a USRP B200 (http://www.ettus.com/product/details/UB200-KIT) board. I
may have found one; it'd be really handy to have a test bench setup that
can transmit ads-b out messages.

And to preempt any concerns, yes I have access to rf shielded test
enclosures and labs.

cyoung, 2016-01-03:
Charge it to the company card!

bradanlane, 2016-01-03:
Here's one data point to consider: I loaded V0.4r4 and had the crash dump I posted above. To get past the grasp on initial start, I removed the SDR and then executed `/etc/init.d/stratux start`

This allowed Stratux to start and then I plugged the SDR in.

This setup has been running for 9.5 hours and counting. 

In that time the USB GPS managed a lock (sitting in my office window) and it received the occasional UAT traffic passing above.

bradanlane, 2016-01-03:
FYI - I just rebooted and Stratux crashed immediately. I unplugged the SDR and tried starting Stratux. It took two attempts but did eventually start. Then I plugged the SDR back in.

I will let it run for as long as it is willing (or when I wake up, which ever is longer).

cyoung, 2016-01-03:
You could modify /usr/bin/start_uat and add to the end ">/var/log/templog 2>&1"

jpoirier, 2016-01-03:
Wish I could but since the purchase of alcatel-lucent any excess
expenditure approvals are difficult to come by! We did have a couple B200
boards in-house that I was able to play with for a bit but they ended up in
Germany after the project that was using them got killed.

cyoung, 2016-01-03:
I mean the stratux company card :)

Well, I've been thinking about getting a HackRF - do you think BladeRF is more appropriate for our purposes?

skypuppy, 2016-01-03:
For me, if I got a HackRF, the temptation would be TOO GREAT to turn it
into ADS-B in and **OUT**.  Which is not only dangerous to me and others
but also somewhat illegal to use as an OUT.  It is very, very tempting,
though.

Be velly, velly careful with transmitters in the aviation band.

<insert evil laugh>

Skypuppy

jpoirier, 2016-01-03:
I think HackRF  would work fine but look here for a comparison:
http://www.taylorkillian.com/2013/08/sdr-showdown-hackrf-vs-bladerf-vs-usrp.html

8-bit IQs, half duplex, slower interface speeds, and lower sample rate kind
of limit its use.

jpoirier, 2016-01-03:
You could use it as a test bench to transmit ads-b provided the tx power is
low enough such that it doesn't radiate out of the confines of your home.

skypuppy, 2016-01-03:
What, you think I got that kind of self-control and willpower? 
Hahahahahaahahaha.

(Just trying to be funny here, guys.  No worries.)

jpoirier, 2016-01-03:
On Sat, Jan 2, 2016 at 10:04 PM, skypuppy <notifications@github.com> wrote:

> What, you think I got that kind of self-control and willpower?
> Hahahahahaahahaha.
>
> (Just trying to be funny here, guys. No worries.)
>
> Mischievous ADS-B transmissions would most probably be looked at as
domestic terrorism and likely get said individual an unlimited free stay at
a domestic rendition center sans due process most riki tic.

bradanlane, 2016-01-03:
Back to the issue (smirk) :-)

Once I started Stratux without the SDR and then plugged it in, the V0.4r4 has been running an additional 9.5 hours.

Now I need to figure out what is causing the startup crash with `ReadSync failed - error: overflow`. My suspicion is some change in the RTL code that gets pulled in during build.

bradanlane, 2016-01-03:
I swapped in a pristine new virgin Pi with V0.4r4 and a single SDR and it booted and ran without error.

This has me concerned that something we are doing is stressing the hardware somehow (or just really bad luck).

bradanlane, 2016-01-03:
I may be crazy but I tried a third pristine new virgin Pi with a fresh image of my V0.4r4 and pristine new virgin SDR. 

It failed to boot. Manually starting `/usr/bin/gen_gdl90` showed the same crash with the `ReadSync Failed - error: overflow`

Manually starting `/usr/bin/gen_gdl90` with the SDR not inserted, starts. Then inserting the SDR and it remains operating.

bradanlane, 2016-01-03:
I'm back on the current code stream. Either by luck or using the "start w/o SDR" trick, I can get Stratux to run. However, with the power saver code active, it will crash sooner or later. With that code disabled, it runs longer - how long is being tested now.

cyoung, 2016-01-03:
This isn't the OrangePi hardware is it?

bradanlane, 2016-01-03:
No. I assume any orange issues are mine to resolve in quiet desperation :-)

mooneyguy, 2016-01-03:
I just bought a new RPi2 because my old one died running the V0.5b3 (worked fine in the plane on the ground but at start up on the first flight, it died). What does that mean? The wifi light and both antennas did not light up but the red and green lights on the RPi2 board were solid red and green). The gps module did not light up at all. The configuration did work fine in the previous version. I put in the new RPi2 last night and everything worked except the gps would only have a power light on solid so I just disconnected the gps for now since I am mostly interested in traffic and weather. However, while on the ground with the new board installed and using V0.5b3 I still get 0 feet altitude for some planes but the other information looks good.

bradanlane, 2016-01-03:
should we merge issue #174 and issue #180 ?

Ergonomicmike, 2016-01-03:
I had to pull the SDR to get the Pi to reboot again today, this time with v0.5b4. (On the ground at home, after a soft-reboot.)

Since it's new firmware but since everything else is the same, I'm beginning to wonder if my particular new problem of having the pull the SDR out before booting isn't a power (a.k.a. KMASHI) issue? Regardless of why my Pi started requiring reboots in flight (although I suppose that could also be a KMASHI issue), it might be that, during initialization, with warm components, there's a power demand when the SDR comes on line that causes the KMASHI to drop too much and stalls the boot process? I say this because I noticed that the green light on the Pi stopped flickering when the SDR's LED came on.

I will see if I have a 5v AC P/S that works with the Pi and will test later to see if I can get thru reboots using that with the SDR connected.

bradanlane, 2016-01-03:
I'm running with a stable regulated power supply (measuring both voltage and current) and I am seeing the same issue:

> had to pull the SDR to get the Pi to reboot again today, this time with v0.5b4

bradanlane, 2016-01-03:
I still don't know what the real issue is, but I can reliably startup with the following change:
````
func (u *UAT) read() {
	time.Sleep(5 * time.Second)
	defer uat_wg.Done()
	log.Println("Entered UAT read() with 5 second delay ...")
````

However, if I have the SDR power saving "cycle down" code active, I still crash sooner or later.

jpoirier, 2016-01-03:
I've only done a cursory look at the power saver code and it looked okay,
so it's possible that there's a problem with constantly enabling and
disabling a dongle from either librtlsdr's and/or libusb's point of view.

Since we're using our own fork of librtlsdr adding diagnostics and/or
improvements will be straight forward.

bradanlane, 2016-01-03:
I agree. I do not believe the crash is in the power saving code. I seems the issue is related to something that is going on in parallel with the SDR setup code. The cycling of the SDR just means we have lots more occurrences of that code being executed.

bradanlane, 2016-01-03:
I need some Golang expertise. I was playing with a simple for-loop with nothing in it but `time.Sleep(1 * time.Second)'

The signal from this sleep timer triggered the crash inside the rtlsdr 'C' code.

Are there limitations or special requirements for executing 'C' code concurrently with GO code ?

jpoirier, 2016-01-03:
On Sun, Jan 3, 2016 at 1:34 PM, bradanlane <notifications@github.com> wrote:

> I need some Golang expertise. I was playing with a simple for-loop with
> nothing in it but `time.Sleep(1 * time.Second)'
>
> The signal from this sleep timer triggered the crash inside the rtlsdr 'C' [...]
Yes there are limitations and not all of them are well defined
(requirements) such that there can be very subtle bugs.

E.g. passing pointers to memory from Go to C code is okay but passing a Go
pointer to pointer to memory to C code was a bug (Go garbage collector
restrictions) that wouldn't cause a compile time error until the recent
version 1.6 of the Go compiler.

bradanlane, 2016-01-03:
@jpoirier - do you know of any limitations related to concurrent code execution?

Is it possible one GO process has called into rtlsdr while another GO process is already executing inside rtlsdr?

Since we have multiple GO processes running concurrently - including the sdrWatcher and syncRead - we may be "thread unsafe".

jpoirier, 2016-01-03:
On Sun, Jan 3, 2016 at 2:15 PM, bradanlane <notifications@github.com> wrote:

> @jpoirier <https://github.com/jpoirier> - do you know of any limitations
> related to concurrent code execution?
>
> Is it possible one GO process has called into rtlsdr while another GO
> process is already executing inside rtlsdr?
>
With multiple dongles this has to happen.

> Since we have multiple GO processes running concurrently - including the
> sdrWatcher and syncRead - we may be "thread unsafe".
>
From Go's and gortlsdr's point of view concurrent code execution shouldn't
be a problem but not sure about librtlsdr. gortlsdr creates a per device
object where methods and state are encapsulated/associated with only that
particular object.

Go uses its own scheduler and very inexpensive goroutines, which are
overlayed on to system threads, for concurrency. By default, when a Go app
runs it creates the number of system threads equal to the number of system
cores (this is configurable programatically). When a call is made in to C
code from Go code the goroutine that the C code was called from is pulled
from the running queue (there's no reason to wait in the Go code because
it's not doing anything) and another goroutine is called, when the C code
returns its goroutine starts back up.

jpoirier, 2016-01-03:
@bradanlane - not sure it'll solve the problem but there's a pull request (
https://github.com/cyoung/stratux/pull/183) that removes a call to an
already configured dongle that you can try.

On Sun, Jan 3, 2016 at 2:35 PM, Joseph Poirier <jdpoirier@gmail.com> wrote:

> On Sun, Jan 3, 2016 at 2:15 PM, bradanlane <notifications@github.com>
> wrote:
>
> With multiple dongles this has to happen.
>
> From Go's and gortlsdr's point of view concurrent code execution shouldn't
> be [...]

bradanlane, 2016-01-03:
the code block I commented out is the following:
````
		// Shutdown UAT for 50 seconds, check every 60 seconds if the count is 0.
		if time.Since(lastUATCheck) >= 1*time.Minute {
			if UATDev != nil && globalStatus.UAT_messages_last_minute == 0 {
				log.Printf("Pausing UAT listening for 50 seconds - none received.\n")
				UATDev.shutdown()
				UATDev = nil
				stopCheckingUATUntil = time.Now().Add(50 * time.Second)
			}
			lastUATCheck = time.Now()
		}
````

[28 more replies]

closed 2016-01-08 by bradanlane