cyoung/stratux #176: GPS "Featue"?

Axtel4, 2015-12-31:
I am not trying to be a pain, but I have been seeing an interesting GPS issue.  

While watching my App, I see the GPS position will drop periodically.  What I discovered is that the longitude will switch signs in the Web IU.  For example if the actual own aircraft position is +38 Lat / -95 Lon it will sometimes swap the Longitude 180 degrees to become +38 Lat / +95 Lon.  When it does this, the App screen blanks (moves off the chart area) and the map presentation in the Web UI moves to some location in West Central China.  There are also some times when the Web UI map shows the aircraft position somewhere in the middle of the ocean while the App screen blanks.

Looking at the GPS log, I see where some of the recorded sentences appear to contain garbage:
Good position data:
124362999804	$GPÇGA	162902.6	3800.00000	N	9500.00000	W

Bad position data:
123365849417	$ÇÐGGÁ		162901.6	3800.00000	N	9500.00000	×
or
123700451214	$PUÂØ	0	162901.9	3800.00000	Î	9500.00000	×

I also see what looks looks garbage for the GPS sentence name in the log:
$GPÇGA
$ÐÕBX
$PÕÂX
$PÕBX
$PUBØ
$ÐUBØ
$PUBØ
$PÕBX
$PUÂX
$ÐUBX
$PUÂX

I am setting stationary, sourcing the GPS with the RY835 and running Stratux 0.5r3.

cyoung, 2015-12-31:
Do you have the latest version of master? Paste your startup line, e.g.:

     Stratux v0.5b3 (f51bd61b3b6474c7a4e596ce7c9adbae20ba7f30) starting.

Axtel4, 2015-12-31:
Will I need to log into console to see it or can I:

pi@raspberrypi ~/stratux $ git describe --abbrev=0 --tags
v0.5b3
pi@raspberrypi ~/stratux $ git log -n 1 --pretty=%H
052b4e28557aa922035271dace5978b23ea07a2a
pi@raspberrypi ~/stratux $

duecedriver, 2015-12-31:
its signal noise on the serial line....

Axtel4, 2015-12-31:
> its signal noise on the serial line.... 

Maybe, maybe not.

cyoung, 2015-12-31:
There was a fix that @AvSquirrel put in just after commit 052b4e28557aa922035271dace5978b23ea07a2a, do a git pull, make, make install.

Axtel4, 2015-12-31:
Thanks.  I'll give it a try later and I need go run some errands. I'll try that before moving the RY835 from the R-Pi to test one variable at a time.

Axtel4, 2015-12-31:
The errand got delayed so I had some time to do a quick test.

Well, it looks like we may have our first confirmed case of EMI. All data now looks good in the log and the GPS position is stable.  I also had a quicker lock to the GPS signal. 

200962534147	$GPGGA		180934	3800.00000	N	9500.00000	W
200991097122	$PUBX	0	180934	3800.00000	N	9500.00000	W

What's different?
I still have the RY835 connected via the GPIO but I had enough wiring to set it beside and parallel to the R-Pi instead of over the SDRs and the R-Pi and WiFi dongle.

duecedriver, 2015-12-31:
tada…

Axtel4, 2015-12-31:
@duecedriver  Just curious, what does your Stratux unit look like and how did you shield it? 

https://www.reddit.com/r/stratux/comments/3w9di9/found_my_system_crash_problem/cxv93nu

cyoung, 2015-12-31:
It didn't have to do with EMI, it was an issue with the GPS initialization sequence.

closed 2015-12-31 by cyoung

Axtel4, 2015-12-31:
@cyoung I did some additional testing after I got back home and it indeed is not an interference issue.  With the previous build on 5 out of 10 starts, the GPS would flip the Longitude and have bad data in the log.  I then pulled and built the most current updates and with 10 of 10 starts the GPS stayed solid.  In addition the Web UI shows a consistent DGPS (SBAS/WASS) whereas before, 3D GPS was predominate with DGPS being hit or miss.  

@AvSquirrel, the Ground Speed flag now shows dashes for invalid GS and the roll and pitch appear to be correct.  Thanks for the fixes.  

Thanks again for all you guys do.