cyoung/stratux #177: RPi datetime being set to bad GPS value

bradanlane, 2015-12-31:
It looks like the GPS code is trying to set the RPi clock even before it has a good datetime. In my output I see if go from "now" back to 1980 *(no DeLorean needed)*.

```
2015/12/31 22:21:55 stats [up since: 3 minutes ago]
2015/12/31 22:21:55  - CPUTemp=57.00 deg C, MemStats.Alloc=8.9 MB, MemStats.Sys=13 MB, totalNetworkMessagesSent=0
2015/12/31 22:21:55  - UAT/min 0/0 [maxSS=0.00%], ES/min 0/0
, Total traffic targets tracked=0
2015/12/31 22:21:55  - Last GPS fix: a long while ago, GPS solution type: 0 using 0 satellites (0/0 seen/tracked), NACp: 0, est accuracy 0.00 m
2015/12/31 22:21:55  - GPS vertical velocity: 0.00 ft/sec; GPS vertical accuracy: 0 m
2015/12/31 22:22:01 setting system time to: 1980-01-06 00:10:10 +0000 UTC
1980/01/06 00:10:34 stats [up since: 36 years from now]
1980/01/06 00:10:34  - CPUTemp=53.00 deg C, MemStats.Alloc=4.7 MB, MemStats.Sys=13 MB, totalNetworkMessagesSent=0
1980/01/06 00:10:34  - UAT/min 0/0 [maxSS=0.00%], ES/min 0/0
, Total traffic targets tracked=0
1980/01/06 00:10:34  - Last GPS fix: a long while ago, GPS solution type: 0 using 0 satellites (0/0 seen/tracked), NACp: 0, est accuracy 0.00 m
```

bradanlane, 2015-12-31:
this has a negative side effect on the new UAD idler code. The UAT SDR was powered down, the the GPS clock went back in time, now the UAT timer never wakes the UAT SDR back up.

cyoung, 2015-12-31:
Ahhh

AvSquirrel, 2016-01-01:
@cyoung -- I've got this one. This will just take adding a bounds check for GPS week into  processNMEALine().

U-blox tries to calculate a date from the GPS week number, even if that value is null or invalid. This incorrect date is then reported in the PUBX,04 message. Contrast to GxRMC, where date is left blank.

```
// cold boot:
00:02:32  $GPTXT,01,01,02,Resetting GPS*68
00:02:32  $GPTXT,01,01,02,ANTSTATUS=INIT*25
23:59:44  $GPTXT,01,01,02,ANTSTATUS=OK*3B
...
// Invalid date #1 (Jan 12, 3236 ...65535 weeks after GPS epoch)
23:59:44  $GPRMC,,V,,,,,,,,,,N*53
23:59:45  $PUBX,04,235945.00,120136,604785.00,65535,16D,0,0.000,21*5A
...
// Invalid date #2 (Jan 11, 1980.. week 0 of GPS epoch)
00:04:44  $GPRMC,000444.00,V,,,,,,,,,,N*79
00:04:44  $PUBX,04,000444.00,110180,432284.00,0,16D,0,0.000,21*5F
...
Valid date (Jan 1, 2016... week 1877 of GPS epoch)
00:06:03  $GPRMC,000603.00,V,,,,,,,010116,,,N*7F
00:06:03  $PUBX,04,000603.00,010116,432362.99,1877,16D,-30960396,0.000,21*7D
```

bradanlane, 2016-01-01:
Good work AvSquirrel.

It seems the code is evolving and stabilizing at a good pace.

skypuppy, 2016-01-01:
There is "gps fix" data on one of the NMEA lines.  You could just
continue to sample that line until the fix becomes good at at least a
2D.  Then the gps clock should be fine and you could use it.  Save a ton
of coding for ya.

AvSquirrel, 2016-01-01:
https://github.com/AvSquirrel/stratux/commit/0f611560d40e8209ae617d9f118e1c61d7e56297

Ublox stores weeks since epoch (1980-JAN-06) as a uint16. This code validates that we're in week 1877 (i.e this week) or later, while protecting against overflow (i.e. the week 65535 bug seen during startup).

I also zero-padded the GPS time string to proper formatting (single-digit minutes and seconds weren't being parsed properly), and tightened up the conditions to reset the Pi clock to GPS time. Used to be only if we were 10 minutes slow; now the clock will reset if it's more than three seconds off UTC.

@skypuppy -- appreciate the sentiment, but this was a straightforward change, and only took a half a dozen LOC. This approach also lets us set the date and time before the GPS has position lock.

PR #178 (2016-01-01): Check for valid GPS date. Fix time string formatting.
Fixes #177. Validated on U-blox 7 (VK-172).
closed 2016-01-01 by cyoung