cyoung/stratux #179: Request: means to configure RPi networking

bkwny, 2016-01-02:
In earlier issue https://github.com/cyoung/stratux/issues/158 the participants discussed the notion of configuring the AP SSID.  In closing the issue @cyoung recognized the requirement and mused about possible solution approaches.  In this issue I am adding my two cents' worth and asking for the issue to remain open and to be marked as future enhancement.

In the few days I have been running I've found more than the AP SSID needing configuring.  To avoid colliding with my SkyRadar unit, I've changed the channel number.  To help keep my neighbors' kids out of my device, I've changed the AP security to WPA-2.  I can see needing to change the SSID to a value I choose, but not just yet.  I made all of my changes by editing /etc/hostapd/hostapd.conf and restarting.  I'll suggest the web interface would be a great place to do these things.  Even better, if there were a way for the web interface to upload my desired hostapd.conf from my laptop, that would be terrific.  An upload scheme would make it easy to cope with micro SD reflashes.   It would be OK to require a Stratux reboot to put the uploaded hostapd.conf into effect.

There might also be a need for the user to be able to edit or upload a new edition of /etc/network/interfaces.  One consideration that comes to mind is changing the wired Ethernet to static IP.  I haven't thought this through yet.  Yes, it would be better if my Linksys router's DHCP server would let me mark certain MACs as always to be given the same IP.

Thanks for listening.
Brian

skypuppy, 2016-01-02:
The Linksys router should be able to do that if you get into the admin menu.

bkwny, 2016-01-03:
Hello @skypuppy, because of your comment I found the DHCP reservation button in my Linksys router.  Thank you.

cyoung, 2016-01-03:
Maybe it's best to do this as an external configurator app, since there are many one-off and sparsely needed features like this. Are you using OSX @bkwny?

bkwny, 2016-01-04:
No, not OSX.  Ubuntu 14.04 LTS or Win 7 or Win XP.

bradanlane, 2016-01-04:
one thing that came up a while back which might be of new consideration:

if we move the stratux web server (servicing of UI files) out to something like `lighttpd' and retain the web interfaces (websockets and REST APIs) within `managementinterface` but on its own port, then others could freely add to the capabilities of the base system.

They could create scripts with web front ends for changing the networking, uploading files, downloading files, perhaps even perform software updates without needing to get into the stratux code.

Is this of interest and has it's time come?

skypuppy, 2016-01-04:
I like it.

ssokol, 2016-01-04:
+1. I've been thinking the same thing for a while. I wonder if we wouldn't
be somewhat better off if we broke the project out into several
sub-projects:

Stratux Core - the gen_gdl90 application (ADS-B, GPS, AHRS). This would
retain the status websocket server and the REST API for management /
configuration.  Ideally this would also include packaging scripts that
produce .deb files for updates and the init script(s) needed to launch /
respawn. (Web interfaces would need appropriate CORS / crossdomain
configuration since they would be running on a separate server from the
static server.)

Stratux Web - the web admin interface. This could be independently
developed and could be expanded to incorporate additional features
including a firmware update utility. It would interact with Stratux purely
through the REST and WebSocket interfaces.

Stratux Platform - the collection of changes that need to be made to the
default Raspbian image to prepare it for distribution. Would include the
installation of hostpd, dhcpd, lighttpd, etc. Probably also all the changes
necessary to operate in a purely read-only format. We might also want to
have a variant for installing a developer system (i.e. includes golang).

cyoung, 2016-03-08:
Anyone up for making an app?

cyoung, 2016-03-08:
Duplicate #228.

closed 2016-03-08 by cyoung