cyoung/stratux #189: Web UI sometimes reports wrong screen resolution

Ergonomicmike, 2016-01-04:
A really small nit, and it might not be important.

I have an Asus ZenPad S 8. It has a screen resolution of 2048 x 1536. But the Web UI reports 987 x 741 on the Logs page. 

That's half of actual.  (I tweaked my ppi from the stock 320 to 332. So I'm really at 1974 x 1482.)

I don't know why the Web UI reports screen resolution. So perhaps the easiest solution is to delete that report?

bradanlane, 2016-01-04:
The WebUI shows the reported viewport size on the Log page for debugging and development purposes. Eventually it will go away.

FYI - the viewport size is what the device browser reports and is not the same as the screen resolution.

Ergonomicmike, 2016-01-04:
Okay. I run the browser at "full screen" (Android) & "Desktop mode." So I expected it to report my native resolution. Sounds like a bug in my browser. But since it's just a report & will go away someday, I'll close this non-issue. Thanks!

closed 2016-01-04 by Ergonomicmike