cyoung/stratux #190: Heading Indication in WebUI

laluzdiy, 2016-01-04:
I don't think this is really an issue but something I observed.  On versions 5b3 and 5b4, the heading indication shown in the WebUI has a 90 degree quadrant that doesn't work.  When slowly rotating the unit - the heading jumps between 135 degrees and 225 degrees (at least on my unit).

I don't think this information is used anywhere except the WebUI and if so it's not really an issue.  Also the heading appear to be 180 degrees off.  I'm assuming here that the forward direction on the RY835AI module should be side with the InvenSense, Honeywell, and Bosch sensors as that is the proper orientation for the AHRS pitch.

AvSquirrel, 2016-01-04:
AHRS heading always initializes to 360°, regardless of the orientation of the sensor. Once Stratux gets a GPS lock and detects a ground speed greater than 7 knots, it will slew the heading to true course. That wasn't being handled right in 0.5b4, but #186 should resolve it.

Jumps could be caused by a low-accuracy GPS lock (i.e. you're stationary, but location shifts) causing a false ground track. The 7 knot filter will eliminate this 99%+ of the time unless reception is especially bad.

We're not doing anything with the magnetometer yet.

cyoung, 2016-01-06:
> I'm assuming here that the forward direction on the RY835AI module should be side with the InvenSense, Honeywell, and Bosch sensors as that is the proper orientation for the AHRS pitch.

That's correct.

Nokomis449, 2016-01-08:
I had the same issue and it's still there in 5.5, as there is an area of the compass (mine seems to be roughly 110deg to maybe 210deg) that the paper airplane will not point to while sitting flat on the table and being slowly turned by hand.  No GPS lock in the house.

cyoung, 2016-01-08:
There's a calibration sequence that needs to be done for the MPU6050, the tools are on there if you're able to use the command line? If not it will be built in to either the web interface or automatic soon.

AvSquirrel, 2016-01-08:
@cyoung -- this one isn't a calibration problem; it's due to libimu not properly handling the lack of compass input in the current codebase.

The Euler angle output from mpu9150, passed through libimu, should be 0 ± π radians (0±180°). What was being passed was 0 ± 0.75π radians (0 ± 135°) ... or 225° - 135° in compass heading.

Why the difference? mpu6050.c aligns gyro yaw to magnetic heading, using `yaw_mixing_factor` [as a proportional gain](https://github.com/cyoung/linux-mpu9150/blob/master/mpu9150/mpu9150.c#L410-L411) to fuse the two inputs:
```
	deltaMagYaw = newMagYaw - newYaw;
        ...
	if (yaw_mixing_factor > 0)
		newYaw += deltaMagYaw / yaw_mixing_factor;
```
This mixing factor is taken as an argument by `mpu9150_init()`. Our problem lies in the fact that libmpu.c [passes a constant](https://github.com/cyoung/linux-mpu9150/blob/master/libimu.c#L55-L66) -- in this case, "4". But, we're not using the mag compass input, so magYaw is always 0.

The net result is that the yaw output is reduced by a factor of 1/4, and that full-scale misses a full 90° of the compass.

The quickest fix short-term fix would be to [update libmpu.h](https://github.com/cyoung/linux-mpu9150/blob/master/libimu.h#L12) to change `DEFAULT_YAW_MIX_FACTOR` to 0. Make sure to force libimu to re-compile (or just delete the .o /.so files) when you `make`.

Longer-term we either get the magnetometer working (feed as I2C slave and/or switch to a 9150/9250) or update libimu and its go wrapper to pass the appropriate gain factor.

skypuppy, 2016-01-08:
YIPPEE!

commit e00c7a82 (2016-01-10): Updated linux-mpu9150.

Fixes #190.
closed 2016-01-10 by cyoung