cyoung/stratux #192: CPU usage changes with SDR remove/insert

bradanlane, 2016-01-06:
Has anyone else observed that the CPU utilization is much higher when Stratux starts from boot-up vs when Stratus is running after an SDR has been removed/inserted?

I typically see about 105% CPU (drops to about 7% with the UAT power savings is active) for gen_gdl90 when Stratus is run, either from boot-up or when started from the command line. This is for a single SDR.

If I pull the SDR and then re-insert it (doesn't matter if it is pulled while Stratux is running or pulled after stopping stratux), then Stratux has a peak CPU of about 45%.

ssokol, 2016-01-06:
Huh. Have not tested that scenario. Will do so now.

Any difference in functionality after a remove / re-insert? Do you know if
its actually receiving properly?

cyoung, 2016-01-06:
v0.5b4, or latest?

bradanlane, 2016-01-06:
Latest code but I also remember seeing it several weeks ago when I was testing the in configured SDR fallback code (dual SDR and pulling 978 switches 1090 to do 978)

bradanlane, 2016-01-06:
As for "still working?" It's not as easy to tell giving I'm testing 979 at my desk. I left it run over night and it did pick up ONE UAT traffic message :-)

Interesting note: if I use the old 4r4 launch method (rtl pipes to dump978 pipes to gen_gdl90) I see the CPU usage at 45%.

cyoung, 2016-01-06:
That's interesting, will try to reproduce the 105% CPU and run the profiler on it.

ssokol, 2016-01-06:
If the weather was better I would run out to the airport and test it with
my airplane.

Really want to buy this:
http://www.barnstormers.com/classified_1087975_SkyguardTWX+ADS-B+with+AHRS+.html

Use it for testing in the office.

laluzdiy, 2016-01-06:
​I was just thinking about ground testing, if a simulated transmission RF
signal could be directly connected via coaxial cable to the SDR antenna
port (at a pico watt level) then RF transmission would not be an issue.
Maybe something like this CC1150RGVT
<http://www.mouser.com/Search/ProductDetail.aspx?R=CC1150RGVTvirtualkey59500000virtualkey595-CC1150RGVT>
FM
transmitter on a chip ​could be used to generate the signal if it can be
programmed outside of it's specified upper frequency range (800 to 928 MHz).
If it can't be programmed outside outside the specified range, then another
option would be (for testing purposes only) to create new test DUMP code
modules with frequencies within the 800 to 928 MHz range.  Of course that
still leaves figuring out the packet and modulation requirements and then
simulating or recording transmissions (including weather?) to use for
testing.
I could see this growing into a project of its own.
Like I said, I was just thinking.

cyoung, 2016-01-06:
Good idea, but to test that the receiver is receiving (and not receiving + demodulating, because the demodulation part has never been an issue) - what about making some really simple on/off self-test sequence hooking up a GPIO pin to the SDR input?

bradanlane, 2016-01-07:
Side effect of the high CPU usage is higher chip temp - about 10C higher.

I'm going to try an experiment to launch dump978 similar to how dump1090 is run.

bradanlane, 2016-01-07:
I've got a running test system where `rtl_sdr | dump978` is run similar to `dump1090` using via `exec.Command`

it seems stable and the CPU for teh processes (for a single SDR + USB GPS) are:
* rtl_sdr 5%
* dump978 33%
* gen_gdl90 7%

So it is right there at 45%. I'll try to flight test this sometime this week ? ... or go park near the UAT tower :-)

bradanlane, 2016-01-07:
Very Strange. Now that I have split the code into separate processes, the high CPU load can be narrowed down.

Processes (for a single SDR + USB GPS) are:
* rtl_sdr 99%
* dump978 1%
* gen_gdl90 5%

If I pull the SDR and re-install it, I see the following:
* rtl_sdr 5%
* dump978 34%
* gen_gdl90 7%

I think the only way to know which of the above are correct is to go park near an ground station with the test system and laptop. I may be able to do this on Sunday.

bradanlane, 2016-01-07:
Has anyone else checked CPU usage?
Is high by default and drops with remove/insert of the SDR?
Is it just my test system (in which case we should close this issue)

cyoung, 2016-01-07:
Not able to reproduce it. You disable the stratux startup process, reboot, run rtl_sdr | dump978 | gen_gdl90? Do you get the 99% CPU on rtl_sdr every time on first boot?

bradanlane, 2016-01-07:
I'm going to close this as "just my problem". If I learn anything useful, I'll add it to this thread.

closed 2016-01-07 by bradanlane

skypuppy, 2016-01-07:
With that large a drop, can you verify that the sdr is actually working?