cyoung/stratux #193: Request: FE redirect immediately before shutdown

jamesshannon, 2016-01-08:
Scenario: I want to shutdown. Go to the FE, choose shutdown on my phone, affirm, then it goes to /shutdown and the unit shuts down. I'm lazy, so I don't close the tab.

Later, I bring the stratux back up, open my browser, see that the 192.168.10.1 tab is open and has a "... connection was lost" message. I hit refresh and ... nothing. In fact, the Wifi dongle stops blinking at about this time.

Turns out that the shutdown page was still "open" and I was refreshing that.

 Easy fix: shutdown and restart pages should be POST's, which is at least semantically correct.

Better fix: they should immediately redirect back to the homepage, before the shutdown starts.

Even better fix: homepage (which is already loaded and in the browser) does a XHR POST to the /shutdown page (technically should have a xss token), which causes the shutdown, but never leaves the loaded webapp. The webapp then goes to 'Status: disconnected' and continues to monitor the server until it comes back up.

Nokomis449, 2016-01-08:
While you're visiting this, please consider moving the reboot and shutdown buttons further apart.  A little turbulence can turn a reboot press into a shutdown press,  which is bad news if the Stratux is on the hat shelf in the back.  I'd like to see them on each side of the div.

ScraboTower, 2016-01-08:
Please put them back on the first page !!!!  They are technically NOT a setting !!!

bradanlane, 2016-01-08:
The likely implementation will be AJAX calls to web service a web service "command" interface. This eliminates any change to the URL. The "shutdown" and "reboot" should default the WebUI back to the "status" page.

Adding more space between the buttons is good feedback.

While the "reboot" and "shutdown" buttons are not technically "settings", they are controls and thus are even less about "status".

We debated even having these buttons as they will not be necessary. However, until Stratux is able to provide the stability of an "appliance" (in part by operating with the SD card read-only) there was enough call for the buttons that they were added.

RobertGary1, 2016-01-08:
The shutdown button is a temporary work around, I wouldn't put too much effort into it. 

Long term  the Stratux image is on a RO file system so you just unplug it. It's an appliance to most users. I've never shutdown mine in fact it powers off when I turn off the master switch.

-Robert

braddward, 2016-01-08:
Good idea on doing the POST.  I can look at adding that and adding some
spacing between the buttons.

skypuppy, 2016-01-08:
Robert,  you are not viewing enough of the picture to be  helpful.

bradanlane, 2016-01-08:
Skypuppy, am I missing something? I would agree with robertgary that the shutdown and reboot buttons are stop-gap for current limitations and issues.

If there were no reports of lockups and drops, I dont think any users would be seeking a "reboot" button. 

Similarly, if pulling the power (or cutting it with a simple switch) had zero negative side effects, I am not aware of any user who would be asking for a "shutdown" button.

skypuppy, 2016-01-08:
Then, obviously,  bad side effects are happening without the graceful
shutdown capability.

bradanlane, 2016-01-08:
This is why so many of use have been discussing the need to make stratux read-only for users.

Axtel4, 2016-01-08:
> Please put them back on the first page !!!! They are technically NOT a setting !!!

+1 this comment.

Ergonomicmike, 2016-01-09:
Even if Stratux goes RO, it will probably need a remote reboot button. (I like to leave mine in the back, which makes power down reboot difficult in flight.) Might as well make it so it doesn't result in a rebootloop sooner rather than later.

RobertGary1, 2016-01-09:
I guess isn't hurt. Although I don't think Stratus has a reboot interface. 

-Robert

Nokomis449, 2016-01-09:
Stratux isn't on v0.5b5, either.

bkwny, 2016-01-09:
I believe a way to power-cycle the Stratux is the only recovery interface needed long-term and that a power-cycle to recover must work 100% of the time barring hardware failure.  The device is an appliance.  So are all of its competitors.  The fact that there's a general-purpose, multi-user, multitasking, MP-capable computing system inside Stratux is an implementation choice that must not show through to its end user.  For my SkyRadar receiver, installed remotely, the only recovery interface I have is to power-cycle it by pulling and resetting its circuit breaker, which is mounted in my instrument panel.  I want the same operational properties for the its eventual replacement.  I want its eventual replacement to be the Stratux.

Ergonomicmike, 2016-01-09:
But this is a portable unit (by design). Even for those who choose to hardwire it to shop's power, it has a battery backup, in case ship's power is lost. So I don't see how pulling a CB could command a reset.

As a precedent for a remote way to reset, my ACK ELT is a portable unit that is strapped to the plane. It runs on batteries but has a panel mounted remote switch that allows the user to reset it. A web UI with a reset button accomplishes vitually the same thing, except without wires. (Except, as I found yesterday, that when the Stratux drops Wi-Fi, the reset button doesn't work.)

Axtel4, 2016-01-09:
I don't use the Web UI Shutdown button much as I installed a hardware switch on my Stratux to shut it down before I pull the plug.  The R-Pi has a Python script running in the background and launches the Shutdown routine when the button is pressed.  I do make use of the Web UI Reset button.

commit 812170d0 (2016-01-10): Merge pull request #199 from braddward/updated-shutdown

Updated shutdown. Fixes #193.
closed 2016-01-10 by cyoung