cyoung/stratux #197: r0.5b5 GPS NEMA Error Issue?

Axtel4, 2016-01-09:
I updated to 0.5b5 and then pulled the latest commits.  After pulling and compiling the latest commit I see invalid NEMA string GPS errors (see the Stratux 2.txt log).  The first log (Stratux 1.txt) was taken from the initial install of the r0.5b5 image where the second log (Stratux 2.txt) was taken after pulling and compiling commit 3c7d700. This also resets my R-Pi clock from local time back to UTC time.  I am running the RY835 connected via the GPIO.  

[Stratux 1.txt](https://github.com/cyoung/stratux/files/83526/Stratux.1.txt)
[Stratux 2.txt](https://github.com/cyoung/stratux/files/83527/Stratux.2.txt)

AvSquirrel, 2016-01-09:
This is beta code, and I'm still being verbose with error handling, especially since it can help troubleshoot GPIO connectivity problems.

All this message means is that the string being passed to the GPS message parser didn't start with a "$", or didn't end have a "*" token (delineates the checksum) somewhere in the message. This is fairly common to see at startup, and is caused by the serial port being opened in the middle of a transmission.

Here's a case where mine caught the end of a UBX,00 message, and threw the error.

```
START,Fri Jan 8 15:39:04 +0000 UTC 2016,Fri Jan 8 15:39:04 +0000 UTC 2016
UNPAUSE,825678305
24086294885,�4.2,5.2,0.742,186.17,0.102,,1.31,1.83,1.38,8,0,0*7F
```

```
2016/01/08 15:39:04 Stratux v0.5b5 (9a1af8e2e1253e756959e2060532dc4b103aa5e4) starting.
...
2016/01/08 15:39:27 Using /dev/ttyAMA0 for GPS
2016/01/08 15:39:28 GPS configuration complete
2016/01/08 15:39:28 GPS error. Invalid NMEA string: Invalid NMEA message
2016/01/08 15:39:47 stats [started: 30 seconds ago]
2016/01/08 15:39:47  - CPUTemp=45.46 deg C, MemStats.Alloc=1.9 MB, MemStats.Sys=6.6 MB, totalNetworkMessagesSent=30
2016/01/08 15:39:47  - UAT/min 0/0 [maxSS=0.00%], ES/min 86/86
, Total traffic targets tracked=8
2016/01/08 15:39:47  - Last GPS fix: now, GPS solution type: 1 using 9 satellites (12/16 seen/tracked), NACp: 10, est accuracy 7.40 m
```

Axtel4, 2016-01-09:
Thanks for the info.  Follow-up question; Why does it have to reset the R-Pi System Time every second?

AvSquirrel, 2016-01-09:
It sets [system time to GPS time](https://github.com/cyoung/stratux/blob/master/main/ry835ai.go#L567-L570) if there's more than a three second difference between the two. That explains the first entry in your log. 

```
2016/01/08 21:17:21 setting system time to: 2016-01-09 04:11:17 +0000 UTC
2016/01/09 04:11:17 setting system time to: 2016-01-09 04:11:18 +0000 UTC
2016/01/09 04:11:18 setting system time to: 2016-01-09 04:11:19 +0000 UTC
```

On the others, that's a timezone but. It turns out that the string being passed to GNU `date` doesn't specify timezone. If the local timezone is set to something other than UTC, `date` assumes local time, and sets the GPS time as local. Hence, your clock was being set hours ahead of real time on the first trip around the loop, and on every subsequent loop (because your clock was more than 3 seconds fast).

I pushed https://github.com/AvSquirrel/stratux/commit/e9e8c5c098b1ff985694547e9ff8f8fa63ca31a8 to resolve this issue. This will correctly set time, even if the Pi is assigned a local time zone:

```
2016/01/09 12:07:15 setting system time to: '20160109 18:05:01.000 UTC'
2016/01/09 12:05:01 Time set from GPS. Current time is 2016-01-09 12:05:01.000835446 -0600 CST
```

cyoung, 2016-01-09:
@Axtel4 - sounds like you upgraded from v0.5b4 to v0.5b5 via git pull. Make sure you disable ntpd on startup, this was done on the release image.

closed 2016-01-09 by cyoung

Axtel4, 2016-01-09:
Thanks.

> sounds like you upgraded from v0.5b4 to v0.5b5 via git pull.

I did indeed.

> Make sure you disable ntpd on startup, this was done on the release image.

Ok, that went over my head.  Where do I make this change? I see in wheezy-stage4 where this is done in a script, 
 
> 44 #disable ntpd autostart 
45 update-rc.d ntp disable 

but I can't see how to do it manually.

Is it just a matter of running "update-rc.d ntp disable" from the command line?

cyoung, 2016-01-09:
>Is it just a matter of running "update-rc.d ntp disable" from the command line?

Yep (as root).

bradanlane, 2016-01-09:
I'll need to re-check but I had build issues with ntpd disabled while doing development. I was getting a compiler warning that the time was slewed.

skypuppy, 2016-01-09:
Just be sure the system time is correct before you turn off the ntp
deamon.  Or doing any programming.
Linux machines are PARTICULAR about system times being correct.

:-)

Sort of killing two birds...  Some gps modules (like the ones from
Adafruit) come with a slot for a coin battery and IIRC, the time of day
(tod) clock (as well as gps init info) is kept current while the battery
is good.  At least it gives you a very quick sat finding/fixing time (if
you haven't moved it too far since last use, like a bunch of miles.) 
And if it does keep the tod clock current as well, then you can set the
RPi2 clock with it anyway.  Guess I'll need to do some tests re clock
time...

Batteries weigh nothing, cost $2 or $3, and keep running for months and
years.

Axtel4, 2016-01-09:
I pulled and compiled your latest commit.  It is working now without continuously resetting the system time.  Note: My Pi is still set to local time and I have not made the change with to update-rc.d.

Thanks.

PR #198 (2016-01-09): GPS time zone fix. Don't assume Pi is set to UTC.
Fixes #197