cyoung/stratux #200: Momentary "blink" in ForeFlight

ssokol, 2016-01-11:
I've noticed ForeFlight's display pause / reset / "blink" in two scenarios. One is when the GPS switches its fix type / accuracy value. Another is when the UAT receiver gains or loses a tower. It appears that when the tower count changes ForeFlight momentarily displays "0 Towers" in red, then updates with the new count. I wish I had a better description or some video of this - sorry.

This might be a ForeFlight thing - hard to tell at this point. Don't know. But... it might be worth adding a debounce timer or some kind of filter to the associated metrics to avoid unnecessarily causing state changes in the app.

Ergonomicmike, 2016-01-11:
I'm sure you know this, but if you're trying to get video of a fast blink, a trick I use when I'm trying to read a BIOS message on my desktop is to set my pocket camera to 60 fps (or more) and play back at 30 on my movie edit software.

cyoung, 2016-01-26:
I get this all the time when offline (not connected to stratux) with FF. It's probably just an issue with their display refresh. Don't think there's much we can do. They have some other issues like the icons that they place not disappearing (like enabling 100LL prices or PIREP layers, then disabling). Have submitted bug reports to them but the issue persists still. Give it some time and reopen if it's still an issue in, say, a month.

closed 2016-01-26 by cyoung