cyoung/stratux #201: gps application formluas

duecedriver, 2016-01-11:
I found this resource that I felt might be of assistance as you try to polish the code and the available output data of stratux especially for GPS

especially useful I feel if you want to provide altitude data with the ry chip is a formula for baro aided GPS altitude that could utilize the onboard pressure sensor.  of course you would have to provide a method of either entering a known QNH or, perhaps like wingx.. just have the ability to set the known elevation and the box calculates the offset.  however in flight as QNH changes when you fly, it would need to be updated.   it would not however be set to 29.92 crossing 18k.. unless you wanted your actual gps/stratux altitude to be the same as ships altimeter operating in the class A... probably not an issue here for like 99% since flight above 18k is rarely done in an unpressurized aircraft and the baro on this chip would obviously not work in a pressurized cabin.. although it could be programmed to give a 'high cabin alt' alert ... but I digress..

https://books.google.com/books?id=t0eGFpSwN0wC&pg=PA223&lpg=PA223&dq=baro+aided+gps+altitude+equation&source=bl&ots=uIcEYFgTqF&sig=2VzyfWGhR6rnBT_fkG06tO6dIwc&hl=en&sa=X&ved=0ahUKEwjh0rLCg6LKAhUT-mMKHV15DBcQ6AEIJjAC#v=onepage&q=baro%20aided%20gps%20altitude%20equation&f=false

skypuppy, 2016-01-11:
I think this is a bad idea.  Interior barometric pressure is absolutely
not reliable, even under ideal conditions.  Link it into the
pitot/static system and you've got something.

duecedriver, 2016-01-11:
agreed, however.. what will be closer to reality.. gps alone or baro corrected?

almost all non pressurized GA aircraft use internal cabin pressure as the alt static

some have a valve that can be opened, others have you break the glass on a gauge

folks here are bitching that the GPS alt is off by several hundred feet, this is the best way to tighten that up..

take it or leave it.. just presenting the reference GPS algorithms for consumption.. 

it was not intended to start yet another useless thread with opinions.. just provide proven algorithms for solving the issues we are currently facing

breaking into the pitot static is a non starter if your aircraft is certified….

ssokol, 2016-01-11:
FWIW - I built a very basic altimeter using the Bosh part a couple of years
ago. It was software corrected based on the OAT and sea-level baro value
(what you get from ATIS / AWOS). With accurate temp and baro values it was
within 20' of my onboard altimeter, and had far less drift over time. It
was not connected to the static line - it just used cockpit pressure.

duecedriver, 2016-01-11:
here is a good website that should give you some indication of how a appropriately installed gps should be functioning.. its the FAA realtime GPS performance site..

the entire US is in the 20 meter or better VDOP (vertical accuracy) bubble again today.. so 60ft or better.. in the military we had very good multipath rejection algorithms and our altitude would usually be MUCH tighter…

emuyshondt, 2016-01-11:
Newer iPads and iPhones haves built-in barometer. As a user, I'd like to use those, before having to buy new hardware.

ssokol, 2016-01-11:
I've been playing with baro altitude from my iPad Mini 4 and iPhone 6S.
They're always off because they're not corrected - ForeFlight does not
allow you to set the baro correction, nor does it appear to automatically
pull the closest from the internet or ADS-B. They also tend to be off
because the sensor is packaged inside a warm electronic enclosure while the
air in which you are flying may be a very different temp.

duecedriver, 2016-01-11:
problems with using altimetry from the apple products

1. not everyone is using an apple product.. thats kinda of a big one...

2. apple really limits what the developers can see, use and manipulate when it comes to the internal hardware so I dont know what kind of hooks the developers would have to work with.. might be worthwhile to look, however it doesnt serve everyone .. see problem one.. so is it worth the effort?

closed 2016-03-08 by cyoung