cyoung/stratux #202: 1090 "drop off" after 5 to 30 minutes

ssokol, 2016-01-12:
Running 0.5b5. Dual band system with NESDR nano 2s + Vk-172. dmurray14 antennas. Ground testing, so no UAT signals.

When I start the system I get a usual 1090-ES stream - typically 500 messages / minute. After some time, between 5 and 30 minutes, the 1090 stream tapers off to very low numbers. As few as 12 messages per second.

Give the box a hard reboot (the reboot button seems to cause it to hang - something we've seen before) and when it comes back up, we're back to 500+ messages / minute.

Any ideas?

Axtel4, 2016-01-12:
I've been running for an hour or so.  Is this bad? I am not sure what the 7,640/10,178 signifies.

> 2016/01/11 21:42:54 stats [started: 1 hour ago]
2016/01/11 21:42:54  - CPUTemp=37.39 deg C, MemStats.Alloc=2.5 MB, MemStats.Sys=13 MB, totalNetworkMessagesSent=253,433
2016/01/11 21:42:54  - UAT/min 0/0 [maxSS=0.00%], ES/min 7,640/10,178
, Total traffic targets tracked=247

v0.5r5, Dual dongles, RY835, 1090 MHz collinear feeding 1090 dongle, L Band blade feeding 978 Dongle.  Outside MCI Mode C veil.

cyoung, 2016-01-12:
7,640/10,178 means 7,640 messages last minute, 10,178 highest message/min.

ssokol, 2016-01-12:
Wow. That's some serious reception. Best I've seen is in the 1300 messages
per minute range.

I'm based on the east edge of the MCI veil at 0N0. Are you local or just
passing through?

duecedriver, 2016-01-12:
I wrote up something similar last week over on one of the Reddit threads. Same behavior. Although putting an ice pack over the sdr increased traffic

When tuning the sdr with sdr sharp it drifts really badly on narrow band fm   

Can anyone see if something is actually getting jammed up in the code?

Axtel4, 2016-01-12:
Local on the southwest side.

Axtel4, 2016-01-12:
If it helps any, here is an ES range plot and the Stratux log for above.

[image]
[Stratux.txt](https://github.com/cyoung/stratux/files/86773/Stratux.txt)

cyoung, 2016-01-12:
@ssokol - try with no cooling, let 'er get heated.

Axtel4, 2016-01-12:
> I'm based on the east edge of the MCI veil at 0N0. 

Roosterville.  Always wanted to land there just to say I did.

duecedriver, 2016-01-12:
Best I have gotten is like 30 miles.that sitting on the ground testing in the house. Is that in the air

skypuppy, 2016-01-12:
Thermal drift of the rtlsdr?

ssokol, 2016-01-12:
Will do.

ssokol, 2016-01-12:
Yep. 18' wide and 2700' long. I learned to fly here - lots of fun. You just
treat the asphalt as a centerline in the middle of a 50'-wide grass runway.

ssokol, 2016-01-13:
Can't reproduce this at the moment, so closing. (But opening another ticket as I think I may have found a related bug.)

closed 2016-01-13 by ssokol