cyoung/stratux #203: Perfromance Anlysis

Axtel4, 2016-01-13:
Chris, if you don't feel this belongs here you may take it down.

I, like many have been curious about the performance of my Stratux unit and as such I wanted to quantify its performance.

My Stratux comprises of the R-Pi 2 B, a RY835 GPS / AHRS, 2 of the original T2  RTL-SDR dongles and various antenna configurations used during testing.  The pictures of my Statux build are available here: http://imgur.com/a/G085R

The antenna configurations will be discussed later.  The test method was to select an antenna configuration, capture the Stratux logs and post process and analyze the data.

Test Configuration Number 1:
The first test configuration consisted of a tuned Diopole Vertical antenna (one at 1090 MHz and one at 978 MHz) connected directly to the SMA connectors on the Startux unit.  Stratux was place on the dash board of my truck and was parked in a field behind the hose with a clear view to the sky to the north.  The cab and body of the truck shielded the Stratux reception to the south.  A 15 minute recording was taken and 1090ES data was post processed.  The data consisted of anywhere from 1,500 to over 5000 1090ES hits.  The range to the detected traffic was calculated using the Haversine Formula with my truck location at the center.  A Track Plot was created to visualize the range and bearing of the traffic. These track plots utilize the same methods to analyze TCAS performance data. This is a Track plot of the first configuration. 
 
[image]

As can been seen in this plot, the range approaches 170 nm.  This is not bad for an off the shelf RF dongle.  Now I had a baseline to use for the rest of the testing.  (It appears the range is limited due to the view angle of the traffic from my location.  I'll add more about the view angle at a future date.)  

Test Configuration Number 2:
This configuration removed both tuned antennas and feed both dongle via the tuned 1090 MHz antenna and an SMA "T".  Conventional wisdom would say that there would be a 3 dB reduction on power to each dongle and thus a reduction in range by at least one eighth.  This is otherwise know as the invers square law: Where if you double the distance, the power at the receiving antenna is one forth the previous power lever.  Let's see how the looks in practice.

[image]

The above plot still shows a range out to 170 nm.  Based on this data, we still haven't lost reception of the traffic before is moves out of our view.

Test Configuration Number 3:
Test configuration was like Configuration 2 except the 1090 MHz antenna was replaced with the tuned 978 MHz antenna.  

[image]

Notice the range and number of targets are reduced.  The range decreased 26% from 170 nm to 126 nm.  What suggests is the tuning of the antenna to the frequency of interest is of more importance than whether it is feeding to the SDR directly or through a "T".  

That's it for this installment.  Next up is an in-flight performance analysis and an analysis of the view angle.

cyoung, 2016-01-13:
Very useful! Thanks for posting.

Axtel4, 2016-01-13:
You're welcome.

dmurray14, 2016-01-13:
Do you have any of my antennas? Would love to have you test them. If not, let me know and I'll get a set out to you.

JohnOCFII, 2016-01-13:
Nicely documented.  Great case and photos too!  I'd love to see the impact of antenna changes on 978 tower reception.

Ergonomicmike, 2016-01-13:
I was taught the conventional wisdom in school that a signal splitter causes a 3dB reduction in power. And I was taught the reciprocity theorem for antennas. But if a receiver has a high impedance input, then  there is little current flow => little power disippated. (In contrast to a 50 Ohm receiver input.) If correct, and if the SDR's use a FET (or equivalent) for a front end, then I'm not surprised that there's no loss of signal strength/voltage when one antenna directly feeds two receivers.

skypuppy, 2016-01-13:
What does one give up to get that?  Or, what is the downside?  More
current and raised noise floor?

ssokol, 2016-01-13:
FWIW - I contacted the folks at NooElec, looking for a tuned / balanced
splitter. They actually made up several and tested them and found that for
the 978 MHz to 1090 MHz range, there was no appreciable advantage to using
a the splitter vs. using a simple Y cable.

The question becomes, do you sacrifice a bit of both by picking a midpoint
frequency for the antenna, or do you go with a 978 (since that's where most
users get the most value - weather) and lose a bit on the 1090-ES side?

skypuppy, 2016-01-13:
If there is actually any appreciable difference, my vote is toward the
978 messages.  As you said, weather and traffic.

Axtel4, 2016-01-13:
@dmurray14 , I don't have a set of your antennas.  I could test them if you wish.

@JohnOCFII, I have flight data that I plan to post for the tower reception.  As far as 978UAT, I can not find transcribbed log like the 1090ES log.  I also want to alayze direct 978UAT traffic reception, but I don't think that data is stored in a Stratux log.

@Ergonomicmike:
> I was taught the conventional wisdom in school that a signal splitter causes a 3dB reduction in power. 

As I have learned and accepted also.  What I found surprising was that I didn't see a reduction in singal reception no matter the antenna configuration I used: be it a home crafted 75 ohm 1090 MHz collinear antenna, the certified L-Band antenna, or the tuned 1090 MHz / 978 MHz whips.  After looking at the data, I was starting to come to the thought you had that we are working into a high impendance system.  Also, now that I think about it a little more, this may explain the "bad" SDRs that people are reporting.   With the high input impedance the devices may be more succeptible to ESD than other portible receivers thus blowing the input through handling.

@ssokol , Good question.  I have looked a different broad band antenna designs, but I haven't seen any yet that didn't involve some sort impedance matching adding to thier complexity.  To split the frequency, we would need a large diameter antenna element to get the desired bandwidth to cover both bands.  If you want to consider an antenna that would cover the frequency range and still have acceptible perfromance look to the common DME blade antenna.  There may be way to incorporate one in the top of a Stratux box so it could be portable.

egid, 2016-01-13:
I agree with @skypuppy - I'd much prefer an antenna optimized for 978.

Based on these results I'm tempted to pick up an SMA -> 2x MCX splitter cable rather than the T-assembly people have been constructing, and give it a shot with @dmurray14's 978 antenna.

gitmoregrits, 2016-01-13:
@Axtel4 , in this new post you share pictures of your build and cite a "red button" for graceful shutdown of the unit.  Can you share this detail and/or implementation please?

Axtel4, 2016-01-13:
Sure. I'll post something later.

Axtel4, 2016-01-14:
Analysis Part 2, Flight data:
This data was recorded on a cross country flight across central Missouri.  The Stratux was set up with the Whip antennas attached to the Stratux respective bulkhead connectors and was setting on the Glare shield.   Ship power was supplied through a 28V / 5V USB converter.  Cruising altitude was at 8000 ft.

This is the 1090 ES track plot.  As with the previous plots, it can bee seen that Stratux has impressive 1090 ES range; tacking traffic in excess of 250 nm. 

[image]

The next plot is a plot of the view angle from Stratux.  It gives an picture of the angle the traffic is from Stratux perspective.  I use it as a crude elevation plot for the antenna.  Different antenna topologies (1/4 wave, 1/2 wave, 5/8 wave, colinear, j-pole, etc.) will have different elevation angles.  Here we see
 the elevation angle covers from -1 degree to +21 degrees.  Note: This is just for the 1090ES traffic that was tracked.  There was other non 1090ES traffic that didn't get recorded (discussed later).

[image]

The last set of plots are the only method I have to gauge 978 UAT performance.  The towers were visible out to almost 100 nm. 

[image]

Even with the low power of the towers compared to the 1090ES signal I was surprised the range wasn't further.  I suspect I may have an under performing UAT dongle but more testing will be required to make a solid conclusion.

This is the Google Earth image of the tracked tower locations.

[image]

A note about the UAT data collection: Stratux does not provide the UAT data in a Human Readable format like the 1090ES data is captured.  To analyze UAT performance I screen capture the Tower data, however it isn't easy.  (This is where a Lab partner comes in handy.)  Since the an airplane is generally moving in a straight line, it is difficult to capture the tower data from a single point of view.  To simulate this I fly a shallow turn on a point and capture screen shoots of the Web UI Tower page as the aircraft is flown around the point.  I post flight plot the track around the point, determine the center of the circle (the point) and use that position was the reference point for the tower range plot.  Maybe someday Stratux will record the UAT and it can be analyzed directly.  (Feature request?)

During the flight I did see additional traffic both visually and via Datalink TIS.  Since the aircraft is not yet Version 2 ADS-B Out Mode S equipped, I was not able to see the ATCRBS or TIS-B uplinked aircraft via Stratux.  While monitoring the Web UI Traffic page I did notice quite a bit of directly received UAT aircraft.  However, since Startux doesn't record UAT traffic in human readable format I wasn't able post flight analyze their ranges.

That's it for now.  I'll return later to add the results of testing with the Certified L-Band antenna (ground based) and the Homebrew 1090 MHz collinear antenna.

skypuppy, 2016-01-14:
> 1090 whip track
> 1090 whip view
> uat tower range
> uat tower ac

UAT data could run into megabytes in very short order.

Excellent work, Axtel4, and thank you for making the very complicated
into a format even I could understand.

Skypuppy

dmurray14, 2016-01-14:
@Axtel4 - send me your address in a private message - dmurray14@gmail.com

gitmoregrits, 2016-01-14:
@Axtel4 - thank you very much for sharing your shutdown button implementation.  Question for @cyoung and others:  Would the associated script for monitoring the Raspberry Pi that @Axtel4 references be monitoring the GPIO of the subject switch be worth incorporating into the Stratux build?  Seems such a capability might partially address the potential for filesystem corruption on shutdown.

duecedriver, 2016-01-14:
the reason I would design the system to favor 978 is that I might have to get info from a tower at the edge of the reception range while the 1090 squit distance is always going to be proximal to my aircraft and anything outside 20nm for 1090 squits is not really useful anyway unless you are flying a Citation X at mach .98 and covering 6 miles a minute...

Ergonomicmike, 2016-01-14:
I'll be very interested to see your antenna test data on the Murray antennas (w/o ground plane) vs. yours w/ ground plane radials.

As an aside, is it possible to saturate the Stratux w/ too much traffic? I suppose that you're showing empirically that we haven't hit a limit yet. ( How would we know? Does the Stratux drop packets or freeze when overwhelmed?)

Axtel4, 2016-01-15:
These are the Whip antennas I use for testing.  The left (shorter) one is the 1090 MHz antenna and the right (longer) one s the 978 antenna.  

[image]

Axtel4, 2016-01-15:
@skypuppy 
> UAT data could run into megabytes in very short order.
I know some of the decompressed ES logs I have analyzed have approached 50 MB.  I was hoping that there was a way to capture basic traffic and tower information in a csv format similar to the 1090ES log.  For example: For the Tower capture the Lat, Lon, and Signal data.  For the UAT downlink data that is directly received from a tracked aircraft capture the Aircraft Address, Time Stamp, Altitude, Latitude, and Longitude.  The TIS-B or WX data would not need to be captured for this dataset.

Ergonomicmike, 2016-01-15:
Capacitance Hat on the 1090?

Axtel4, 2016-01-15:
@Ergonomicmike 
> As an aside, is it possible to saturate the Stratux w/ too much traffic? 

That is a good question.  I have seen 1090ES receive rates in excess of 10k / min and Stratux doesn't seem to miss a beat or drop targets.  One way to tell, maybe, is to analyze the track plot for each unique Mode S address that was received and see if any expected messages are missing.  The DF17 should be squittered every two seconds and any gaps in the message interval could indicate a drop in the data.  However, is there are drops it would not be know if it was due to saturation, antenna shading, multipath distortion, or a the minimum receive level.

Axtel4, 2016-01-15:
> Capacitance Hat on the 1090?

Yes.  Somebody got over exuberant when tuning the antenna and didn't want to throw it away. So, he called a audible and threw the top hat on it. :-)  It tuned up with about -15 dB of return loss at 1090 MHz.

Axtel4, 2016-01-26:
For those that might be interested, I have documented some emission testing I performed on my Stratux build.

[Stratux Emission Testing.pdf](https://github.com/cyoung/stratux/files/104418/Stratux.Emission.Testing.pdf)

cyoung, 2016-01-26:
Great stuff.

Axtel4, 2016-01-26:
I have a couple more I an working on.  @cyoung  I think what I'd like to do, if it is ok with you, is update the OP and use it a as a repository for PDF files.  I will then edit the OP to upload new files and  information thus keeping them all in one place.

cyoung, 2016-01-26:
Perfect!

Axtel4, 2016-01-29:
For those that are interested I posted a new analysis document at the top of the OP.

Axtel4, 2016-02-05:
Adding a new installment:  Impedance Analysis of the dongles, a splitter, and homebrew antennas.

[Impedance Analysis.pdf](https://github.com/cyoung/stratux/files/119981/Impedance.Analysis.pdf)

Ergonomicmike, 2016-02-14:
Wow - just discovered that you've updated this "issue" and done a lot of quantitative testing. I'm surprised that the Pi is as quiet as it is, unshielded as it is. But I can't argue w/ an HP Analyzer. Nice work.

Axtel4, 2016-03-17:
Latest Stratux Performance Analysis installment: 978 MHz UAT antennas
[978 MHz Antenna Analysis.pdf](https://github.com/cyoung/stratux/files/178694/978.MHz.Antenna.Analysis.pdf)

skypuppy, 2016-03-17:
How critical is the shape and size of the metal base?

Axtel4, 2016-03-17:
I am assuming you are taking about using the Standard antenna.  

 

The shape or size isn’t to critical, but ideally you want a radial system that has a radius of one quarter wavelength at the lowest operating frequency.  The metal base I used for the testing was 5” x 7” because that is what I had laying around in the shop.  The antennas were mounted about a half inch in on mid span of the narrow side with the Stratux mounted between the antennas.  

 

One could take a one inch wide 6” long strip and place the antenna in the middle of the strip.   You can bend the strip to fit within the enclosure.  You can see what mine looked like with copper tape and the bulkhead connectors above in the thread, but the same concept would apply with the Stock antenna.

 

To calculate the minimum radial radius for 978 MHz:

1 Wavelength = (300 M Meters / second) / Frequency in Hertz

1 Wavelength = 300 /978 MHz

1 Wavelength = 300/978 MHz

1 Wavelength = .3067 meters

1/4 Wavelength = .3067 meters / 4

1/4 Wavelength = .0767 meters

Convert to inches:

1/4 Wavelength = .0767 meters * 39.4

1/4 Wavelength = 3.07 inches = Radius of ground plane

Ergonomicmike, 2016-03-21:
Half wave end fed? (Just the inner conductor at a half wave long, taking into account velocity factor.) As I've reported in the reddit, they work fantastic for me taped to the rear window of the Glasair.

Axtel4, 2016-03-23:
This is a method to demonstrate how a radial (i.e. counter poise) can improve signal reception.  It dovetails with the Antenna analysis posted above.

The 1090ES message rate inside my house for a test antenna attached to my Stratux build without a ground plane was around 100 messages / minute.

I took a small paper clip and straightened it to form a radial creating a radial that is approximately one quarter electrical wavelength.
[image]

I clipped it to the bulkhead SMA connector at the base of my 1090 MHz antenna.  
[image]

It made a 10x improvement in the 1090 MHz message rate.
[image]
 

Removing the “radial” the message dropped back to approximately 100 messages / minute again.
[image]

Axtel4, 2016-03-24:
Updated 978 UAT Antenna Analysis
This is an analysis of various 978 MHz UAT antennas.  Version 2 added "no ground" data for the remaining antennas.
[978 MHz Antenna Analysis V2.pdf](https://github.com/cyoung/stratux/files/187217/978.MHz.Antenna.Analysis.V2.pdf)

skypuppy, 2016-03-24:
I want some of your homebrew antennae!  :)

Skypuppy

JohnOCFII, 2016-03-24:
> Updated 978 UAT Antenna Analysis
> This is an analysis of various 978 MHz UAT antennas. Version 2 added "no ground" data for the remaining antennas.

Great information - Thanks!

mooneyguy, 2016-03-24:
@axtel4 what are the dimensions of your wire?

 

 <https://cloud.githubusercontent.com/assets/15059010/13974639/7d74eaf2-f07b-11e5-874e-d69967f0b774.jpg> 20160322_213130

 <https://cloud.githubusercontent.com/assets/15059010/13974620/4eabc394-f07b-11e5-8c16-8dccfd197922.jpg> 20160322_213937

 <https://cloud.githubusercontent.com/assets/15059010/13974800/09a8af76-f07d-11e5-9c70-c09a92d34427.jpg> after

 <https://cloud.githubusercontent.com/assets/15059010/13974730/4b3265c8-f07c-11e5-9355-8e99ad7ee99e.jpg> b4

Axtel4, 2016-03-25:
The wire is a standard 1.25" paper clip I unwound from the inside out leaving the outside bend to clip around the SMA bulkhead connector.  From the top of the crook to the end if the wire is approximately 2 7/8".  This is just shy of a 1/4 wavelength at 1090 MHz.

Axtel4, 2016-04-25:
Stratux Gain testing:

[Stratux Gain Notes.pdf](https://github.com/cyoung/stratux/files/235610/Stratux.Gain.Notes.pdf)

Axtel4, 2016-04-27:
Some SDR numbers.

Prompted by some discussions on Reddit, this post will look at some performance numbers for the Nano 2 SDR.

Nano 2,
SDR# tuned to NOAA WX Radio 162.475 MHz
Heat Sinking - Plastic Shell removed. 2 wraps Kapton Tape over SDR.  2 wraps copper tape over Kapton tape.
LNA gain = 49.6 dB (Max gain)

Part 1: A look at temperature rise v. LNA gain.
Dongle  allowed to stabilize for 15 minutes.  The dongle was centered on the NOAA frequency using the SDR# PPM offset.  PPM setting 35 PPM.  Dongle Temperature 64.4 degrees C

Dongle gain reduced to 38.6 dB and allowed to stabilize for 15 minutes.  PPM = 35; Temp = 65 C

Dongle gain reduced to 28 dB and allowed to stabilize for 15 minutes.  PPM = 35; Temp = 65 C.
Observation: Reduction in gain does not appear to affect temperature rise of the SDR.

Part 2: Add forced air cooling and determine frequency shift.
Gain reset to max and allowed to stabilize for 15 minutes.
Start: PPM = 35; Temp = 65 C
Stop: PPM = 31; Temp = 30.5 C
PPM Shift = approximately 0.12 PPM / Degree C

skypuppy, 2016-04-27:
Random data  point.  A Nano2's shell read a max of 109F today (on continuously for several days) with an average of about 104F.  Read by a non-contact thermometer over two minute periods.

Axtel4, 2016-04-27:
Is that on the outside of the plastic shell?  Were there any hot spots?

I forgto to mention, the temperature measurement method I used was direct contact thermal couple on the copper foil with heat sink compound to eliminate any air gap.

cyoung, 2016-04-27:
The Mini 2:

[image]

[image]

skypuppy, 2016-04-27:
Jeez, I want a FLIR camera so badly I could just spit.  <sigh>
Yes, the hot spot on the Nano2 was very close to where the antenna cable plugs in.  The rest of the plastic case was high 90'sF.

Ergonomicmike, 2016-05-02:
@Axtel4 So then, on your Stratux Gain Notes, is the conclusion to back the gain down 3 to 6 dB (as some where beginning to experiment with in the reddit a couple months ago) in order to get better S/N?

Axtel4, 2016-05-04:
> So then, on your Stratux Gain Notes, is the conclusion to back the gain down 3 to 6 dB (as some where beginning to experiment with in the reddit a couple months ago) in order to get better S/N?

 

Currently it is undetermined. I got mixed results reducing the gain with the SDRs connected to the R-PI.  I want to perform more testing but time hasn't allowed.  Also, the latest AvSquirrel additions to monitor signal strength /  position in real time should speed up testing various configurations.  Now if I could figure out how to change the gain “on the fly” without having to rebuild the code each time it is changed.

Axtel4, 2016-08-25:
[978 MHz Antenna Analysis V3.pdf](https://github.com/cyoung/stratux/files/436011/978.MHz.Antenna.Analysis.V3.pdf)
Added the Digikey (Linx) ANT-916-CW-HWR-SMA-ND antenna to the 978 MHz Antenna Analysis document.

Axtel4, 2016-08-25:
Added the Digikey (Linx) ANT-916-CW-HWR-SMA-ND antenna to the 978 MHz Antenna Analysis document.
[978 MHz Antenna Analysis V3.pdf](https://github.com/cyoung/stratux/files/436016/978.MHz.Antenna.Analysis.V3.pdf)

Axtel4, 2016-08-26:
Revised document to correct Plot labels.  Combined the split charts into one chart based on GPS time stamp (i.e Plot 2a and 2b combined into a single plot 2).  Added a histogram summary chart for all antennas that were mounted over a ground plane.
[978 MHz Antenna Analysis V3a.pdf](https://github.com/cyoung/stratux/files/438398/978.MHz.Antenna.Analysis.V3a.pdf)

Axtel4, 2016-08-27:
[image]

R Pi 3 EMI plot WiFi Full Power Ch 1 from FCC test report.

cyoung, 2018-01-17:
Would anyone care to consolidate all of this useful information into the [Wiki](https://github.com/cyoung/stratux/wiki)?