cyoung/stratux #204: 1090 does not gracefully restart

ssokol, 2016-01-13:
Running 0.5b4. Dual band w/ NooElec Nano 2 NESDRs. 

Tried restarting gen_gdl90 using 'service stratux restart' and found that after the restart I was unable to receive any 1090-ES messages. Message count remained at zero. Nothing in the log file. Tried killing dump1090 and it went [defunct]. Had to do a full power cycle before it resumed receiving.

Tested the "reboot" button and it that seems to work properly (i.e. it does not hang the reboot).

Anyone else seen this?

cyoung, 2016-01-13:
"service stratux restart" doesn't restart until the latest commits.

ssokol, 2016-01-13:
Oops - I left that out. I updated to the latest commit.

cyoung, 2016-01-13:
Hmm, and you were sure to do a "make install"?

I know that it's not exiting dump1090 cleanly when gen_gdl90 is killed - need to add some signal handlers. Try adding "pkill dump1090" before "pkill gen_gdl90" in the init script and see how it does.

ssokol, 2016-01-13:
No luck. I don't seem to be getting the [defunct] at this point, but if
regardless of the order of the killing, 1090 fails to restart cleanly. I
seem to always get one 1090 message and that's it.

commit 57a9fbf3 (2016-01-19): Catch SIGINT and SIGTERM - shutdown SDR functions.

Fixes #204.
closed 2016-01-19 by cyoung