cyoung/stratux #205: Weird, probably self-induced compile issue

ssokol, 2016-01-14:
This is probably me, not anything to do with Stratux. So first some background...

I've been working on solving a couple things. I've started with the 0.5b5 image and:

- Added a FAT32 partition for logs and config (/mnt/log) - 512 MB
- Added a FAT32 partition for update uploads (/mnt/update) - 512 MB
- Changed Stratux to write logs and config files to /mnt/log/ rather than /var/log and /etc/
- Adapted the fake-hwclock by changing its data file location to /mnt/log
- Revised the /etc/init.d script for hostapd to create a new hostapd.conf file that uses the MAC address as a suffix on the SSID
- Make the /boot and / filesystems read only
- Added scripts to switch / between read-only and read-write

Everything was working except for the Stratux config file. I realized I had a typo in my revision to gen_gdl90.go. So I switched to read-write, went back in, corrected the error and went to run 'make' again. Now it seems to be failing on the link stage:

    root@raspberrypi:~/stratux# make xgen_gdl90
    go get -t -d -v ./main ./test ./linux-mpu9150/mpu ./godump978 ./mpu6050 ./uatparse
    go build -ldflags "-X main.stratuxVersion=`git describe --tags --abbrev=0` -X main.stratuxBuild=`git log -n 1 --pretty=%H`" main/gen_gdl90.go main/traffic.go main/ry835ai.go main/network.go main/managementinterface.go main/sdr.go main/uibroadcast.go main/monotonic.go
    # command-line-arguments
    /root/go/pkg/tool/linux_arm/link: running gcc failed: exit status 1
    collect2: ld returned 1 exit status
    
    Makefile:16: recipe for target 'xgen_gdl90' failed
    make: *** [xgen_gdl90] Error 2
    root@raspberrypi:~/stratux#

Anyone have any idea what kind of stupid thing I've done. I KNOW it's not that the filesystem is still read-only. Tested that one. I complied 30 minutes before without any problem, so it did work. Any suggestions would be greatly appreciated.

ssokol, 2016-01-14:
I never figured out what caused this, but I imaged up another card and the problem did not resurface so I'm closing this.

closed 2016-01-14 by ssokol

cyoung, 2016-03-15:
Steve, I just got this error. The partition that I was compiling on filled up.

ssokol, 2016-03-15:
So should I change the rw / ro scripts to clear and unmount the /tmp tmpfs
partition, which should put it back to a standard directory on the root
partition?

cyoung, 2016-03-15:
That would work, or just expand the tmpfs part