cyoung/stratux #206: FWIW, SkyRadar AHRS info

Nokomis449, 2016-01-14:
For the people working on AHRS code - I started toying around with the AHRS on my SkyRadar DX, and noticed that the AHRS does not move at all (it indicates level flight) UNTIL its GPS light comes on.  At that point, the AHRS comes alive and will indicate correctly.  This is true in the moving plane and on the kitchen table.  At the table I can then disconnect the GPS antenna, and the AHRS continues to function correctly.  So the GPS seems to be providing something to the AHRS code that is needed for initial startup.  Just thought I'd pass this along.

duecedriver, 2016-01-14:
gps data is probably being fed to a kalman filter with the magnetometer data to provide heading 

in garmin logic, if the gps signal to the ahrs is down but the magnetometer is working, you will just get a gps degrade flag.. if the magnetometer is also down, you will get no heading

ssokol, 2016-01-14:
From what I remember from Samuel Cowan's articles on inertial compensation,
you can use rate-of-turn data from either a magnetometer or a GPS. Mag is
closer to real-time. GPS is less given to errors.

cyoung, 2016-01-26:
@Nokomis449 - thanks for the information.

closed 2016-01-26 by cyoung