cyoung/stratux #207: OT: Disk image question

ssokol, 2016-01-14:
Slightly off topic...

Anyone know the secret to making a disk image from a uSD card without the image being the full size of the storage media? I'm playing with adding a read/write partition for various items and making the root partition read only. When I go to create an image using dd, it images the full 8 GB of the card, rather than the 3.5 GB of the partitions on the card. The original Stratux images are, by comparison, much smaller - 3.1 GB before compression. Any hints?

Thanks,

-S

skypuppy, 2016-01-14:
0. write the image to your local hard drive.

1. try to reduce the image to 4 gb.

2. whether that works or not, use bzip to compress the image to 4 gb or
less.  It will probably work.

ssokol, 2016-01-14:
Hmm... Here's what I've tried thus far:

1) I imaged 0.5b5 onto a 32 GB card. It took up 3.1 GB of the space on the
card, leaving 28.9 GB available, presumably.
2) I created a couple of additional partitions, each 512 MB FAT32
3) I used dd to copy the card back to my hard drive

The resulting image was 15GB. The diskutil utility on the Mac shows the
card as having 15.9 GB of partitioned space:

/dev/disk4
   #:                       TYPE NAME                    SIZE
IDENTIFIER
   0:     FDisk_partition_scheme                        *15.9 GB    disk4
   1:             Windows_FAT_32 boot                    58.7 MB    disk4s1
   2:                      Linux                         3.2 GB     disk4s2
   3:                 DOS_FAT_32                         512.0 MB   disk4s3
   4:                 DOS_FAT_32                         512.0 MB   disk4s4

Weird, considering that the card is 32 GB, but perhaps the way it's
partitioned or maybe Macs can't handle 32 GB mixed filesystem cards? I
noticed that even Linux tops out at 5 partitions on the card.

I checked out an 8 GB card that just has the default Stratux image on it.
It shows up as:

/dev/disk4
   #:                       TYPE NAME                    SIZE
IDENTIFIER
   0:     FDisk_partition_scheme                        *7.9 GB     disk4
   1:             Windows_FAT_32 boot                    58.7 MB    disk4s1
   2:                      Linux                         3.2 GB     disk4s2

I assume if I were to make an image it would come out at ~8 GB.

@cyoung - are you just using a very small card? Or does the spindle process
somehow generate the image using some other process?

bradanlane, 2016-01-14:
There is a script out on the internet somewhere. I've got a copy of it that I've tweaked but I won't be at my computer for at least a day. 

The way it works is you make an image of the whole SDD. Then run the script on the img file. By default it will locate the last partition and then cut the img file down to the end of that partition. I customized the script to optionally take a size so it will adjust the last partition and then shrink the img down to either the size I specify or the smallest possible size / which ever is greater.

ssokol, 2016-01-14:
Found it - or at least instructions that got me where I wanted to go. For posterity, here's the deal:

First, get the partition from the disk using fdisk:

````
root@raspberrypi ~ # fdisk -l

Disk /dev/mmcblk0: 31.9 GB, 31914983424 bytes
4 heads, 16 sectors/track, 973968 cylinders, total 62333952 sectors
Units = sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disk identifier: 0x00066f9b

        Device Boot      Start         End      Blocks   Id  System
/dev/mmcblk0p1            8192      122879       57344    c  W95 FAT32 (LBA)
/dev/mmcblk0p2          122880     6399999     3138560   83  Linux
/dev/mmcblk0p3         6400000     7399999      500000    b  W95 FAT32
/dev/mmcblk0p4         7400000     8399999      500000    b  W95 FAT32
````

You only need to dd the blocks that matter, so in the above example, you need to go one past the last block (8399999) which would be 8400000. So the command to capture the relevant blocks is:

    dd if=/dev/rdisk4 bs=512 count=8400000 of=stratux_test_image.img

Making the image now. If it works, I'll come back in and close this.

ssokol, 2016-01-15:
Worked like a charm. Closing.

closed 2016-01-15 by ssokol