cyoung/stratux #209: WingX weirdness

duecedriver, 2016-01-17:
latest WingX release 5.5b

WingX will report say 9 1090 traffic but only be displaying 4

the stratus web page traffic pages generally shows what WingX is plotting say the 4... so where is the 9 coming from?

egid, 2016-01-19:
Any chance it's filtering distant targets? I remember there's an option for this in the totally impenetrable WingX settings, but I can't recall where.

duecedriver, 2016-01-19:
nope. I have the altitude filter turned off so it should be showing everything...

egid, 2016-01-20:
Huh. I'd give it a shot, but my Stratux is in pieces and it's raining cats and dogs out here today!

ScraboTower, 2016-01-20:
Same combo as the OP and not seeing the same issue.

duecedriver, 2016-01-20:
So the number of 1090 traffic reported in the wingx stratux pull down tab always matches the number of targets displayed on screen? 

Huh... I wonde what's wrong with mine. 

I still don't get but a handful and only out to 40 miles I wonder if I have a mediocre sdr

cyoung, 2016-03-08:
Reopen with more info to reproduce if still an issue.

closed 2016-03-08 by cyoung