cyoung/stratux #210: WebUI shows Weather "Receiving" even on 1090 only

Ergonomicmike, 2016-01-19:
Small nit.

Single Dongle Plain Vanilla Stratux running v0.5b5. The weather page shows me "Receiving" weather (Green tag) even when on only 1090. This persists after a reboot. Of course, Watching and Reports are zero.

cyoung, 2016-01-19:
The "Receiving" indicator is showing (correctly) that the web browser is receiving weather updates from the device - maybe the meaning of "Receiving" could be clearer or should be reevaluated.

Ergonomicmike, 2016-01-20:
Perhaps eliminated then? I thought it was an Activity Checker, indicating that weather products were coming thru. Since we already have a "Connected" marker in the status page, what new info does "Receiving" indicate?

Axtel4, 2016-01-20:
Maybe "Weather Connected / Not Receiving", "Weather Connected / Receiving"?

cyoung, 2016-01-26:
Need: three state display. Connected, Disconnected, Receiving. Or clarify the current "Receiving". @bradanlane, let's go with what you think is best.

bradanlane, 2016-01-26:
I'll normalize the messages to "connected"/"disconnected" and will verify it is logical an intuitive in all cases.

cyoung, 2016-01-29:
https://github.com/cyoung/stratux/commit/bbbcd3bbe16f67febb9955ff7d92908b82d2fa24

closed 2016-01-29 by cyoung