cyoung/stratux #212: FlyQ

yasbean, 2016-01-20:
Not really an issue, but just to let it be known to the developers, I finally had a chance to take my new Stratux v0.5b3 (978UAT + 1080ES + GPS + AHRS) box up for a flight.  I was busy trying get approaches in during variable weather, but I had time to notice that although the FlyQ ADS-B indicator was green over a 2.5 hour flight, I did not see any traffic or weather.  I wrote to Seattle Avionics, saying just this, then asking:  "It may, of course, just be a matter of spending more time playing with settings or rebooting, but is there a known problem at this point with FlyQ and the newest image of Stratux?"

I received the following reply from SA:

Great question!

We have been aware of many aviation related SDR projects for some time. Stratux is a interesting open source project for US market because of its support for 978MHz. Personally I look forward to the day when Stratux has better integration for GPS, 1090Mhz and AHRS.

We've heard form users who say its works fine for them but due to the experimental nature of the project its not supported by us. Meaning we're unable to help with setup, configuration, and validation. In fact we didn't modify FlyQ EFB to work Stratux, Stratux modified its computer code to work with FlyQ EFB. In any case, I'll write this up for our dev team to review.

Fly Safe!

Regards,
Katie Sue

Ergonomicmike, 2016-01-20:
Perhaps vote with your wallet then and try iFly v9.4 when it comes out? Unlike Seattle Avionics, iFly has moved to support Stratux, in both words and deeds.

yasbean, 2016-01-20:
Too late for that as I have a lifetime subscription. I have been using their x86 based Voyager since 2005, and got a deal too good to refuse.

cyoung, 2016-01-20:
https://seattleavionics.wordpress.com/2015/10/22/flyq-efb-version-2-1-1-now-available-fixes-for-ios-9-and-more/

I'm not sure about the "In fact we didn't modify FlyQ EFB to work Stratux, Stratux modified its computer code to work with FlyQ EFB." part - so stratux has gained rogue support from FlyQ? :) @stevepodradchik

@yasbean - I think the issue might have more to do with the setup. Have you tried looking at http://192.168.10.1/ while in the air to see if you're getting UAT and 1090ES?

yasbean, 2016-01-20:
While still on the pad, I was happily surprised by the number of hits on both frequencies. If others have been successful, I will just have to go out and fly again in better weather and under more leisure.

ScraboTower, 2016-01-20:
Perhaps vote with your wallet then and try WingX, at least Hilton was open to it and embraced it with open arms......unlike some.

duecedriver, 2016-01-20:
this is the way its going to be from now on in aviation... as technology and innovation catch up to legacy players, new and exciting products and tech ranging from DIY to new manufactures willing to offer good products at better prices... 

not holding my breath on regulator or liability reform from the FAA's long overdue part 23 re-write... but that could also open the flood gates.. best we can hope for is that the FAA begins to allow quality avionics that have been used in the experimental side over to part 91 operations.

personally I think that should be the driving dividing line in regulatory guidance, to a point.. part 91 (private not for profit) vs for profit part 135,121 et.al

we shall see..

for those devs willing to stick it out with the stratux development process and even more so to those actively contributing here to make it better... this buds for you...

jzeevi, 2016-01-20:
Um, are you having that beer or getting the dev team one? 

 

                J

 

(note, I’m not on the dev team – just commenting and testing)

duecedriver, 2016-01-20:
I am semi retired right now so any time is a good time for a bud...

I had input during the upgrade of 2 USAF major weapons systems by tier 1 defense contractors to glass cockpits as well as training the aircrew on the transition.  Going back to the early 2000s consulting with a number of civilian aviation apps in the windows CE environment well before the EFB concept was ever embraced, to the present... so I know how hard these guys work to get it right, innovate, and try to set themselves above the competition....

duecedriver, 2016-01-20:
@jzeevi 

besides.. it goes good with your pizza

yasbean, 2016-01-26:
I just upgraded to v0.5b6, and from the comfort of the ground, FlyQ is now showing plenty of traffic. I can even hear it flying around as I watch it on FlyQ.

closed 2016-01-26 by yasbean

cyoung, 2016-01-27:
Excellent, thanks for checking back.