cyoung/stratux #214: AHRS Cycles on and off

jbast, 2016-01-21:
RY835AI is soldered to the pi. When enabled, the AHRS cycles on and off in the stratux interface. No data seems to make it to Foreflight. GPS works fine. Suggestions?

ScraboTower, 2016-01-21:
Foreflight does not want to play in the Stratux sandbox anymore ... 

quote "The simultaneous feeding of data to both our operational use and flight simulator use ports was undesirable (unexpected, untested, and not a priority for us to debug, harden, and ultimately support at this time), as was the injection of Stratux AHRS data for operational use. We know what we get at this point with the Stratus 2 and FlightStream AHRS - both Appareo and Garmin have years of experience building certified inertial systems. Lots of time and testing is invested in getting that performance really dialed in, and other priorities are at the top of the stack currently"

jbast, 2016-01-22:
Same behavior in WingX, and the AHRS cycle is reflected in the Stratux UI. Boo to foreflight for not playing fair!

AvSquirrel, 2016-01-22:
The on-off cycling is a sign that the AHRS chip and pressure sensor aren't talking to the RPi. How are you connected? At a minimum, you'll need the following pins connected for AHRS:

VDD --> GPIO Pin 2 (+5V)
GND --> GPIO Pin 6 (Ground)
SDA1 --> GPIO Pin 3
SCL1 --> GPIO Pin 5

ScraboTower, 2016-01-22:
I had that same issue a while back, my issue was the SCL1 to pin 5 was not seated fully in the Pi side. As @AvSquirrel  says, check your pin mapping and that all connections are good.

jbast, 2016-01-22:
Thanks @AvSquirrel that fixed the issue. got my wires crossed. Appreciate it!

closed 2016-01-22 by jbast