cyoung/stratux #215: User Suggestion: VHF Radio Receiver

ssokol, 2016-01-21:
Disclosure: This might be completely stupid. Just tossing it out for a thorough beating by the group...

I had a backer call me up and ask if it would be possible to use the SDRs as a backup radio. He was initially thinking of a full transceiver, and I explained that the SDRs were strictly receivers. So he suggested that they could still be a good back up receiver - perhaps for listening to ATIS / AWOS.

I realize that the antennas we're using are not tuned for the low MHz range of VHF, but what do you think? Would it be useful to be able to pull one of the SDRs off of data modem duty and assign it as an AM receiver? Perhaps output the audio to the Pi's analog jack or stream it to a web page?

skypuppy, 2016-01-21:
Stick another $20 sdr on?  Or even use a separate RPi2 with it's own
receiver, just for "miscellaneous" other receiver functions?  :)
ATIS/AWOS, instrument landing system with much better graphics
(backup!), and so on and so on...

cyoung, 2016-01-21:
Great idea! Streaming via WiFi and/or playing to the RPi's audio-out are both options.

While we're talking about new features, is there any interest in serialout? I made an image for someone to test on a Chelton EFIS. Anyone else want to test? https://www.dropbox.com/s/owvhmhxt3w6c31n/stratux-serialout_test.img.zip?dl=0 + http://amzn.com/B0007T27H8. I've bench tested it.

ssokol, 2016-01-21:
I'm sure there's user demand for a serial out option. I don't have an EFIS (funny, just wrote a blog post about that) but I bet there are plenty of people over on vansairforce.net who would probably jump at the chance.

ScraboTower, 2016-01-21:
I would love an "IN" so I could feed the ZAON data into WingX via Stratux WIFI - currently its either ZAON or Stratux I have connected via Wifi. 

The iLevil products allows you connect ZAON serially to it and it feeds both ADS-B and ZAON traffic.

Maybe I am the last of a dying breed using ZAON but its on every flight.

cyoung, 2016-01-21:
ZAON send out GDL90?

ScraboTower, 2016-01-21:
Don't have the documentation at hand but @HiltonG might know, he implemented it into WingX about 3+ years ago.

224XS, 2016-01-24:
First package I ran on my mac with my first SDR was GQRX.  Had it tune my field's Unicom and zap!
Nice AM reception on 122.72 MHz with reasonable sensitivity using the supplied short whip. 
Equivalent code exists for the Pi. Unlikely that the AM receiver code would co-execute 
with Stratux, and  it needs a terminal front-end for tuning, setting demodulation, etc.  
Probably cheaper and more practical to get a receive-only airband handheld receiver. 
Of course, if you want to experiment, one could rewrite the Pi version of GQRX or RTL-SDR 
to use a wifi link to a tablet front-end and maybe even bluetooth audio out receiver

ssokol, 2016-01-24:
It's interesting in part as an experiment, and in part because of the price
point. VHF COM is nice, but I think the "money" is probably VOR / ILS. You
could never get it certified, but it could make one heck of a good backup.

peepsnet, 2016-01-31:
I like the idea of a backup receiver. But the ability to easily tune the freq is a bit of a task. I would assume I would have open the webGUI and enter the freq there. That seems cumbersome. 

If it was possible to have the EFB sent http or ssh commands to switch an SDR and stream or send to audio out and use an adaptor for your headset or something like that. 

You could get the ATIS at the APT that way for sure

Some might say the backup receiver would be an option but to be useful you would need an amplified speaker or hook up your headset. if your headset is hooked up then you cannot use the a/c transmitter(if working). 

As an AirTraffic Controller I can tell you that if you have a com fail it would be nice to still be able to hear ATC. "If you can hear the tower, IDENT" type stuff. 

As far as ILS. AHRS should be enough in an emergency? The VOR is an option but now it looks like an APP would be needed to handle to custom capabilities of the "FlightBox"

If there was a companion APP to the Flight Box then we could take full advantage of any "extra" functionality we would like to add. On android I can envision an APP that has a status bar icon when connected to the "Stratux" SSID that way any other functions can be accessed quickly. The app could function in an overlay to the current active application allowing you to change functions without leafing the EFB app

cyoung, 2016-03-08:
Consolidating to #228.

closed 2016-03-08 by cyoung